RIOT-OS/RIOT #6281: Hello world with Eclipse IDE

juttTechnology, 2017-01-02:
i am working with eclipse IDE to build the hello world example.
i am blocked on the setp Number 5 of this tutorial. [Link](https://github.com/RIOT-OS/RIOT/wiki/Using-the-Eclipse-IDE-for-C-and-CPP-Developers,-Howto)
when i try to run this command 
make BOARD=nrf52dk eclipsesym
i got the following error.
[image]

thank you

OlegHahm, 2017-01-03:
Well, my first guess, looking at the output, would be that you will have to install wget or curl.

juttTechnology, 2017-01-05:
i am using windows 10. when i try to install the curl, i was unsuccessful.
i seems like, he is working on linux or ubuntu.
on windows its not easy to put this working.

gebart, 2017-01-05:
On Thu, Jan 5, 2017 at 1:49 PM, juttTechnology <notifications@github.com> wrote:
>
> i am using windows 10. when i try to install the curl, i was unsuccessful.
> i seems like, he is working on linux or ubuntu.
> on windows its not easy to put this working.
>

Most RIOT developers are using some variant of Linux or MacOS for
development. There are some people writing on the mailing lists about
working with RIOT in Windows, but I haven't followed the developments.

closed 2017-01-12 by juttTechnology