RIOT-OS/RIOT #6282: git-cache broken ?

aabadie, 2017-01-03:
Just tried a fresh build of nanocoap_server against latest master () but failed to clone the kaspar030/sock repository.

The incriminated commit is 63bfbfabecf555b58d15dadf7ae16d83f710f28f.

The error message I have is:
```
Building application "nanocoap_server" for "samr21-xpro" with MCU "samd21".

"make" -C /home/aabadie/softs/src/riot/RIOT/pkg/nanocoap
rm -Rf /home/aabadie/softs/src/riot/RIOT/examples/nanocoap_server/bin/pkg/samr21-xpro/nanocoap
mkdir -p /home/aabadie/softs/src/riot/RIOT/examples/nanocoap_server/bin/pkg/samr21-xpro/nanocoap
/home/aabadie/softs/src/riot/RIOT/dist/tools/git/git-cache clone "https://github.com/kaspar030/sock" "cf2aff1b4322cee06bb5b24d0215bb764372fba7" "/home/aabadie/softs/src/riot/RIOT/examples/nanocoap_server/bin/pkg/samr21-xpro/nanocoap"
fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
Cloning into '/home/aabadie/softs/src/riot/RIOT/examples/nanocoap_server/bin/pkg/samr21-xpro/nanocoap'...
fatal: reference repository '/home/aabadie/.gitcache' is not a local repository.
/home/aabadie/softs/src/riot/RIOT/pkg/pkg.mk:23: recipe for target '/home/aabadie/softs/src/riot/RIOT/examples/nanocoap_server/bin/pkg/samr21-xpro/nanocoap/.git-downloaded' failed
make[1]: *** [/home/aabadie/softs/src/riot/RIOT/examples/nanocoap_server/bin/pkg/samr21-xpro/nanocoap/.git-downloaded] Error 128
/home/aabadie/softs/src/riot/RIOT/Makefile.include:317: recipe for target '/home/aabadie/softs/src/riot/RIOT/examples/nanocoap_server/bin/samr21-xpro/nanocoap.a' failed
make: *** [/home/aabadie/softs/src/riot/RIOT/examples/nanocoap_server/bin/samr21-xpro/nanocoap.a] Error 2
```

OlegHahm, 2017-01-03:
I failed to reproduce. Which git version do you use?

aabadie, 2017-01-03:
```
$ git --version
git version 2.9.3
```
with Ubuntu 16.10

OlegHahm, 2017-01-03:
Okay, it doesn't seem to be a problem of the git version. I don't get the error with `git version 2.9.3`, either. Strange.

gebart, 2017-01-04:
Do you have git-cache installed?
Is it initialized? Could be that git cache doesn't handle errors gracefully
when there's something wrong with the cache directory.

aabadie, 2017-01-04:
----- Mail original -----

> Do you have git-cache installed?
Yes, git-cache is provided by RIOT itself (in `dist/tools/git`) 

> Is it initialized? Could be that git cache doesn't handle errors gracefully
> when there's something wrong with the cache directory.
it's working for me with 5fd26e28e7cf7450344ffe22abf97a74d8069526, the commit just before 0c47cf70931942b353eb9fc04f3bae3e1ea92c2e (the one that fails). 

Looking at the changes, it's the `git clone --reference` that fails.

aabadie, 2017-01-04:
Thanks @BytesGalore.

I should have read the README first. My problem was that I had a .gitcache directory but it was empty (not initialized with `git-cache init`).

#6289 makes sense BTW by not activating git caching by default.

BytesGalore, 2017-01-04:
@aabadie it fails also if the `.gitcache` folder is not present, at least for me.

aabadie, 2017-01-05:
> @aabadie it fails also if the .gitcache folder is not present, at least for me.

Same here but your PR fixes it.

I may have missed something but after playing a bit with this git cache functionality, I found the actual behavior a bit weird : if the git cache is correctly configured (`dist/tools/git/git-cache init`), the repositories are still cloned in `<application>/bin/pkg/<pkg-name>` and `~/.gitcache` only contains refs commits but no objects.
I'm wondering what's the benefit here ?
Example with microcoap:
```
$ dist/tools/git/git-cache list
git://github.com/1248/microcoap.git
$ du -sh bin/pkg/samr21-xpro/microcoap/
324K	bin/pkg/samr21-xpro/microcoap/
$ du -sh ~/.gitcache/
164K	/home/aabadie/.gitcache/
```
The content of the microcoap pkg is still duplicated in the `bin/pkg` of each application that uses it.

PR #6289 (2017-01-05): git-cache: consider if git_cache dir is present
fixes #6282
closed 2017-01-05 by kYc0o