RIOT-OS/RIOT #6287: stm32f1: xtimer clock running at twice the speed

bergzand, 2017-01-04:
I've noticed this when running the timer_periodic_wakeup example from the RIOT examples directory.
cpu: STM32F103RCBT6
Hardware: [maple-mini](https://github.com/phectori/RIOT/tree/maple-mini/boards/maple-mini) (Clock configuration seems identical to the nucleo-f103 board

expected to receive a message every second. A message is received every 500ms.
RIOT seems to think that a second has elapsed:
```
main(): This is RIOT! (Version: 2017.01-devel-295-g6a757-dystopia-maple-mini-work)
slept until 1014640
slept until 2014640
slept until 3014640
```

bergzand, 2017-01-04:
This seems to be caused by the unification of the timer code for the stm32 series. Looking at the code previously used for the [stm32f1 for the timer prescaler](https://github.com/RIOT-OS/RIOT/blob/2016.10/cpu/stm32f1/periph/timer.c#L59):
```dev(tim)->PSC = ((CLOCK_CORECLOCK / freq) - 1);```
and now it is:
```dev(tim)->PSC = ((periph_apb_clk(timer_config[tim].bus) / freq) - 1);```
apb_clk is the APB1 bus, running at half the speed of the core clock, thus generating a prescaler value of half the required value.

Although the timers of the stm32f1 are connected to the APB1 bus and clock, when looking at the manual page 125, there seems to be a multiplier block in between which I suspects generates a clock at core clock speed for those timers.

OlegHahm, 2017-01-04:
I can confirm the bug for the IoT-LAB M3 platform and `git bisect` also pointed to #6184.

haukepetersen, 2017-01-05:
fixed with #6291

PR #6291 (2017-01-05): cpu/stm32: fix timer speed for STM32F1 and F2
fixes #6287
closed 2017-01-05 by OlegHahm