RIOT-OS/RIOT #6305: pyterm feature request: blank mode and `>` handling

haukepetersen, 2017-01-10:
When doing some measurements on a node, I often copy-paste output from the shell and use it for quick analysis in spreadsheets or similar. For this some parts of the pyterm output is in the way, so here my idea:

a) can we add an option to pyterm that will prevent pyterm from printing the date in front of each output line?
b) we could make the handling of the RIOT shell's `>` character smarter: if input is coming from the keyboard and is send to the board, print the `>`, if the `>` by RIOT is followed by further input from the RIOT board, drop the `>`.

@OlegHahm, what do you say?

cgundogan, 2017-01-10:
adding to this, but not addressing the feature request; if the macro variable `SHELL_NO_PROMPT` is set, then the output of `>` is ignored.
This could be used in anticipation before compiling the binary for performing measurements etc.

OlegHahm, 2017-01-10:
> can we add an option to pyterm that will prevent pyterm from printing the date in front of each output line?

From the top of my head: the output is configurable. Let me take a look...

> b) we could make the handling of the RIOT shell's > character smarter: if input is coming from the keyboard and is send to the board, print the >, if the > by RIOT is followed by further input from the RIOT board, drop the >.

Sure, makes sense.

OlegHahm, 2017-01-10:
See #6308 (the feature was there, but broken). If you want to have no prefix at all, add this to your pyterm config (per default `${HOME}/.pyterm/pyterm-${HOSTNAME}$.conf`:
```ini
fmt_str =
```

If you have no pyterm config file yet, you can create one like this:
```ini
[general]
port = /dev/ttyUSB1
fmt_str =
```

haukepetersen, 2017-01-10:
I see. But am I able to define a custom config file through some command line arg through the RIOT build system? As an option like overriding the fmt string is something I would do only on specific occasions and I don't want to edit the pyterm conf file every time...

OlegHahm, 2017-01-10:
You can specify the config file via the command line:
```
  -c CONFIG, --config CONFIG
                        Specify the config filename, default is pyterm-${HOST}.conf
```
Reading the formatter from the command line seems clumsy to me, but I could give it a try.

OlegHahm, 2017-01-10:
How about #6309 - `-np`/`--noprefix` does what you requested.

OlegHahm, 2017-01-10:
> b) we could make the handling of the RIOT shell's > character smarter: if input is coming from the keyboard and is send to the board, print the >, if the > by RIOT is followed by further input from the RIOT board, drop the >.

Hm, thinking about this again: what happens if the executing of a RIOT command takes longer, e.g. the ping command? But maybe pyterm could scan for RIOT's `>` anyway and print it immediately, i.e. without waiting for the newline.

haukepetersen, 2017-01-10:
> How about #6309 - -np/--noprefix? It should do what you requested.

Yes, that was what I was looking for

haukepetersen, 2017-01-10:
> But maybe pyterm could scan for RIOT's > anyway and print it immediately, i.e. without waiting for the newline.

I think I'd like that

OlegHahm, 2017-01-10:
See #6310 for b).

haukepetersen, 2017-02-07:
For me everything I asked for is working now -> closed

closed 2017-02-07 by haukepetersen