RIOT-OS/RIOT #6330: #6321 broke git-cache in OS X

kYc0o, 2017-01-12:
While merging #6321 no package can be downloaded using git-cache on OS X.

This is due to the addition of line 48 in git-cache.

```console
RIOT/dist/tools/git/git-cache: line 48: flock: command not found
```

`flock` doesn't exist in OS X, and I didn't find any replacement so far. Is this really necessary?

The same "command not found" is happening with `md5sum` but a direct replacement exists as `md5`.
It's on line 111:

```console
RIOT/dist/tools/git/git-cache: line 111: md5sum: command not found
```

But this doesn't lead to a crash (expected behaviour?).

miri64, 2017-01-12:
Maybe unrelated but why does `git-cache` use md5. Isn't Git all about SHA-1?

kaspar030, 2017-01-12:
> flock doesn't exist in OS X, and I didn't find any replacement so far. Is this really necessary?

The locking is necessary :( I'll research how to do it on OSX.

> Maybe unrelated but why does git-cache use md5. Isn't Git all about SHA-1?

git-cache uses md5sum to create a unique name for each URL. git needs a name in addition to an URL for every remote, so I just use the md5sum of the URL.

@kYc0o Does OSX have sha1sum?

kYc0o, 2017-01-12:
> @kYc0o Does OSX have sha1sum?

OS X have both MD5 and SHA-1, but with a different name: `md5` for md5sum and `shasum -a 1` for sha1sum.

miri64, 2017-01-12:
> git-cache uses md5sum to create a unique name for each URL.

Aren't URL's unique by nature? Unless you are trying to save space it would suffice to use their identity. On the other hand with SHA-1 you could store them as blobs in your git-cache repository and check if they already exist just by asking git about the existence of the blob :D

smlng, 2017-01-12:
to solve this, I installed the following with Homebrew:

```
brew install md5sha1sum
brew tap discoteq/discoteq
brew install flock
```

see also [here for OSX flock](https://github.com/discoteq/flock)

@kaspar030 wouldn't it be better/safer to lock a file, e.g. `git-cache.lock`, instead of filediscriptor number `9`?

kYc0o, 2017-01-12:
I know, but we are adding more and more requirements to have a "stable" environment in OS X, I'd like to have something less hacked...

smlng, 2017-01-12:
@kYc0o I agree, we should try to use (native) tools that are available on Linux and other Unix-like environments such as macOS by default.

miri64, 2017-01-12:
That's why it might be a good idea to use git itself to store the URLs. This you should have installed ;-). And git porcelain is very powerful.

kaspar030, 2017-01-12:
> That's why it might be a good idea to use git itself to store the URLs.

Thing is, git-cache needs to do "git remote add $name $url" at some point, but it doesn't care about the name. It just needs to be unique for the cache. using md5sum seemed simple. :)

miri64, 2017-01-12:
> Thing is, git-cache needs to do "git remote add $name $url" at some point, but it doesn't care about the name. It just needs to be unique for the cache. using md5sum seemed simple. :)

Ahh, I thought you use it for some comparison. But still, use git for that:

```bash
_remote_name() {
    echo "$*" | git hash-object [-w] --stdin
}
```

(The `-w` would write the created object into the git repo if desired)

kaspar030, 2017-01-12:
> use git for that:

good idea!

kaspar030, 2017-01-12:
@kYc0o could you test https://github.com/kaspar030/git-cache/commits/master pls? (still no locking, but uses git for hashing)

miri64, 2017-01-12:
And regarding `flock`, haven't found any yet, but considering that many git commands are written in shell script (`git bisect` or `git rebase` to name a few popular ;-)) it wouldn't surprise me if there is a locking mechanism somewhere in git

kaspar030, 2017-01-12:
@kYc0o I've also added locking for OSX, using shlock.

@miri64 Didn't find anything, either. Probably the do it in perl...

kYc0o, 2017-01-12:
Works like a charm! PR it quickly I'm anxious to see this beauty on master! 😃

PR #6334 (2017-01-12): dist: tools: git-cache: bump version
adds OSX fixes for md5sum, flock

Fixes #6330.
closed 2017-01-12 by miri64