RIOT-OS/RIOT #6331: RTC alarm triggered with 1s delay

jia200x, 2017-01-12:
Hi.

I have been testing RTC features. 
When I try to do stuffs like:

```
struct tm time;
rtc_get_time(&time);
time.tm_sec  += 10;
rtc_set_alarm(&time, cb, 0);
```

, the alarm is triggered n+1 seconds after the rtc_set_alarm call (11 seconds in this case). I added 2 printf in the cb function of periph_rtc example and this is the output:

```
2017-01-12 10:54:20,987 - INFO # RIOT RTC low-level driver test
2017-01-12 10:54:20,991 - INFO # This test will display 'Alarm' in 10 seconds
2017-01-12 10:54:20,991 - INFO # 
2017-01-12 10:54:20,993 - INFO # Initializing the RTC driver
2017-01-12 10:54:21,007 - INFO # Setting clock to 2011-12-13 14:15:15
2017-01-12 10:54:21,051 - INFO # Clock set to 2011-12-13 14:15:15
2017-01-12 10:54:21,054 - INFO # Setting alarm to 2011-12-13 14:15:25
2017-01-12 10:54:21,069 - INFO # Alarm set to 2011-12-13 14:15:25
2017-01-12 10:54:21,074 - INFO # The alarm should trigger every 10 seconds for 4 times.
2017-01-12 10:54:32,024 - INFO # Alarm!
2017-01-12 10:54:32,062 - INFO # ISR: rtc_get_time 2011-12-13 14:15:26
2017-01-12 10:54:32,065 - INFO # ISR: rtc_get_alarm 2011-12-13 14:15:25
2017-01-12 10:54:42,024 - INFO # Alarm!
2017-01-12 10:54:42,062 - INFO # ISR: rtc_get_time 2011-12-13 14:15:36
2017-01-12 10:54:42,065 - INFO # ISR: rtc_get_alarm 2011-12-13 14:15:35
2017-01-12 10:54:52,024 - INFO # Alarm!
2017-01-12 10:54:52,062 - INFO # ISR: rtc_get_time 2011-12-13 14:15:46
2017-01-12 10:54:52,065 - INFO # ISR: rtc_get_alarm 2011-12-13 14:15:45
2017-01-12 10:55:02,024 - INFO # Alarm!
2017-01-12 10:55:02,062 - INFO # ISR: rtc_get_time 2011-12-13 14:15:56
2017-01-12 10:55:02,065 - INFO # ISR: rtc_get_alarm 2011-12-13 14:15:55
```

rtc_get_time > rtc_get_alarm by 1 sec, so the cb is being triggered every 9+1 seconds. The test passes, but I'm not sure if this is the expected behavior.

Any comments?
Cheers

gebart, 2017-01-12:
Which board are you running on? This is likely a bug in the periph driver for that CPU.

gebart, 2017-01-12:
I'm seeing the same issue in tests/periph_rtc on kinetis (mulle)

jia200x, 2017-01-12:
Hi.

I have been using an NZ32-SC151 (stm32l1) and a samr21. Both have the same
behavior.

Cheers

<https://github.com/RIOT-OS/RIOT/issues/6331#issuecomment-272174987>, or mute
the thread

kYc0o, 2017-04-25:
Can you verify (or someone) that this is solved by #6504 ?

jia200x, 2017-08-12:
I'm still seeing the issue in samr21.

gebart, 2017-08-15:
@jia200x you should look up the manual for your CPU if it has the same behaviour as Kinetis. On those devices, the alarm interrupt is triggered when the seconds register is equal to the alarm register, and the seconds register increments. Which means that reading the current time after the alarm has triggered would show that the time is one second past the alarm.
What we expected, and I would guess most API users would expect, is that the test in the top post would make the alarm trigger immediately when the time increments to become equal to the alarm time. So, check your CPU reference manual/datasheet and verify that the periph/rtt driver is doing the right thing.

aabadie, 2017-09-13:
> What we expected, and I would guess most API users would expect, is that the test in the top post would make the alarm trigger immediately when the time increments to become equal to the alarm time

+1

PR #7476 (2017-10-09): kinetis: Adjust RTT alarm trigger time
The RTT alarm will trigger when the RTT seconds register is equal to the set alarm target, instead of triggering at the following second. This makes the API behave more like what users seem to expect.

Fixes #6331 for Kinetis
closed 2017-10-09 by kYc0o