RIOT-OS/RIOT #6358: arduino uno : xtimer accuracy problem

mali, 2017-01-14:
xtimer accuracy is bad under arduino uno and duemilanove platforms.
e.g. examples/timer_periodic_wakeup should wake up every second.
Instead it wakes up very often, something like 50ms or so.

mali, 2017-01-15:
Does it works well on atmega2560 ?
For what I understand, xtimer use Timer0 for wich atmega2560 and atmega328p have the same specs.
By the way, I don't understand mega2560 xtimer configuration : 
`#define XTIMER_WIDTH                (16)` should be `(8)`
`#define XTIMER_SHIFT                (2)` should be `(4)` 
Anybody with a meg2560 who can confirm ?

kYc0o, 2017-01-16:
I did extensive tests on this platform and works correctly with current configuration. Can you elaborate a bit more what you don't understand?

mali, 2017-01-16:
@kYc0o :
I think that current xtimer implementation of mega2560 use TIMER0 which is an 8 bits timer, but it's defined as `XTIMER_WIDTH (16)` which means 16 bits resolution.
The same way, according to [XTIMER_SHIFT doc](https://riot-os.org/api/group__sys__xtimer.html#gafd8fce8e5a8b0042399523706cfb6f60), and knowing that atmega2560 has a 16Mhz Clock, it should be defined `XTIMER_SHIFT (4)` not `(2)`

kYc0o, 2017-01-16:
OK, I'll test asap since I'm experiencing various XTIMER problems on arduino-mega2560.

kYc0o, 2017-01-16:
Well actually we are using TIMER1 which is 16 bits. For the XTIMER_SHIFT the timer will run at 250KHz so the value it's right.

mali, 2017-01-17:
Ok, got it [RIOT TIMER0](https://github.com/RIOT-OS/RIOT/blob/master/boards/arduino-mega2560/include/periph_conf.h#L45) is in fact mcu TIMER1.
Anyway, these two boards are similar in terms of TIMER0/1/2 and it's done the same way for [Uno](https://github.com/RIOT-OS/RIOT/blob/master/boards/arduino-common/include/periph_conf.h#L46) as it's done for the Mega2560. So problem should be elsewhere ....

haukepetersen, 2017-01-18:
@kYc0o are you still looking into this?

haukepetersen, 2017-01-18:
see also #6392

kYc0o, 2017-01-20:
My lasts tests showed a good xtimer behaviour with this board. @mali can you make a last check and see if it still fails?

mali, 2017-01-20:
@kYc0o : which commit fix this please ?

kYc0o, 2017-01-20:
Unfortunately I don't know and I didn't dig into it, but it was strange...

mali, 2017-01-22:
I've run test/xtimer_usleep, examples/timer_preriodic_wakeup and tests/saul and well .. it works :)

mali, 2017-01-23:
Don't know how nor why, but it has been fixed by PR #6320 
I close

closed 2017-01-23 by mali

kamejoko80, 2017-04-02:
No concerned with this topic but I have a question, how to implement xtimer with count down mode timer ?

I'm porting RIOT to ARM cortex-A9, the CPU has 32bit timers with count down mode only, it will generate interrupt when timer reaches to Zero. Now I'm stuck at xtimer implementation, almost another platform are using count up mode timer. 
Please show me some hints. My porting task is almost done except xtimer.

gebart, 2017-04-02:
The kinetis hardware lptmr (periph/timer.c) only support down counting mode, you could look there for ideas.

haukepetersen, 2017-04-03:
Hej @kamejoko80: actually the solution is not too difficult: you don't have to care about `xtimer`, all you need to care about is the implementation `periph/timer.h`. And a here the only thing that is a little tricky is the `time_read()` function. Here you probably have to us some kind of SW counter on the side.

kamejoko80, 2017-04-11:
Hello haukepetersen,

Thanks so much for your reply.

Yes, I've implemented the following wrapper layer functions for xtimer.

void timer_irq(void);
int timer_init(tim_t tim, unsigned long freq, timer_cb_t cb, void *arg);
unsigned int timer_read(tim_t tim);
int timer_set_absolute(tim_t tim, int channel, unsigned int value);

But I'm not clear about xtimer behavior in RIOT, are almost platform using up-counter timer which counts from 0-0xFFFFFFFF and generates interrupt when it reaches 0xFFFFFFFF ? 

My platform CPU is S5P4418, it has down-counter with counter buffer register TCNTB0, firstly it loads TCNTB0 value into timer counter and counts down from this value, when the counter reaches to zero it generates interrupt request to the CPU. In RIOT point of view does it expect that the counter counting down from 0xFFFFFFFF to zero instead of TCNTB0 value?

haukepetersen, 2017-04-11:
Hej, I am not quite sure if I understand your question. As already said: you don't have to care for xtimer. Just implement the `periph/timer.h` interface as if it was an upcounting timer, then you don't have any trouble.

kamejoko80, 2017-04-12:
Hello haukepetersen,

I didn't touch xtimer, just implemented the following functions : 

void timer_irq(void);
int timer_init(tim_t tim, unsigned long freq, timer_cb_t cb, void *arg);
unsigned int timer_read(tim_t tim);
int timer_set_absolute(tim_t tim, int channel, unsigned int value);

I've tested with example timer_periodic_wakeup and realized that it didn't work properly.
I'm investigating timer_read() and timer_set_absolute() to know their behavior and adjust the implementation.

kamejoko80, 2017-04-12:
Hello

I think I know the problem

Normally up-counter timer counts from 0x0 to 0xFFFFFFFF, timer_read() function returns the current counter value, and timer_set_absolute() sets the compare value (counter matching register), when counter matches compare value then timer interrupt happen. I missunderstood compare value with timer reload value. 

Unfortunately, S5P4418 doesn't have any compare register, it has only reloaded register TCNTB0 and down-counter mode. I've used a common variable to accumulate the clock ticks, and pass the offset value from xtimer implementation to TCNTB0  register. Now timer_periodic_wakeup seems work well, I'll upload my repo into GitHub server later.

kamejoko80, 2017-04-16:
Hello haukepetersen,

I've finished porting RIOT to my custom board and I've uploaded my workspace into github server as below link :

https://github.com/kamejoko80/RIOT-S5P4418

In my porting task has three  parts : 

1> Add support for ARMv7-A (in /cpu/armv7a_common),  we can reuse this module if we want to implement with another platform in the future such as RPi, Beagle Bone Black...

2> Add support for S5P4418 CPU /cpu/s5p4418

3> Add support for Jellyfish (JF4418) custom development board, and my porting task maybe can run on NanoPi2 development board also. 

Details of the commit as below: 

https://github.com/kamejoko80/RIOT-S5P4418/commit/7961e0c35169b641e3c6cbae50a48279b78916f3
https://github.com/kamejoko80/RIOT-S5P4418/commit/9108e4011c089a9f24bdfa3d9c11419273542ca2
https://github.com/kamejoko80/RIOT-S5P4418/commit/d1e537a0c8d33df9eda4bbd54b5783f76635d133

I've also tested with some examples such as timer_periodic_wakeup, ipc_pingpong, gnrc_networking and see that it worked well.

May I distribute my porting task into RIOT mainline? Would you please help for reviewing?
Thank you very much.

miri64, 2017-04-16:
Hi @kamejoko80,
Did you have a look at [our Wiki](https://github.com/RIOT-OS/RIOT/wiki/Contributing-to-RIOT#contribute-code)? You need to put the commits into a GitHub [fork](https://help.github.com/articles/about-forks/) of RIOT to create a [pull request](https://help.github.com/articles/about-pull-requests/) (PR for short).
Good thing is: you already seem to [have one](https://github.com/kamejoko80/RIOT). Since your other repository has none of RIOT's history (you just seem to have copied the files) you need to transfer the patches in another way: In your terminal go to your local repository you pushed to https://github.com/kamejoko80/RIOT-S5P4418 from and type

```sh
git format-patch ce52f9b
```

into your terminal. This will create three patch files corresponding to your three commits. Now go to your local copy of your RIOT fork (or clone it freshly `git clone git@github.com:kamejoko80/RIOT.git`) and first update the master branch of your fork:

```sh
git remote add upstream https://github.com/RIOT-OS/RIOT.git
git fetch upstream
git checkout -f master
git reset --hard upstream/master # the "nicer" would be to pull from upstream master here, just making sure this comes out cleanly
```

Then go ahead and create a new branch with your patches, your PR will be based on:

```sh
git checkout -b pr/s5p4418-initial  # or whatever you would like to call it
git apply path/to/RIOT-S4P4418/*.patch
git push -u origin pr/s5p4418-initial
```

Now go to https://github.com/kamejoko80/RIOT. Usually the suggestion to create a PR now pops up above the code. If not (which probably means you did something wrong ;-)), press the "New pull request" button, select RIOT-OS/RIOT's master as the base and the branch created above from kamejoko80/RIOT as compare. Write up some some nice title (prefixed with the name of your ported CPU) and description and hit submit. Then someone will start to review your code (which, considering the size of your change set, might take a while and could also include a lot of nitpicky style comments, just so you are warned ;-)).

kamejoko80, 2017-04-17:
Hello miri64,

Thanks so much for your kindly support. I've created a pull request https://github.com/RIOT-OS/RIOT/pull/6915.

This is the first time that I work with RIOT and git, and I'm sorry for my mistake if any.

Thank you very much.