RIOT-OS/RIOT #6367: doc: doxygen's automatic file matching files

OlegHahm, 2017-01-15:
Currently we get a large number for warning like this when building doxygen:
```
/tmp/RIOT/cpu/atmega1281/include/cpu_conf.h:12: warning: the name `/tmp/RIOT/cpu/atmega1281/include/cpu_conf.h' supplied as the second argument in the \file statement matches the following input files:
   /tmp/RIOT/cpu/atmega1281/include/cpu_conf.h
Please use a more specific name by including a (larger) part of the path!
```
This looks IMO strange since apparently only one matching input file was found, so no problem should exist. Of course, the simple solution could be to specify explicitly things like
```
@file cpu/atmega1281/include/cpu_conf.h
```
for each CPU and board, but I thought doxygen should be able to resolve this itself.

gebart, 2017-01-16:
This bothered me before as well. I even did try to dig a little in the Doxygen source code, since it seems like a bug in Doxygen, but it's a mess and I didn't have time to dig deeper.

Did you try specifying the name on the line? IIRC another warning showed up instead.

OlegHahm, 2017-01-16:
Indeed:
```/home/oleg/git/RIOT/cpu/atmega1281/include/cpu_conf.h:12: warning: the name `cpu/atmega1281/include/cpu_conf.h' supplied as the second argument in the \file statement is not an input file
```
if I try to use `cpu/atmega1281/include/cpu_conf.h`.

miri64, 2017-01-17:
The problem might be, that doxygen's path is `doc/doxygen` (at least I suspect it to be) or it just works on absolute paths.

OlegHahm, 2017-01-17:
I tried with relative paths (`../../cpu.../`) but it didn't work. Didn't want to try with absolute paths since this wouldn't make sense.

gebart, 2017-01-17:
Could you please try it with absolute paths as well? I think the original
warning will come back then.

OlegHahm, 2017-01-17:
True.

A-Paul, 2017-01-17:
The warnings disappear when in *riot.doxyfile* the line :
> STRIP_FROM_INC_PATH    = $(STRIP_FROM_INC_PATH_LIST) # Exported from Makefile.

is reset to default.

> STRIP_FROM_INC_PATH    =

The change was introduced by https://github.com/RIOT-OS/RIOT/pull/4890.

OlegHahm, 2017-01-17:
Ah, thanks! Great finding. Now we have to think how we can solve this issue without re-introducing #4794.

miri64, 2017-01-17:
Maybe set `STRIP_FROM_INC_PATH_LIST` just to `$RIOTBASE` (and rename it to something more non-list-related then of course)? This would keep cpu- and board-related headers from colliding

miri64, 2017-01-17:
Nope, that keeps the include path's confusing :-/

miri64, 2017-01-17:
Another alternative would be to have the doccheck script just not report

```
${file}:\d+: warning: the name `${file}' supplied as the second argument in the \file statement matches the following input files:
   ${file}
```

as an error.

OlegHahm, 2017-01-17:
As long as we make sure that we never supply a second argument to `@file`, this could work.

miri64, 2017-01-17:
So `git grep "@file.\+$" && echo "Evil begun!"`, too?

A-Paul, 2017-01-17:
I think the exclusion is the fastest solution, for now. It should be commented as a workaround with reference to the PR.

miri64, 2017-01-17:
Yes, also quite visible so people don't wonder about the wall of text when they compile their doc locally.

A-Paul, 2017-01-17:
The doxygen documentation says:
> Important:
    The documentation of global functions, variables, typedefs, and enums will only be included in the output if the file they are in is documented as well.

Is this the reason '\file' tags without parameters are used anyway?

gebart, 2017-01-17:
Based on the findings of @A-Paul, I opened #6395 which avoids adding the cpu and board include directories to the include path strip list. The warnings show up because of multiple files in the include path with the same name (mostly cpu_conf.h and periph_conf.h).

PR #6395 (2017-01-20): doc: Doxygen strip include fix
Fixes #6367 for all occurrences except w5100_params.h

Piggybacked a tiny enhancement for the clean Makefile target in doc as well.
closed 2017-01-20 by OlegHahm