RIOT-OS/RIOT #6379: nrf52dk/nordic_soft_device: not working anymore?

PeterKietzmann, 2017-01-16:
Following [this guide](https://devzone.nordicsemi.com/blogs/981/setting-up-ipv6-over-ble-using-nrf52-series-and-ri/) I can not use the nrf52dk with Nordic soft device BLE support any more. A program including the soft device module won't even allow interaction with a shell. @kaspar030 can you confirm?

PeterKietzmann, 2017-01-31:
@kaspar030 ping can you give it a quick try?

kYc0o, 2017-04-25:
I tried and it actually doesn't work anymore, it works for release 2016.10 but it stopped to work on 2017.01.

What I suggest is just to remove it from the boards with radio capabilities and point to the exact branch/commit on which the softdevice used to work.

Of course, the best thing to do is to fix it, but I have no idea how.

triple7, 2017-06-06:
Hi, has anyone been able to fix this? We
re currently stuck with trying to get it going, however we don't want to mess with the dependencies all over the board.

kaspar030, 2017-06-06:
I'll take a look at it later today!

triple7, 2017-06-18:
Any progress on this?

aabadie, 2017-07-02:
ping @kaspar030, would be great to fix this for the release.

R1oT01, 2017-08-16:
@kaspar030 PING. Any ideas yet?

R1oT01, 2017-08-23:
Ok, I tested some things, but still no output in terminal.
I flashed the SoftDevice.hex and microcoap-server manually with JLinkExe and nrfjprog command line tool, but still didn't work. I try to debug, but couldn't connect with the Segger JLink Debugger Device to the board. I tried to search the code for inconsistancies, but was overwhelmed by the amount of code to look into. Still have no clue where to search for the problem. Besides that, I'm a little frustrated about the fact, that i got no reaction yet. I'm a rookie in programming, especially with RIOT and Nordic devices. I would appreciate a little help.

PR #8068 (2018-02-06): pkg/nordic_softdevice_ble: fix broken package
This PR intends to fix #6379 

Interaction with shell is now working when the `nordic_softdevice_ble` pkg is enabled
This PR is WIP & RFC because this is a very ugly revert of f656a31 (#6638) when the `nordic_softdevice_ble` package is used.

What must be done before merging : 

- ~~Find a proper fix for VTOR when softdevice is enabled (not this ugly hack)~~ EDIT: Done
- ~~Fix the reset (when I push the reset button, the shell is not restart. How it is even possible ?)~~
- ~~Remove export FLASH_ADDR := 0x1f000 within boards/makefile.include and use [... cut]
closed 2018-02-06 by kaspar030