RIOT-OS/RIOT #6405: Some drivers with saul capabilities are not tested automatically by the saul test

aabadie, 2017-01-18:
This is the case for:
* si70xx
* bmp180
* jc42
* and soon bme280 (after #6156 is merged)

I open this issue to discuss a solution that could be applied to the CI scripts : [dist/tools/compile_test/compile_test.py](https://github.com/RIOT-OS/RIOT/blob/master/dist/tools/compile_test/compile_test.py)) or the [JenkinsFile](https://github.com/RIOT-OS/RIOT/blob/master/Jenkinsfile).

@PeterKietzmann suggested a simple solution in this [comment](https://github.com/RIOT-OS/RIOT/pull/6156#issuecomment-272965500), e.g call building `tests/saul` for each sensor drivers having a saul implementation, to verify it builds. For example:
```bash
drivers="bmp180 jc42 si70xx"
for driver in $drivers; do USEMODULE=$driver make -C tests/saul clean all; done
```

If you feel it's worth, I propose myself for a PR.

aabadie, 2017-06-27:
@smlng, some of your PRs yesterday made me remember this issue. In `made buidtests`, the idea would be to check if the application to be built is `examples/saul` and loop on all drivers in `sys/auto_init/saul`. Then the SAUL build of drivers will be part of the CI.

What do you think of the idea ?

I'd like to look at it for this release.

smlng, 2017-06-27:
@aabadie had the same idea yesterday, when working on those SAUL refinements - and basically I did as you suggested: take `examples/saul` and compile every SAUL capable driver like:

```
USEMODULE=<driver-module> BOARD=samr21-xpro make -C examples/saul clean all
```

The question is how to proceed and make it future prove? Integrate with `make buildtests` wouldn't be _my_ favorite, nevertheless, better than nothing.

kaspar030, 2017-06-27:
> USEMODULE=<driver-module> BOARD=samr21-xpro make -C examples/saul clean all

There are test applications for most drivers. How about just including saul in them?

While the CI could do "for driver in drivers; do USEMODULE=driver make -C examples/saul clean all", it would completely break both ccache and the git-caching scheme used to avoid a new checkout for each build.

smlng, 2017-06-27:
> There are test applications for most drivers. How about just including saul in them?

Though that might be the easiest way, I don't think its a good idea. Because most driver tests are functional tests, and would then break when somethings wrong with SAUL - or the other way around.

Further, I'm huge _fan_ of keeping tests small and simple - so they (hopefully) fit and execute on even the smallest platforms, like Arduino Uno (remember: just 2K of RAM).

aabadie, 2017-06-27:
Maybe we could add some logic for getting saul capable drivers in compile_test.py ?

This won't solve the ccache issue if I understand correctly.

smlng, 2017-06-27:
For #7241 (by @kaspar030 ) to work, each drivers has to state its required features. Could we extend (or reverse) that logic so that drivers also can state which features they provide, i.e., SAUL for instance? 

And if that's possible, could we exploit that for the tests and/or CI building process to check SAUL if a driver implements SAUL interface?

aabadie, 2017-06-27:
In python, this can be done in one line (PEP8 inside):
```python
drivers = [os.path.splitext(i)[0].split('_')[-1] for i in os.listdir('sys/auto_init/saul') if i != 'Makefile' and i not in  ('auto_init_adc.c', 'auto_init_gpio.c')]
```

kaspar030, 2017-06-27:
How about a "saul_drivers" test application, building *all* saul enabled drivers? (This is about compiling, right?)

aabadie, 2017-06-27:
> How about a "saul_drivers" test application, building all saul enabled drivers?

The problem will be with low memory boards.

smlng, 2017-06-27:
Yes, a compile test would be enough for the beginning. Still would be nice to have an automatic way to get _all_ SAUL enabled drivers, and not adapt the test for every new SAUL adaption.

miri64, 2017-06-27:
> The problem will be with low memory boards.

They still compile when they are in `BOARD_INSUFFICIENT_MEMORY`, just the linking is skipped.

aabadie, 2017-06-27:
Ok, then let's go for the saul_drivers application. I'll try to push a PR later today.

aabadie, 2019-09-10:
This issue is still valid. Maybe the underlying idea of #11676 could help a bit here.

closed 2020-11-16 by stale[bot]