RIOT-OS/RIOT #6410: xtimer_core.c DEBUG does not compile (at least) for AVR platforms

kYc0o, 2017-01-18:
If one tries to enable debug on `sys/xtimer/xtimer_core.c` several errors will appear:

```console
In file included from /Users/facosta/git/RIOT-OS/RIOT/sys/xtimer/xtimer_core.c:29:0:
/Users/facosta/git/RIOT-OS/RIOT/sys/xtimer/xtimer_core.c: In function '_xtimer_set':
/Users/facosta/git/RIOT-OS/RIOT/sys/xtimer/xtimer_core.c:131:11: error: format '%lu' expects argument of type 'long unsigned int', but argument 3 has type xtimer_ticks32_t {aka struct <anonymous>}' [-Werror=format=]
     DEBUG("timer_set(): offset=%" PRIu32 " now=%" PRIu32 " (%" PRIu32 ")\n", offset, xtimer_now(), _xtimer_lltimer_now());
           ^
/Users/facosta/git/RIOT-OS/RIOT/core/include/debug.h:63:33: note: in definition of macro 'DEBUG_PRINT'
 #define DEBUG_PRINT(...) printf(__VA_ARGS__)
                                 ^~~~~~~~~~~
/Users/facosta/git/RIOT-OS/RIOT/sys/xtimer/xtimer_core.c:131:5: note: in expansion of macro 'DEBUG'
     DEBUG("timer_set(): offset=%" PRIu32 " now=%" PRIu32 " (%" PRIu32 ")\n", offset, xtimer_now(), _xtimer_lltimer_now());
     ^~~~~
/Users/facosta/git/RIOT-OS/RIOT/sys/xtimer/xtimer_core.c: In function '_timer_callback':
/Users/facosta/git/RIOT-OS/RIOT/sys/xtimer/xtimer_core.c:439:34: error: incompatible type for argument 1 of '_xtimer_lltimer_mask'
             _xtimer_lltimer_mask(xtimer_now()), _xtimer_lltimer_mask(0xffffffff - xtimer_now()));
                                  ^
/Users/facosta/git/RIOT-OS/RIOT/core/include/debug.h:63:33: note: in definition of macro 'DEBUG_PRINT'
 #define DEBUG_PRINT(...) printf(__VA_ARGS__)
                                 ^~~~~~~~~~~
/Users/facosta/git/RIOT-OS/RIOT/sys/xtimer/xtimer_core.c:438:5: note: in expansion of macro 'DEBUG'
     DEBUG("_timer_callback() now=%" PRIu32 " (%" PRIu32 ")pleft=%" PRIu32 "\n", xtimer_now(),
     ^~~~~
In file included from /Users/facosta/git/RIOT-OS/RIOT/sys/include/xtimer.h:585:0,
                 from /Users/facosta/git/RIOT-OS/RIOT/sys/xtimer/xtimer_core.c:24:
/Users/facosta/git/RIOT-OS/RIOT/sys/include/xtimer/implementation.h:53:24: note: expected 'uint32_t {aka long unsigned int}' but argument is of type 'xtimer_ticks32_t {aka struct <anonymous>}'
 static inline uint32_t _xtimer_lltimer_mask(uint32_t val)
                        ^~~~~~~~~~~~~~~~~~~~
In file included from /Users/facosta/git/RIOT-OS/RIOT/sys/xtimer/xtimer_core.c:29:0:
/Users/facosta/git/RIOT-OS/RIOT/sys/xtimer/xtimer_core.c:439:81: error: invalid operands to binary - (have 'long unsigned int' and 'xtimer_ticks32_t {aka struct <anonymous>}')
             _xtimer_lltimer_mask(xtimer_now()), _xtimer_lltimer_mask(0xffffffff - xtimer_now()));
                                                                                 ^ ~~~~~~~~~~~~
/Users/facosta/git/RIOT-OS/RIOT/core/include/debug.h:63:33: note: in definition of macro 'DEBUG_PRINT'
 #define DEBUG_PRINT(...) printf(__VA_ARGS__)
                                 ^~~~~~~~~~~
/Users/facosta/git/RIOT-OS/RIOT/sys/xtimer/xtimer_core.c:438:5: note: in expansion of macro 'DEBUG'
     DEBUG("_timer_callback() now=%" PRIu32 " (%" PRIu32 ")pleft=%" PRIu32 "\n", xtimer_now(),
     ^~~~~
```

My compiler:

```console
avr-gcc (GCC) 6.2.0
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

OlegHahm, 2017-01-18:
I couldn't reproduce, but can you check if #6411 fixes the issue for you?

kYc0o, 2017-01-18:
Yes it does.

closed 2017-01-18 by PeterKietzmann

kYc0o, 2017-01-18:
#6414 does actually.

PR #6411 (2017-01-18): xtimer: fix formatter for xtimer_now in DEBUG
Fixes #6410.