RIOT-OS/RIOT #6426: Jenkins: Show status of each individual build group as red/green

gebart, 2017-01-19:
I don't know Jenkins very well, so excuse me if this already exists somewhere.
I would like to have a status page where we can see the status of each build group in the CI build, something like what Travis used to show when we used the build matrix feature to select settings.
I have seen this page: 
https://riot-ci.inet.haw-hamburg.de/job/RIOT/view/change-requests/job/PR-3263/2/flowGraphTable/
but it only shows success on everything until the last step where it shows a red ball "Error signal - (1 ms in self)"

miri64, 2017-01-19:
In general some parsed output of the log files would be nice. Oftentimes they are not really helpful in finding the error unless you use `Ctrl+F`

miri64, 2017-01-19:
(To my knowledge there are plugins to parse GCC's output, so this shouldn't be that much of a deal [baring the usual config hell of Jenkins of course])

smlng, 2017-07-06:
Jenkins is no more 😢 hence, closing

closed 2017-07-06 by smlng