RIOT-OS/RIOT #6427: xtimer: infinite loop with xtimer_mutex_lock_timeout

vincent-d, 2017-01-19:
Hi,

I tried to use xtimer_mutex_lock_timeout but if I try to lock the mutex again when the timeout has expired, the mutex_lock is stuck in an infinite loop.

By changing the `_mutex_timeout` function as follow, it solves the issue:
```C
static void _mutex_timeout(void *arg)
{
    mutex_thread_t *mt = (mutex_thread_t *)arg;

    mt->timeout = 1;
    list_remove(&mt->mutex->queue, (list_node_t *)&mt->thread->rq_entry);
    if (mt->mutex->queue.next == NULL) {
        mt->mutex->queue.next = MUTEX_LOCKED;
    }
    sched_set_status(mt->thread, STATUS_PENDING);
    thread_yield_higher();
}
```

kaspar030, 2017-01-19:
@vincent-d Would you mind opening a PR with your change? Like this it is hard to see what you have changed...

lebrush, 2017-02-24:
Solved by #6428

closed 2017-02-24 by lebrush