RIOT-OS/RIOT #6431: Segmentation Fault running Shell

antmerlino, 2017-01-19:
Hi all,

I'm working through the tutorials trying to checkout some of the features of RIOT OS.  In task-02 you are asked to add a shell command.  Here is my code:

```
#include <stdio.h>
#include <string.h>

#include "shell.h"

int echo(int argc, char **argv)
{
    (void)argc;
    (void)argv;
    return 0;
}

static const shell_command_t commands[] = {
    {"echo", "Write", echo}
};

int main(void)
{
    puts("This is Task-02");

    char line_buf[SHELL_DEFAULT_BUFSIZE];
    shell_run(commands, line_buf, SHELL_DEFAULT_BUFSIZE);

    return 0;
}
```

But, if I type in anything in the terminal after adding this command, I get:

```
main(): This is RIOT! (Version: 2016.10-devel-445-g3d237-880bcc294d12-HEAD)
This is Task-02
> help
help
/data/riotbuild/task-02/../RIOT/Makefile.include:336: recipe for target 'term' failed
make: *** [term] Segmentation fault
make: Leaving directory '/data/riotbuild/task-02'
```

Not sure if I am doing something fundamentally wrong, or there is an issue here.

OlegHahm, 2017-01-19:
Which toolchain/compiler version are you using?

antmerlino, 2017-01-19:
I've tried with both vagrant and docker with the same result.  So I'm going to assume GCC, which according to the docker container

gcc (Ubuntu 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609

I'm going to guess Vagrant is on a similar version

OlegHahm, 2017-01-19:
Can you post your Makefile as well, please?

antmerlino, 2017-01-19:
Sure, though it is the stock one from the tutorial.

```
# name of your application
APPLICATION = Task02

# If no BOARD is found in the environment, use this default:
BOARD ?= native

# This has to be the absolute path to the RIOT base directory:
RIOTBASE ?= $(CURDIR)/../RIOT

# Comment this out to disable code in RIOT that does safety checking
# which is not needed in a production environment but helps in the
# development process:
CFLAGS += -g -DDEVELHELP

# Change this to 0 show compiler invocation lines by default:
QUIET ?= 1

# Modules to include:
USEMODULE += shell
USEMODULE += shell_commands
USEMODULE += ps

include $(RIOTBASE)/Makefile.include
```

OlegHahm, 2017-01-19:
Ah, sorry, for the noise - I should have checked if I can reproduce your problem first - which in fact I can.

antmerlino, 2017-01-19:
All good, I did notice I made one small change to the Makefile which was to add the -g option to CFLAGS.  On a side note, what is the intended way of getting make to generate symbols using Riot's build system.  Is there a different target?  I tried all-debug but that didn't seem to do anything.

Sorry for the side question.

OlegHahm, 2017-01-19:
```C
static const shell_command_t commands[] = {
    {"echo", "Write", echo}
};
```
The list of  `shell_command_t` must be terminated by an empty entry. It should be
```C
static const shell_command_t commands[] = {
    {"echo", "Write", echo},
    {NULL, NULL, NULL}
};

OlegHahm, 2017-01-19:
> All good, I did notice I made one small change to the Makefile which was to add the -g option to CFLAGS. On a side note, what is the intended way of getting make to generate symbols using Riot's build system. Is there a different target? I tried all-debug but that didn't seem to do anything.

The easiest way would be indeed to do `make clean all-debug debug`.

antmerlino, 2017-01-19:
Ahhh! Okay, maybe we should add that to the notes in the tutorial task? Or link the shell documentation?  Either way thank you for your very quick response!

OlegHahm, 2017-01-19:
> Ahhh! Okay, maybe we should add that to the notes in the tutorial task? Or link the shell documentation? 

Good idea.

> Either way thank you for your very quick response!

Okay to close the issue?

antmerlino, 2017-01-19:
I wouldn't mind contributing to the documentation when I come across things like this.  What are your contribution guidelines or procedures for updating documentation?

Good to close the issue.

OlegHahm, 2017-01-19:
> I wouldn't mind contributing to the documentation when I come across things like this. What are your contribution guidelines or procedures for updating documentation?

Oops, then I was too fast. ;-) In general, opening a pull request should work in any RIOT related repository.

closed 2017-01-19 by OlegHahm

antmerlino, 2017-01-19:
Sounds good, thanks again!