RIOT-OS/RIOT #6436: unittests: tests-tweetnacl crashes on ARM

gebart, 2017-01-20:
Tested: 
 - Works on native
 - Reboots after tweetnacl tests finish on mulle
 - Works on mulle when I increase `THREAD_STACKSIZE_DEFAULT` to 2048 (up from 1024)

No error messages printed or any status of the tests, just silent reboot.
Likely a stack overflow.

kYc0o, 2017-01-20:
The [README.md](https://github.com/RIOT-OS/RIOT/blob/master/pkg/tweetnacl/README.md) states it, you need at least 2K more stack in constrained nodes.

Should I change it to explicitly say that it will work out of the box in native but this change is mandatory on small nodes?

gebart, 2017-01-20:
We need the unit tests to run to completion with the default build on all platforms, so we need to either skip the test or change the settings for the thread stacks on unit tests.

miri64, 2017-01-20:
>  or change the settings for the thread stacks on unit tests.

Should be doable by setting the stack-size via the CFLAGS in the test-suite's `Makefile.include`.

gebart, 2017-01-20:
On the other hand, changing the default stack size in the unit tests might
hide stack overflow errors in other modules.

kYc0o, 2017-01-20:
>Should be doable by setting the stack-size via the CFLAGS in the test-suite's Makefile.include.
+1 👍 

I can PR the change.

miri64, 2017-01-20:
> On the other hand, changing the default stack size in the unit tests might
hide stack overflow errors in other modules.

That's why I proposed to only change it in the test suite. By testing the other modules separately you can still detect those (or by deleting the tweetnacl test-suite and building the rest).

kaspar030, 2017-01-20:
IMHO the compile of the default unittests should also be runnable. By setting it only in the suite's makefile, the unittest.elf containing all tests would fail.

We could create a thread with bigger stacksize for tests-tweetnacl and run the tests in there?

miri64, 2017-01-20:
> IMHO the compile of the default unittests should also be runnable. By setting it only in the suite's makefile, the unittest.elf containing all tests would fail.

Huh? The unittests are compiled in a way that all variables from all suites are pulled in during compilation.

> We could create a thread with bigger stacksize for tests-tweetnacl and run the tests in there?

That's also an alternative, but embUnit and threading had proven to be tricky at best in the past.

miri64, 2017-01-20:
(remember the dots per test is just *one* way the unittests can output, they also have an XML format and a more verbose output than the dotted tests)

kYc0o, 2017-01-20:
Is that unittest.elf flasheable to constrained devices?

miri64, 2017-01-20:
Look at the number of boards in `BOARD_INSUFFICIENT_MEMORY` of `tests/unittests` and tell me yourself ;-)

kYc0o, 2017-01-20:
Well, mulle is not there, only for that it's worth to find a solution.

gebart, 2017-01-20:
I usually run the complete unittests build on mulle, but even if you're using one of the boards listed under `BOARD_INSUFFICIENT_MEMORY` you can still run the tests if you build a subset using `make tests-core` for example.

miri64, 2017-01-20:
> I usually run the complete unittests build on mulle, but even if you're using one of the boards listed under BOARD_INSUFFICIENT_MEMORY you can still run the tests if you build a subset using make tests-core for example.

I know. I wrote this feature ;-)

miri64, 2017-01-20:
For the record I meant to provide a fix in the style of

```diff
diff --git a/tests/unittests/tests-tweetnacl/Makefile.include b/tests/unittests/tests-tweetnacl/Makefile.include
index 0cfb96f..9c8efe2 100644
--- a/tests/unittests/tests-tweetnacl/Makefile.include
+++ b/tests/unittests/tests-tweetnacl/Makefile.include
@@ -1,2 +1,4 @@
 USEMODULE += random
 USEPKG += tweetnacl
+
+CFLAGS += -DTHREAD_STACKSIZE_MAIN=\(2*THREAD_STACKSIZE_DEFAULT+THREAD_EXTRA_STACKSIZE_PRINTF\)
```

gebart, 2017-01-20:
It was primarily meant as a response to the question whether you can run
the unit tests on constrained devices.

kYc0o, 2017-01-26:
See #6485

closed 2017-01-26 by kYc0o