RIOT-OS/RIOT #6437: periph/spi: Leftovers from SPI rework in #4780

PeterKietzmann, 2017-01-20:
This issue acts as a place to collect and track leftovers and TODOs from the SPI rework done in  #4780.

- [ ] Test peripheral driver on lpc11u34
- [ ] Test sensor driver adt7310
- [ ] Test ethernet driver encx24j600
- [ ] Test radio driver cc110x once debugged
- [ ]  Adapt u8g2 package patch
- [ ] Performance analysis/optimization for all SPI drivers
    - [ ] atmega_common
    - [ ] cc2538
    - [ ] kinetis_common
    - [ ] lpc11u34
    - [ ]  lpc2387
    - [ ]  msp430fxyz
    - [ ]  nrf5x_common
    - [ ]  sam21_common
    - [ ]  sam3
    - [ ]  stm32_common

haukepetersen, 2017-01-24:
Nice!

PeterKietzmann, 2017-01-25:
@ks156 this is a gentle reminder to test SPI on the WeIO board. Please report errors here or better, check the box in the issue description on success. You can find a procedure that we used for testing SPI drivers in [this table](https://docs.google.com/spreadsheets/d/1_IrBtKyhkOyKK8NxQIHOJ2fdyExqNqMucxDWPLdBNQI/edit?pli=1#gid=0)

lebrush, 2017-01-26:
Please add to the list that the `pn532` driver needs adaption

PeterKietzmann, 2017-01-26:
Dammit! Totally forgot that one. Thanks for the hint

haukepetersen, 2017-01-26:
@lebrush  no problem, I can do the adaption quickly now, if you are volunteering to test it (as I don't have the device...).

PeterKietzmann, 2017-01-26:
We have one here

haukepetersen, 2017-01-26:
and here is the fix: #6478

haukepetersen, 2017-01-26:
build the test-app with `PN532_MODE=spi BOARD=xxx make ...`

haukepetersen, 2017-01-26:
Actually, the `nvram` driver on the `mulle` was just working fine, I just did not configure the correct CS pin when running the test application...

ks156, 2017-01-26:
@PeterKietzmann I'm sorry but I won't be able to test the SPI now ... I don't have an oscilloscope under the hand anymore, and I don't currently have SPI devices for tests...

kaspar030, 2017-01-30:
Tested encx24j600 on nucleo-f334, works like a charm after disabling of onboard led (#6501)

PeterKietzmann, 2017-01-30:
@kaspar030 did you also test it on a stm32f4discovery board?

PeterKietzmann, 2017-01-31:
@haukepetersen when I run SPI (with `POL:0, PHASE:0`) it appears that CLK is on high level during idle. Just when the CS line pulls low, the SCK line goes low as well and when CS goes high, the CLK line goes high as well. That seems a bit strange to me. Did you observe the same during your tests?

PeterKietzmann, 2017-01-31:
BTW: Same applies to `POL:0, PHASE:1`. With `POL:1, PHASE:*` we also come from high level and result in high level before and after the transmission -as expected.

PeterKietzmann, 2017-02-08:
I unset the box for lis3dh because of #6567

immesys, 2017-02-24:
The MISO pin is configured incorrectly on samr21 again. Same as  #5868. Please make sure that MISO is always configured as IN_PD or IN_PU, never just IN.

BytesGalore, 2017-02-28:
I appended a problem with HW CS on the Kinetis board to the OT list
@ 1MHz the HW CS rises back after transmission, while it doesn't @100kHz and 400 kHz

gebart, 2017-02-28:
The kinetis problem could be that the module is disabled before the clock CS line has risen. Try removing the clock disable line in spi_release, see if that fixes it.

immesys, 2017-02-28:
Can I ask why we want to disable module and clock on release? Is it just to save energy or is there a different reason? I ask because I don't think it's a good idea but if I want to convince people to revert it I will need to prove it doesn't serve its intended purpose.

BytesGalore, 2017-02-28:
@gebart I commented out the complete line for powering off [1], but no luck.

[1] https://github.com/RIOT-OS/RIOT/blob/master/cpu/kinetis_common/periph/spi.c#L155

gebart, 2017-03-01:
@BytesGalore Sorry, I meant https://github.com/RIOT-OS/RIOT/blob/master/cpu/kinetis_common/periph/spi.c#L71. Leave the MDIS stuff in. If it doesn't make a difference, try removing both the SIM_SCGC line and the SPI_MCR MDIS line.

BytesGalore, 2017-03-01:
@gebart preventing clearing the SIM_SCGC when powering off (@ line [71](https://github.com/RIOT-OS/RIOT/blob/master/cpu/kinetis_common/periph/spi.c#L71) ) did the trick.
I tested it with 100kHz, 400kHz and 1MHz (my logic analyzer doesn't sample correctly higher frequencies).

gebart, 2017-03-01:
I think turning off the clock domain for the SPI is a wrong move on Kinetis. It doesn't affect the power consumption as far as I can measure, and it have caused issues on multiple occasions.

immesys, 2017-03-01:
It doesn't affect power consumption positively on SAMR21 either...

dylad, 2017-03-21:
SPI doesn't work on SAML21 anymore.
see #6686 for the issue reference.
I also come up with the required fix #6687

PeterKietzmann, 2018-06-27:
@gebart I'd like to get rid of this old-timer. The only uncovered points are:
- Test sensor driver adt7310
- Problems with hardware CS on Kinetis (@ 100kHz and 400kHz)

Do you have a chance to test the adt7310? I would guess that after 1.5 years a SPI defect in this driver would have already popped up. For the latter issue I propose to cover it within #6567.

gebart, 2018-06-27:
My adt7310 breakout is in a box somewhere but I will not be able to get to it until August.

Regarding the Kinetis SPI module disable issue: #9438

PeterKietzmann, 2018-08-21:
Yes, thank you very much!

closed 2018-08-21 by PeterKietzmann