RIOT-OS/RIOT #6439: Jenkins: Shows unrelated commits under "Changes" heading

gebart, 2017-01-20:
Change list shows commits which have been added to master after the PR was branched. It should only show the commits on the PR branch, not the master changes.

Example: https://riot-ci.inet.haw-hamburg.de/job/RIOT/view/change-requests/job/PR-5688/8/

[image]

smlng, 2017-02-03:
it shows changes between consecutive builds of that specific PR, **not** between PR and master. Mostly after a rebase you see lots of changes ... so its (only) a bit counter-intuitive what _changes_ means (or what Jenkins refers to) here

smlng, 2017-07-06:
Jenkins is no more 😢 so closing

closed 2017-07-06 by smlng