RIOT-OS/RIOT #6447: pyterm AttributeError: SerCmd instance has no attribute 'fmt_str'

nikosft, 2017-01-20:
With python 2.7.12 on ubuntu 16.04 I get the following error when I run RIOT/dist/tools/pyterm/pyterm

```
Traceback (most recent call last):
  File "[...]/pyterm", line 728, in <module>
    args.log_dir_name, args.newline)
  File "[...]/pyterm", line 128, in __init__
    if self.fmt_str == None:
AttributeError: SerCmd instance has no attribute 'fmt_str'
make: *** [term] Error 1
```
if I put in comments lines 129-132 (the if statement) and if I add 
```
self.fmt_str == None
```
After line 109, it works as expected

OlegHahm, 2017-01-20:
Are you on master?

OlegHahm, 2017-01-20:
The problem should have been fixed by #6309.

nikosft, 2017-01-20:
You are right, I was using a previous version of master (dcfd0d656baecdfb786781daeb06bd7bfc837681) I updated and it works fine

closed 2017-01-20 by nikosft