RIOT-OS/RIOT #6451: Gain max. SRF08?

radioelf, 2017-01-23:
Hi.
Is this correct? RIOT/drivers/include/srf08.h line 65 "#define SRF08_MAX_GAIN 0x25"
In the documentation https://www.robot-electronics.co.uk/htm/srf08tech.html it indicated 0x35 (1025)

PeterKietzmann, 2017-01-23:
Where did you see `0x35` on the provided site? What I found there was
`31 	0x1F 	Set Maximum Analogue Gain to 1025` which still differs from your description. When writing this file I used [this document](http://coecsl.ece.illinois.edu/ge423/DevantechSRF08UltraSonicRanger.pdf) as reference which states `0x25`. However, looking at the "Analogue Gain" table I think the mapping **might** be wrong. Is there anyone with more experience about that device?

radioelf, 2017-01-23:
HI, It was a mistake, its 0x31....

PeterKietzmann, 2017-01-24:
No. In the link you sent it is `0d31` equals `0x1f`. The link I sent did not translate the decimal to the hex values correctly. Thus I assume your documentation is correct. Do you want to provide a fix for the header file :-)?

radioelf, 2017-01-24:
No, better ensure the correct documentation

PeterKietzmann, 2017-01-26:
@radioelf do you have new insights which data sheet is the correct one? Otherwise I assume the issue still persists, right?

radioelf, 2017-01-26:
Hi Peter, Safely NO, yes the issue still persists.

kYc0o, 2017-11-27:
#7956 was merged, can we consider this as solved?

PeterKietzmann, 2017-11-27:
Whoops, thought I already closed this issue. Yes, I see it as solved now.

closed 2017-11-27 by PeterKietzmann