RIOT-OS/RIOT #6456: Windows + eclipse + RIOT  Hello_world example build error

juttTechnology, 2017-01-23:
i am using windows 10 + eclipse + RIOT hello world example.
toolchain is cygwin 
Hardware is nrf52dk
here is build info 

make BOARD=nrf52dk info-build all 
Warning: no PORT set!
Datei *-common nicht gefunden
process_begin: CreateProcess(NULL, command -v arm-none-eabi-gcc, ...) failed.
APPLICATION: hello-world

supported boards:

BOARD:   nrf52dk
CPU:     nrf52
MCU:     nrf52

RIOTBASE:  C:/IoT/RIOT
RIOTBOARD: C:/IoT/RIOT/boards
RIOTCPU:   C:/IoT/RIOT/cpu
RIOTPKG:   C:/IoT/RIOT/pkg

DEFAULT_MODULE: auto_init board core core_msg cpu sys
DISABLE_MODULE: 
USEMODULE:      cortexm_common isrpipe newlib newlib_nano newlib_syscalls_default nrf5x_common_periph periph periph_common tsrb uart_stdio

ELFFILE: C:/IoT/RIOT/examples/hello-world/bin/nrf52dk/hello-world.elf
HEXFILE: C:/IoT/RIOT/examples/hello-world/bin/nrf52dk/hello-world.bin

FEATURES_REQUIRED (excl. optional features):
         -none-
FEATURES_OPTIONAL (strictly "nice to have"):
         -none-
FEATURES_PROVIDED (by the board or USEMODULE'd drivers):
         cpp periph_cpuid periph_flashpage periph_gpio periph_hwrng periph_rtt periph_timer periph_uart radio_nrfmin
FEATURES_MISSING (incl. optional features):
         -none-
FEATURES_MISSING (only non-optional features):
         -none-

FEATURES_CONFLICT:     
FEATURES_CONFLICT_MSG: 

CC:      arm-none-eabi-gcc
CFLAGS: 
	-Werror  
	-mcpu=cortex-m4  
	-mlittle-endian  
	-mthumb  
	-mfloat-abi=soft  
	-mno-thumb-interwork  
	-ffunction-sections  
	-fdata-sections  
	-fno-builtin  
	-fshort-enums  
	-ggdb  
	-g3  
	-Os  
	-std=c99  
	-fno-delete-null-pointer-checks  
	-Wstrict-prototypes  
	-Werror=strict-prototypes  
	-Wold-style-definition  
	-Werror=old-style-definition  
	-fno-common  
	-Wall  
	-include  
	C:/IoT/RIOT/examples/hello-world/bin/nrf52dk/riotbuild/riotbuild.h

CXX:     arm-none-eabi-g++
CXXUWFLAGS: 
	-Wstrict-prototypes  
	-Wold-style-definition  
	-std=%
CXXEXFLAGS:

LINK:    arm-none-eabi-gcc
LINKFLAGS: 
	-LC:/IoT/RIOT/cpu/nrf52/ldscripts  
	-LC:/IoT/RIOT/cpu/cortexm_common/ldscripts  
	-Tnrf52xxaa.ld  
	-Wl,--fatal-warnings  
	-mcpu=cortex-m4  
	-mlittle-endian  
	-mthumb  
	-mfloat-abi=soft  
	-mno-thumb-interwork  
	-ggdb  
	-g3  
	-Os  
	-static  
	-lgcc  
	-nostartfiles  
	-Wl,--gc-sections  
	-specs=nano.specs  
	-lc  
	-lnosys

OBJCOPY: /cygdrive/c/cygwin64/lib/ARMEmbeddedToolchain/bin/arm-none-eabi-objcopy
OFLAGS:  -O binary

FLASHER: C:/IoT/RIOT/dist/tools/jlink/jlink.sh
FFLAGS:  flash

TERMPROG:  C:/IoT/RIOT/dist/tools/pyterm/pyterm
TERMFLAGS: -p "" -b "115200"
PORT:      

DEBUGGER:       C:/IoT/RIOT/dist/tools/jlink/jlink.sh
DEBUGGER_FLAGS: debug

DOWNLOAD_TO_FILE:   /usr/bin/wget -nv -c -O
DOWNLOAD_TO_STDOUT: /usr/bin/curl -s
UNZIP_HERE:         /usr/bin/unzip -q

DEBUGSERVER:       C:/IoT/RIOT/dist/tools/jlink/jlink.sh
DEBUGSERVER_FLAGS: debug-server

RESET:       C:/IoT/RIOT/dist/tools/jlink/jlink.sh
RESET_FLAGS: reset

MAKEFILE_LIST: 
	C:/IoT/RIOT/examples/hello-world/Makefile  
	C:/IoT/RIOT/Makefile.include  
	C:/IoT/RIOT/Makefile.docker  
	C:/IoT/RIOT/Makefile.scan-build  
	C:/IoT/RIOT/Makefile.pseudomodules  
	C:/IoT/RIOT/Makefile.defaultmodules  
	C:/IoT/RIOT/boards/nrf52dk/Makefile.include  
	C:/IoT/RIOT/boards/Makefile.include.jlink  
	C:/IoT/RIOT/boards/Makefile.include.serial  
	C:/IoT/RIOT/cpu/nrf52/Makefile.include  
	C:/IoT/RIOT/cpu/nrf5x_common/Makefile.include  
	C:/IoT/RIOT/cpu/Makefile.include.cortexm_common  
	C:/IoT/RIOT/cpu/cortexm_common/Makefile.include  
	C:/IoT/RIOT/cpu/Makefile.include.gnu  
	C:/IoT/RIOT/Makefile.dep  
	C:/IoT/RIOT/boards/nrf52dk/Makefile.dep  
	C:/IoT/RIOT/drivers/Makefile.dep  
	C:/IoT/RIOT/Makefile.dep  
	C:/IoT/RIOT/boards/nrf52dk/Makefile.dep  
	C:/IoT/RIOT/drivers/Makefile.dep  
	C:/IoT/RIOT/Makefile.cflags  
	C:/IoT/RIOT/sys/Makefile.include  
	C:/IoT/RIOT/sys/newlib/Makefile.include  
	C:/IoT/RIOT/drivers/Makefile.include  
	C:/IoT/RIOT/Makefile.buildtests  
	C:/IoT/RIOT/boards/nrf52dk/Makefile.features  
	C:/IoT/RIOT/Makefile.vars  
	C:/IoT/RIOT/dist/tools/desvirt/Makefile.desvirt  
	C:/IoT/RIOT/Makefile.bindist  
	C:/IoT/RIOT/Makefile.modules
process_begin: CreateProcess(NULL, /bin/echo -e "\033[1;32mBuilding application \"hello-world\" for \"nrf52dk\" with MCU \"nrf52\".\033[0m", ...) failed.
make (e=2): The system can not find the specified file.

make: *** [..build-message] Error 2

Here is the include and symbol Image
[image]
 
thank you

PeterKietzmann, 2017-01-24:
I really can't help you with this but did you consider using a VM? Please have a look at the description [here](https://github.com/RIOT-OS/Tutorials) under section "Quick Setup using a Virtual Machine". There is also an interesting wiki page, if you haven't already seen it [here](https://github.com/RIOT-OS/RIOT/wiki/Build-RIOT-on-Windows-OS). But **might** be outdated. We don't officially support development under windows, so this issue might get closed soon.

juttTechnology, 2017-01-24:
1- with VM everything is working perfectly.
2- now i am testing on windows. Regarding to the above error i found the error line.
from the above build info data. on line number 3(File * -common not found)
this error is coming from the Makefile.buildtests
and line who is generating this error is this one 
BOARDS := $(filter $(if $(BOARD_WHITELIST), $(BOARD_WHITELIST), %), $(BOARDS))

rmadhuraj, 2017-09-22:
@juttTechnology  . I have the thought of Building RIOT on Ecllipse from windows. Are you succeeded with your testing? If yes, i think you will help me to achieve the same.

juttTechnology, 2017-09-22:
not really..

PeterKietzmann, 2017-11-27:
@juttTechnology, @rmadhuraj we do not officially support development under windows, except the virtual box that we provide. This does't mean these efforts are not worth a try, it's even adorable. However, if you actually start to work on this, I'd like you to open a fresh issue where you can tackle down and track your tasks. That's why I'm closing this one now. 
Besides, don't hesitate to yous our mailing list for further questions!

closed 2017-11-27 by PeterKietzmann

miri64, 2017-11-27:
> This does't mean these efforts are not worth a try, it's even adorable.

I think you meant "admirable". "Adorable" has more the connotation of "cute" ;-).

Now that I have a pure Windows machine at home again, I was also thinking of providing something on the side like this. However, I'm also happy if someone else provides the port and I would just give my testing power and RIOT expertise towards that goal ;-).