RIOT-OS/RIOT #6460: stm32f4: xtimer clock running at twice the speed

bergzand, 2017-01-23:
So, this evening I experienced the same problem as with #6287 on my nucleo-f401 board.
Tested with the timer_periodic wakeup example again.
CPU: stm32f401RE
board: Nucleo-f401

expected to receive a message every second. A message is received every 500ms.
RIOT seems to think that a second has elapsed:
```
main(): This is RIOT! (Version: 2017.01-devel-173-gab12d6-dystopia-HEAD)
slept until 1012141
slept until 2012141
slept until 3012141
slept until 4012141
```

Problem seem to be introduced by https://github.com/RIOT-OS/RIOT/pull/6184
Might be related to the fix in #6287 that the stm32f4(01) needs to be added to the solution from #6291

PeterKietzmann, 2017-01-27:
Should be fixed with #6492. @bergzand can give it a quick try and reopen the issue in case it is not solved?

closed 2017-01-27 by PeterKietzmann

bergzand, 2017-01-27:
@haukepetersen, @PeterKietzmann: Issue is solved, thank you.