RIOT-OS/RIOT #6461: Support semihosting for ARM targets without available UART

antmerlino, 2017-01-23:
Similar to the J-Link Segger RTT, semihosting can provide an IO interface for boards without an additional UART.  Semihosting uses SVC instructions to signal semihosting operations to the debug agent. 

To use semihosting, you must link librdimon which implements the semihosting versions of syscalls from newlib.  

Here is currently what I've identified as the steps to get this working

1. Change build setup to allow no definition of newlib syscalls so that they may be linked from librdimon.
I cannot find a way to specify no implementation of syscalls through the makefile setup currently.  Is there a way?  If not, can we add a `newlib_syscalls_none` option or something that doesn't define them at all.  Alternatively, we can just look for a SEMIHOSTING flag and not use any newlib_syscall_XXX module.

2. Need to add linker flags in order to include the librdimon library
```--spec-rdimon.specs -lc -lrdimon```
Not sure the best place to put this.  Again, maybe we use a SEMIHOSTING flag and then in the cortexm_common Makefile we optionally add the flags?

3. In order for semihosting to be initialized properly, initialize_monitor_handles() is supposed to be called in crt0.  I'm thinking use the SEMIHOSTING flag and call it from cortexm_init.c.  (or maybe vectors_cortexm.c)

4. The current debug target using the openocd.sh script launches openocd and gdb in the same terminal.  The problem is gdb is "on top", but the semihosting interface needs to occur in the openocd window.  Not sure if there is an option here to make it easier for the user.  One option is to just use the debug-server option and then tell the user to manually connect via another terminal.  Otherwise, maybe we could use something like screen or tmux to give them both through the script.

Feedback and/or advice is appreciated.  Let me know if you think this feature is something people would want, or if I should just keep this in my own fork.

-Anthony

antmerlino, 2017-01-24:
I've just stumbled upon the PSEUDOMODULE concept.  Seems like that would be a good way of doing this.  

Can a pseudomodule change the LINKFLAGS variable?  I am trying it but it doesn't seem to work the way I expected.

gebart, 2017-01-24:
I briefly looked at semihosting as an alternative a few years ago when we were testing different parts of the software, but I never saw that there were any open source alternatives for handling the PC side of it (OpenOCD). How do you enable it in OpenOCD?

The stdio write functionality on the device side is quite simple to implement IIRC, the basic idea is to put the characters in a register and call `BKPT #0xfd` or something.

antmerlino, 2017-01-24:
There is some information here under the ARM section http://openocd.org/doc/html/Architecture-and-Core-Commands.html

Basically you just need to do 
```arm semihosting enable```

As far as the stdio functionality, it's all done for you via librdimon.  My issues are more concerned with how to get it integrated into a reasonable workflow with Riot now. 

So far I have:
1. Added a pseudomodule called semihosting_stdio 
2. Used the pseudomodule inclusion to remove the newlib_syscalls_default from Makefile.dep
3. Used the pseudomodule inclusion to add the appropriate linker flags in sys/Makefile.include 

I'm faced with two new issues now:

1) The librdimon looks for symbol __end__ from the linker script in order to know where it can begin allocating memory.  Most examples I've found use this:
```
._user_heap_stack :
{
  . = ALIGN(4);
  PROVIDE ( end = . );
  PROVIDE ( _end = . );
  PROVIDE ( __end__ = . ); 
  . = . + _Min_Heap_Size;
  . = . + _Min_Stack_Size;
  . = ALIGN(4);
} >RAM
```
But the Riot linker scripts for cortexm_common are much different. Does anyone know how to integrate this?

2. The newlib init function is looking for a reference to _init.  Previously, I had no understanding of startup files etc. so mind me if I'm wrong.  From my understanding though, Riot uses custom init functions instead of the standard ones.  Is there something that needs to be done by the linker to label Riot's reset_handler_default as the _init method?  Hoping someone knows about this stuff and can show me the right way to handle this.

Thansk for the help!

antmerlino, 2017-01-27:
So, I got this working.  It wasn't necessarily ready to land, but I confirmed it at least could work.  However, I should have done more reading before doing it.  Semihosting is ungodly slow.  It prints something out like every 1 second.  And it keeps halting the processor over and over again.  Some say it depends on the debugger setup, so maybe I just have a really slow setup. But either way, since it constantly halts the processor, its useless for any real time operations.

So unless anyone thinks it's still worthwhile for me to finish it, I'm going to abandon and close this issue.

closed 2017-01-30 by antmerlino