RIOT-OS/RIOT #6462: boards/nucleo-l1: crashes when setting time to RTC

aabadie, 2017-01-24:
I noticed this issue while reviewing #6448.

How to reproduce:
1. Use a nucleo-l1 board, checkout current master
2. build and flash `tests/periph_rtc` application
```
make BOARD=nucleo-l1 flash term
```
3. You get the following error messages:
```
main(): This is RIOT! (Version: 2017.01-devel-770-g119a3-trek-saclay)

RIOT RTC low-level driver test
This test will display 'Alarm' in 10 seconds

Initializing the RTC driver
tSetting clock to 2011-12-13 14:15:15
*** RIOT kernel panic:
HARD FAULT HANDLER

*** rebooting...
```

OlegHahm, 2017-01-25:
Did you check at which line it triggers the hard fault?

haukepetersen, 2017-01-27:
I guess this is also connnected to #6499

aabadie, 2017-01-27:
I confim this is fixed @haukepetersen with #6499. Closing.

closed 2017-01-27 by aabadie