RIOT-OS/RIOT #6466: doc: @brief where @name is supposed to be

miri64, 2017-01-24:
In many places in the doc this structure is used:

```C
/**
 * @brief name of group
 * @{
 */
#define FOOBAR (0)   /**< some doc that sometimes isn't here, but that's okay */
/** @} */
```

The intend is to add FOOBAR to a group named "name of group", but this is not what that results in. Rather is "name of group" parsed as the brief for FOOBAR, while the original `@brief` (if present) is parsed as its `@details`. The correct command would be `@name`.

miri64, 2017-01-24:
See e.g. http://doc.riot-os.org/boards_2nucleo-l1_2include_2periph__conf_8h.html as an example.

OlegHahm, 2017-01-25:
In the example you give: do you know why "Clock system configuration" has a different hierarchy level than "I2C configuration"?

miri64, 2017-01-26:
So? Both are using the `@name` tag, as you can see in [the header](https://github.com/RIOT-OS/RIOT/blob/master/boards/nucleo-l1/include/periph_conf.h). But `DAC_NUMOF` is e.g. part of `Clock system configuration` and the intended group name (DAC configuration) became the macro's description.

OlegHahm, 2017-01-26:
I'm not sure how this answers my question, but it actually raises a new one: shouldn't https://github.com/RIOT-OS/RIOT/blob/master/boards/nucleo-l1/include/periph_conf.h#L50 close the group for `Clock system configuration`?

My initial question was: why does `Clock system configuration` show under `Macros`, while `I2C configuration` shows as a group on its own.

kYc0o, 2018-01-29:
IIRC, @aabadie has already started this job. Can you (@aabadie) complete this issue to show the progress? If you prefer we can close this one and open a new one which tracks the efforts.

miri64, 2019-08-10:
Fixed by #8962

closed 2019-08-10 by miri64