RIOT-OS/RIOT #6468: Backburner PRs

emmanuelsearch, 2017-01-24:
How about creating a "Backburner" label which would be applied before closing a PR that has seen too little (or no) update from the author for a while, e.g. maximum timelapse = release cycle of 3 month?

This way we'd have an improved process to log and close PRs that are not making progress -- some of which are piling up undesirably as the number of open PRs grow...

emmanuelsearch, 2017-01-24:
we could start right away with this release by the way ;)

miri64, 2017-01-24:
This would fit nicely with a kind of GitHub watchdog daemon I was thinking about implementing after our discussions yesterday (my main focus was more to limit the number of PRs assigned to one person / limit the number of PRs a person may open).

miri64, 2017-01-24:
To explain: label set => "start timeout" in that daemon to close as soon as <configurable time x> has passed.

kaspar030, 2017-01-24:
> a kind of GitHub watchdog daemon I was thinking about implementing after our discussions yesterday

If you do, you should take a look at the murdock codebase, as it already includes logic to react on pull request actions, etc.

miri64, 2017-01-24:
> If you do, you should take a look at the murdock codebase, as it already includes logic to react on pull request actions, etc.

I know, and would've. But I don't think that's something a CI server should be bothered with.

PeterKietzmann, 2017-01-24:
Sorry for being so negative but I'm personally not a fan of this kind of last-minute-fixes. Even though this might be an opportunity to improve our PR development in the future, I'd vote to postpone this plan until the Release is out. It might be worth a try but maybe we should give it some more time (even to get the other maintainers opinions)

miri64, 2017-01-24:
Sorry I think I stirred the discussion in the wrong direction there. I don't think that adding a new label is a last-minute-fix. The daemon was just a proposal to automate this behavior (instead of requiring any one of our time to review old PRs for every release) ;-)

cgundogan, 2017-01-24:
Hm, to be honest, I have difficulties to see the advantage (other than for statistical reasons) of marking inactive PRs. Nobody will ever look at them anyways. Do we really want that extra overhead of keeping track of inactive PRs (let it be manual or automatic overhead)?

miri64, 2017-01-24:
> Hm, to be honest, I have difficulties to see the advantage (other than for statistical reasons) of marking inactive PRs. Nobody will ever look at them anyways. Do we really want that extra overhead of keeping track of inactive PRs (let it be manual or automatic overhead)?

I think if we mark inactive PRs we have less overhead (and issue pressure to a contributor to react). If a PR is marked as such and the last entry in this PR is 

> **user** added the Backburner label 3 month ago

we can just close (in a manual) it without rereading old discussions from 3 month ago. With an automatic solution even this isn't required anymore. I don't see where you see more overhead.

kYc0o, 2017-01-24:
What about those PRs that are very nice but the committer is very busy (the case of PRs abandoned by our maintainers team) or simply the PR got a long review time so the committer just doesn't work on the same thing anymore? I think for those cases someone else can take over the PR and revive it for RIOTs sake.

miri64, 2017-01-24:
> What about those PRs that are very nice but the committer is very busy (the case of PRs abandoned by our maintainers team) or simply the PR got a long review time so the committer just doesn't work [...]

The idea is that the number of PRs is so low that our maintainer team doesn't forget about a PR anymore ;-). But in general: it's still the maintainers prerogative to set this label, so if they deem it worthy to wait they don't have to set the label.

emmanuelsearch, 2017-01-24:
keeping P

emmanuelsearch, 2017-01-24:
mishap close ;)

PeterKietzmann, 2017-01-24:
>we could start right away with this release by the way ;)

But starting with a new workflow just some hours before feature freeze might introduce  further disturbance and uncertainty about the progress. As you can see, there is already a discussion and a couple of "strong-minded" maintainers didn't even state their opinions. So I propose to let this issue open for discussion but not apply for this Release.

emmanuelsearch, 2017-01-24:
Fine. Let's discuss that later.

kaspar030, 2017-01-24:
The question would be if it has value to tag a PR/issue when closing it because of prolonged inactivity. That only makes sense if we want to search specifically for those later.

emmanuelsearch, 2017-01-24:
This could be useful to quickly sort out stuff that was not fully exploited by the community (not brought to a mergeable state), from stuff that was rejected (because flawed in some sense), while diminishing systematically the number of open PRs.

emmanuelsearch, 2017-01-24:
This may help in the long run (i) maintainers that were not reviewers, and (ii) outsiders which are looking to pick up something "valuable" where others left off.

OlegHahm, 2017-01-25:
I'm not sure how this label would differentiate from the "memo" label.

emmanuelsearch, 2017-01-25:
good point.

emmanuelsearch, 2017-01-25:
the one thing we don't have, though, whatever the label, is a policy when to close PRs that drag when the author unresponsive for months.

PeterKietzmann, 2018-06-26:
No progress/outcome since 1.5 years -> I tend to close this issue. @danpetry what do you think, is it worth a note in the [Maintainer Guidelines](https://github.com/RIOT-OS/RIOT/blob/master/MAINTAINING.md)? If not, where else?

danpetry, 2018-06-26:
@PeterKietzmann I would say that PRs older than three months ideally and six at the most, with no activity, should be closed. I get the impression that opinion is divided, though. Might be useful to try and raise a PR against the maintainer guidelines or something and see how that pans out.
IMO issues are ok, though. A lot of open source projects I come across have a longer issue list, I think that's alright as it reminds us the things we need to do.
I think the point though is organisation - it's not the actual number at the top of the page which causes a problem, it's the "big ball of mud" pile of work that we dig through. Would be easier to work together if the structure of our work was clearer, IMO.

danpetry, 2018-06-26:
(Worth mentioning that people form task forces and that kind of thing for packets of work, so it's not completely amorphous.)

PeterKietzmann, 2018-06-26:
> I would say that PRs older than three months ideally and six at the most, with no activity, should be closed. I get the impression that opinion is divided, though. Might be useful to try and raise a PR against the maintainer guidelines or something and see how that pans out.

I agree with you. Would you mind to do so? I guess @emmanuelsearch would be a good assignee and we could close this issue afterwards.

Once I read from @miri64 something like " close PRs that had no activity for six month so" but I couldn't find the source of it, though I share that thinking.

danpetry, 2018-06-26:
Sorry about this, but I've actually got quite a lot on my plate at the moment and I've just spent a fair amount of time on a process discussion getting this merged. I would be happy to help when I've got more time - I don't need to be the one opening all the PRs against the guidelines, though.

miri64, 2018-06-26:
@PeterKietzmann I did this basically on the basis of "feeling within the community". Nothing is lost if you close a PR, so I did not see i t as something that needs long discussion. If other maintainers disagreed, they could just reopen it again (or the contributor could ask for it). See, e.g. #2968 ;-). This however is only valid for feature PRs. Bugfixes (which do not involve major remodeling) or issues should not be closed IMHO, just because they are stalled. I agree with @danpetry that we need some kind of organizing system. He mentioned once the labeling system used by the Rust community, which taken at verbatim I think is quite overkill, but I think looking into that would be a good starting point for someone(TM) to find a solution for structurizing PRs and issues.

PeterKietzmann, 2018-06-26:
I have the impression you didn't get ee right. Let me summarize: As @emmanuelsearch stated:

> the one thing we don't have, though, whatever the label, is a policy when to close PRs that drag when the author is unresponsive for months in a row.

My proposal was to spend a note on the newly introduced maintainer guidelines that says "if unresponsive for time X, close .... blabla can always be re-opend...". If we put that in actual words, we need to define a reasonable "expiration date".

PeterKietzmann, 2018-06-29:
Offline @danpetry and @miri64 convinced me that a simple note about closing strategies neither solves or improves the logging concerns nor it improves our PR/issue organization in general. So let's keep this issue for now

miri64, 2018-07-04:
Regarding issue labeling, there is this issue too btw https://github.com/RIOT-OS/RIOT/issues/6267 (just because it popped up in my timeline again ;-))

miri64, 2019-08-10:
I guess @emmanuelsearch's request in the original post is fulfilled now by the "State: stale" ;-)

closed 2019-08-10 by miri64