RIOT-OS/RIOT #6473: List of compiling issues on OS X

kYc0o, 2017-01-25:
I just ran

```bash
for i in $(ls); do if [[ -d "$i" ]]; then echo $i;  make -C $i; fi;  done;
```

on the RIOT root to know which things are free of error on OS X, and that gave results which can be useful

- [ ] **unittests**

```console
"/Applications/Xcode.app/Contents/Developer/usr/bin/make" -C /Users/facosta/git/RIOT2/RIOT/pkg/relic
rm -Rf /Users/facosta/git/RIOT2/RIOT/tests/unittests/bin/pkg/native/relic
mkdir -p /Users/facosta/git/RIOT2/RIOT/tests/unittests/bin/pkg/native/relic
/Users/facosta/git/RIOT2/RIOT/dist/tools/git/git-cache clone "https://github.com/relic-toolkit/relic.git" "cdcfaeef101d18c3231c3b46359c519dd72682e8" "/Users/facosta/git/RIOT2/RIOT/tests/unittests/bin/pkg/native/relic"
Cloning into '/Users/facosta/git/RIOT2/RIOT/tests/unittests/bin/pkg/native/relic'...
remote: Counting objects: 16213, done.
remote: Total 16213 (delta 0), reused 0 (delta 0), pack-reused 16213
Receiving objects: 100% (16213/16213), 3.69 MiB | 181.00 KiB/s, done.
Resolving deltas: 100% (13018/13018), done.
Checking connectivity... done.
HEAD is now at cdcfaee... Fix cycle counter in ARM.
touch /Users/facosta/git/RIOT2/RIOT/tests/unittests/bin/pkg/native/relic/.git-downloaded
git -C /Users/facosta/git/RIOT2/RIOT/tests/unittests/bin/pkg/native/relic checkout -f cdcfaeef101d18c3231c3b46359c519dd72682e8
HEAD is now at cdcfaee... Fix cycle counter in ARM.
git -C /Users/facosta/git/RIOT2/RIOT/tests/unittests/bin/pkg/native/relic am --ignore-whitespace "/Users/facosta/git/RIOT2/RIOT/pkg/relic"/patches/*.patch
Applying: fixed signedness of counter variable
Applying: don't redefine ALIGN
Applying: require only CMake version 2.8
touch /Users/facosta/git/RIOT2/RIOT/tests/unittests/bin/pkg/native/relic/.git-patched
./fix-util_print_wo_args.sh /Users/facosta/git/RIOT2/RIOT/tests/unittests/bin/pkg/native/relic
sed: RE error: illegal byte sequence
make[1]: *** [fix_source] Error 1
make: *** [/Users/facosta/git/RIOT2/RIOT/tests/unittests/bin/native/relic.a] Error 2
```

- [ ] **warn_conflict** Errors expected but no errors…
- [ ] **saul** Why it works? Does native provide support for SAUL?
- [ ] **pkg_u8g2** Why it works? Is it possible to use it in native?
- [ ] **driver_pir** Why it works? It should send an error since the pin for PIR is not provided in native
- [ ] **driver_lpd8808** Why it works? It's a LED strip driver
- [ ] **tests_driver_kw2xrf** 

```console
RIOT/drivers/kw2xrf/kw2xrf.c:376:23: error: comparison of integers of different signs: 'int' and 'unsigned int'
      [-Werror,-Wsign-compare]
    for (int i = 0; i < IEEE802154_LONG_ADDRESS_LEN; i++) {
                    ~ ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~
RIOT/drivers/kw2xrf/kw2xrf.c:432:23: error: comparison of integers of different signs: 'int' and 'unsigned int'
      [-Werror,-Wsign-compare]
    for (int i = 0; i < CPUID_LEN; i++) {
                    ~ ^ ~~~~~~~~~
RIOT/drivers/kw2xrf/kw2xrf.c:1234:27: error: comparison of integers of different signs: 'int' and 'size_t' (aka 'unsigned long')
      [-Werror,-Wsign-compare]
        for (int i = 0; i < pkt->size; i++) {
                        ~ ^ ~~~~~~~~~
RIOT/drivers/kw2xrf/kw2xrf.c:1255:27: error: comparison of integers of different signs: 'int' and 'size_t' (aka 'unsigned long')
      [-Werror,-Wsign-compare]
        for (int i = 0; i < payload->size; i++) {
                        ~ ^ ~~~~~~~~~~~~~
```

- [ ] **tests_driver_isl29125** 

```console
RIOT/tests/driver_isl29125/main.c:66:23: error: comparison of integers of different signs: 'int' and 'unsigned long'
      [-Werror,-Wsign-compare]
    for (int i = 0; i < sizeof(modes); i++) {
                    ~ ^ ~~~~~~~~~~~~~
```

 - [ ] **tests_driver_hih6130** 

```console
RIOT/tests/driver_hih6130/main.c:74:42: error: using integer absolute value function 'abs' when argument is of floating point type
      [-Werror,-Wabsolute-value]
            (int)integral, (unsigned int)abs(fractional * 10000.f));
                                         ^
RIOT/tests/driver_hih6130/main.c:74:42: note: use function 'fabsf' instead
            (int)integral, (unsigned int)abs(fractional * 10000.f));
                                         ^~~
                                         fabsf
RIOT/tests/driver_hih6130/main.c:77:42: error: using integer absolute value function 'abs' when argument is of floating point type
      [-Werror,-Wabsolute-value]
            (int)integral, (unsigned int)abs(fractional * 10000.f));
                                         ^
RIOT/tests/driver_hih6130/main.c:77:42: note: use function 'fabsf' instead
            (int)integral, (unsigned int)abs(fractional * 10000.f));
                                         ^~~
                                         fabsf
```

 - [ ] **tests_driver_encx24j600**

```console
RIOT/drivers/encx24j600/encx24j600.c:316:23: error: comparison of integers of different signs: 'int' and 'unsigned int'
      [-Werror,-Wsign-compare]
    for (int i = 0; i < count; i++) {
                    ~ ^ ~~~~~
```

 - [ ] **tests_driver_enc28j60**

```console
RIOT/drivers/enc28j60/enc28j60.c:235:23: error: comparison of integers of different signs: 'int' and 'unsigned int'
      [-Werror,-Wsign-compare]
    for (int i = 0; i < count; i++) {
                    ~ ^ ~~~~~
RIOT/drivers/enc28j60/enc28j60.c:316:19: error: comparison of integers of different signs: 'int' and 'unsigned int'
      [-Werror,-Wsign-compare]
        if (res++ >= STARTUP_TIMEOUT) {
            ~~~~~ ^  ~~~~~~~~~~~~~~~
```

- [ ] **tests_driver_dht** Why it works? How can we attach this kind of sensor on native?
- [ ] **tests_driver_at86rf2xx**

```console
RIOT/tests/driver_at86rf2xx/addr.c:23:23: error: comparison of integers of different signs: 'int' and 'size_t'
      (aka 'unsigned long') [-Werror,-Wsign-compare]
    for (int i = 0; i < addr_len; i++) {
                    ~ ^ ~~~~~~~~
```

I'll try to run the same on the examples folder. This is just a trial towards #1121 .

basilfx, 2017-01-29:
> pkg_u8g2 Why it works? Is it possible to use it in native?

There are four output backends, of which the first redirects to stdout, the second to SDL, the third to SPI and the fourth to I2C.

kYc0o, 2017-01-29:
>There are four output backends, of which the first redirects to stdout, the second to SDL, the third to SPI and the fourth to I2C.

>First two can run on native without problems.

Super, I'll mark it as done.

smlng, 2017-06-06:
@kYc0o please not that with #6444 and #6599 the default board for certain boards changed. Which means, that even without supplying a specific board via `BOARD=<something>` some tests will use e.g. `BOARD=samr21-xpro` instead of `native`. Specifically this holds for most periph tests, that require SPI or I2C <- not available for `native` (yet).

That said: as I use macOS myself and do also care about those (nasty) compile errors or warnings: could you please update your list. Many issues are resolved in the mean time anyway, so could also be reasonable to close this one (for now).

kYc0o, 2017-10-25:
I removed non-related OS X errors from the list according to @smlng [comment](https://github.com/RIOT-OS/RIOT/issues/6473#issuecomment-306437140).

However I still found errors and proposed a fix for one of them. @miri64 can you check the ipv6 related ones?

I'll open another issue for board related errors found by clang.

miri64, 2017-10-25:
> However I still found errors and proposed a fix for one of them. @miri64 can you check the ipv6 related ones?

Will do after the release.

kYc0o, 2018-01-16:
AFAIK all these issues were addressed, there are new ones but I'll open another issue with them. Closing.

closed 2018-01-16 by kYc0o