RIOT-OS/RIOT #6476: GNRC TCP: code review

OlegHahm, 2017-01-25:
Based on a rather superficial review of GNRC TCP I found a couple of issues that must be addressed:

## examples
* examples are tests
* examples are missing a README
* cli is a misleading name (it does not provide a Command Line Interface, but a CLIent)
* client examples contain a hardcoded IP address for the communication peer, but this address is not set on the server side
* example seems to be unnecessary complex

## gnrc_tcp
* at least parts of the documentation still refer to ng_... (e.g., ng_pktsnip)
* parts of the code do not obey RIOT coding conventions (else in the same line as the closing brace, camelCase...)
* several functions exceed 100 lines 
* header guards with trailing underscores are re-introduced
* gnrc_tcp_open_passive() (which I don't find a very speaking name) requires one thread per open TCP port, IMO this could (and should) be solved more efficiently with GNRC. E.g., one function could register the TCP thread to wait for incoming connections and actually in
coming connection requests could use TCP. Not even POSIX requires one thread per connection.
* gnrc_tcp_calc_csum() should IMO not be part of the main public API for gnrc_tcp() 
* gnrc_tcp.c: 
    - includes more than 20 headers - I cannot believe that this is mandatory
    - contain switch-case statements with only one case
    - gnrc_tcp_tcb_init() can be significantly be shortened using memset()
* gnrc_tcp_eventloop.c allocates a message queue on the function stack
* gnrc_tcp_fsm.c - why do some function have a bool pointer(!) as output parameter? An int return value offers more than enough options to indicate different states
* Why does _gnrc_tcp_pid() start with an underscore and why is it declared as extern in multiple headers?

brummer-simon, 2017-01-28:
Hi Oleg,

Thanks for your Input. I'll try to address the Issues asap. I would use this Issue to discuss some things.

First of all: Whats wrong with "gnrc_tcp_calc_csum()" as Part of public API? I put it there because UDP does this as well.

brummer-simon, 2017-01-29:
Oh and about include cleanup. Whats RIOTs usual policy there?

In a Header File I would include every Header that contains a Type used in the function prototypes and in the Source Files I would include everything that is used except includes that were part of the associated header file. Right?

brummer-simon, 2017-01-29:
And about the examples. Those were actually thought as test program. I think about removing them as soon as I integrated TCP into the gnrc_networking example. What do you think @OlegHahm ?

miri64, 2017-01-29:
> And about the examples. Those were actually thought as test program. 

Then why were they in `examples/` and not in `tests/` in the first place?

>  I think about removing them as soon as I integrated TCP into the gnrc_networking example.

I'm not sure integrating TCP into `gnrc_networking` is such a good idea :-/. Bloat in this application is already showing. I'd rather prefer having two separate applications for the two transport protocols. Regarding removing the examples: how about moving them to `tests/` instead?

brummer-simon, 2017-01-29:
Hi miri64,

The original Idea of my examples was to show potential user how to use TCP (thats why it ended up in examples) it became more and more my personal test application. I am fine with a second application similar to gnrc_networking (should it contain udp as well?)

I move the examples to tests  #6498. I do some cleanup mentioned here as well.

miri64, 2017-01-29:
> I am fine with a second application similar to gnrc_networking (should it contain udp as well?)

I don't see why UDP as well would be necessary. An example should be as much to the point as possible. Maybe take the TCP part of my [lwIP tests](https://github.com/RIOT-OS/RIOT/pull/6372) as an example. Though they are based on `sock` they show-case the different functionalities both a TCP client and server need to provide quite nicely, in my opinion.

> I move the examples to tests in #6498. I do some cleanup mentioned here as well.

Already offered my complain about doing this in #6498 in that PR. Doing it in a separate PR is cleaner, as it has nothing to do with the topic of #6498 (improving option handling, according to the title).

brummer-simon, 2017-02-01:
#6541= Renames "_gnrc_tcp_pid" to "gnrc_tcp_pid", removes trailing Underscores in header guards, centralizes the declaration of extern variables into a single header (common.h)

PeterKietzmann, 2017-02-03:
@brummer-simon or @OlegHahm can you provide checkboxes above so we can track the progress?

OlegHahm, 2017-02-03:
> First of all: Whats wrong with "gnrc_tcp_calc_csum()" as Part of public API? I put it there because UDP does this as well.

Sorry for the delayed reply:
I think it would make sense if the public API only provide those functions which are supposed to be called by the upper layer (both for TCP and UDP). If additional functions need to be exposed I would put them into a separate header, in order to keep the interface slim.

brummer-simon, 2017-02-26:
Whats the Problem with this 'gnrc_tcp_eventloop.c allocates a message queue on the function stack' point?

I mean the message queue of gnrc tcp's eventloop thread is allocated on the thread function. The message queue lives as long the thread lives and the thread is an eventloop that never returns...

brummer-simon, 2017-03-01:
With the merging of  #6541, the issue List can be updated.  #6541 addresses:

- at least parts of the documentation still refer to ng_... (e.g., ng_pktsnip)
-  parts of the code do not obey RIOT coding conventions (else in the same line as the closing brace, camelCase...)
- several functions exceed 100 lines
- header guards with trailing underscores are re-introduced
- includes more than 20 headers - I cannot believe that this is mandatory
- Why does _gnrc_tcp_pid() start with an underscore and why is it declared as extern in multiple headers?

OlegHahm, 2017-03-02:
> I mean the message queue of gnrc tcp's eventloop thread is allocated on the thread function. The message queue lives as long the thread lives and the thread is an eventloop that never returns...

Exactly. So, what is the point in having the queue inside the stack when the memory is never freed anyway?

OlegHahm, 2017-03-02:
For the discussion see: https://github.com/RIOT-OS/RIOT/issues/2851

brummer-simon, 2017-03-02:
Okay I get what you mean. I'll put the message queue outside of the thread function.

brummer-simon, 2017-03-06:
> Sorry for the delayed reply:
 I think it would make sense if the public API only provide those functions which are supposed to be   called by the upper layer (both for TCP and UDP). If additional functions need to be exposed I would put them into a separate header, in order to keep the interface slim.

I am not sure about that. The API here is not intended to be the 'public' API. GNRC TCP should be used via Sock or any other general Network API, that might come in the future. I would prefer for leave the functions where they are.

brummer-simon, 2017-03-06:
About:
> - contain switch-case statements with only one case

I know this is bad practice in general, in this case I did it on purpose. It is in places where something Network Layer specific must happen. I added the switch statements to make it obvious that code handling AF_INET or AF_LOCAL must be added there. Is it okay to leave it here, or should I change this?

OlegHahm, 2017-03-06:
Concerning the API, from the perspective of someone who prefers GNRC over sock (because he never read the doc :wink:) I don't share your view, but I have no strong opinion.

brummer-simon, 2017-03-06:
Does this mean we can cross:
> - gnrc_tcp_calc_csum() should IMO not be part of the main public API for gnrc_tcp()

Of the list?

smlng, 2017-03-06:
>Does this mean we can cross:
>
>       gnrc_tcp_calc_csum() should IMO not be part of the main public API for gnrc_tcp()
>
>of the list?

Though I understand the reasoning by @OlegHahm that this function should not be public, UDP and ICMPv6 implement this the same way as @brummer-simon did. Hence, I'd say this should be solved (if this is really an issue) once and for all protocols - in a separate PR, that is.

miri64, 2017-03-07:
> Though I understand the reasoning by @OlegHahm that this function should not be public, UDP and ICMPv6 implement this the same way as @brummer-simon did. Hence, I'd say this should be solved (if this is really an issue) once and for all protocols - in a separate PR, that is.

May I intercept this discussion: This is done intentionally. The checksum is generated by the `gnrc_ipv6` module *after* the source address was determined (since it is required for the checksum). So unless you want to do some `extern` magic in `gnrc_ipv6` `gnrc_*_calc_csum()` has to be public.

smlng, 2017-03-07:
@miri64, still I think @OlegHahm has a point: as a (higher layer) user (application) you would not invoke (and should not (?)) `gnrc_*_calc_csum()` when using icmp, tcp, or udp. So why expose it the user anyway?

miri64, 2017-03-07:
> @miri64, still I think @OlegHahm has a point: as a (higher layer) user (application) you would not invoke (and should not (?)) gnrc_*_calc_csum() when using icmp, tcp, or udp. So why expose it the user anyway?

I do not consider the network internal stuff as a "higher layer user API". Contradicting @OlegHahm: That's what sock is for. In this case they are used by another (sub-)module, so they need to be exposed somehow.

OlegHahm, 2017-03-07:
I've never objected putting this function in any header, I would just differentiate between the API exposed to the application and the API exposed to other layers. If the idea is that GNRC's transport layer should be only accessed via sock, this is not necessary, agreed. So far I tried to avoid sock mostly, because I found netapi more convenient.

brummer-simon, 2017-03-21:
About:

-  gnrc_tcp_fsm.c - why do some function have a bool pointer(!) as output parameter? An int return value offers more than enough options to indicate different states

The int return value for the functions with the bool pointer, encode already three different states in the return value. Positive values are used for the amount of bytes sent/received, zero for no error ocured and negative values are for errors. I do not want to encode additionally, that a thread that is blocked by the gnrc_tcp - API should be notified. Thats the reason for the bool pointer here.

I see three options here to solve this issue:
1. leave it as it is
2. Put the flag out of the function context as static variable.
3. Try to encode the boolean into the negative number space, normally reserved for error states.

My vote goes for option 2. Any thoughts ?

brummer-simon, 2017-04-26:
With the merging of #6742, a few points can be crossed of the list:

- examples are missing a README
- client examples contain a hardcoded IP address for the communication peer, but this address is not set on the server side
-  contain switch-case statements with only one case
-  gnrc_tcp_tcb_init() can be significantly be shortened using memset()
-  gnrc_tcp_eventloop.c allocates a message queue on the function stack
- gnrc_tcp_fsm.c - why do some function have a bool pointer(!) as output parameter? An int return value offers more than enough options to indicate different states

miri64, 2017-04-26:
> - gnrc_tcp_tcb_init() can be significantly be shortened using memset()

Still some redundant assignments to zero in there:

```C
tcb->address_family = AF_UNSPEC;
/* ... */
tcb->local_port = PORT_UNSPEC;
tcb->peer_port = PORT_UNSPEC;
tcb->state = FSM_STATE_CLOSED;
/* ... */
tcb->owner = KERNEL_PID_UNDEF;
```

Rest I can confirm (and sorry for not pointing out the short-coming mentioned above in #6742 already).

brummer-simon, 2017-04-26:
It is true that the Defines are mostly Zero as well. I still would like to keep those.
It is possible that values of those Macros change from zero to something else, introducing bug that are hard to track.

miri64, 2017-04-26:
`KERNEL_PID_UNDEF` is unlikely to change, so is `AF_UNSPEC`. The latter [is even standardized](http://pubs.opengroup.org/onlinepubs/9699919799/basedefs/sys_socket.h.html):

> The value of AF_UNSPEC shall be 0.

brummer-simon, 2017-04-26:
Okay, I'll remove the rest of them.

brummer-simon, 2017-04-26:
@miri64 - Would you do me a favor and review #6972 ?

miri64, 2017-04-26:
Done.

brummer-simon, 2017-04-26:
The remaining Issue:

- gnrc_tcp_open_passive() (which I don't find a very speaking name) requires one thread per open TCP port, IMO this could (and should) be solved more efficiently with GNRC. E.g., one function could register the TCP thread to wait for incoming connections and actually in
coming connection requests could use TCP. Not even POSIX requires one thread per connection.

Requires a bit more preparation. I plan to introduce a posix-ish listen and accept call to replace gnrc_tcp_open_passive(). It should follow a similar semantic as, in the already defined sock tcp interface, therefore the port to sock should be easy and straight forward. Any objections ?

miri64, 2017-04-26:
> several functions exceed 100 lines

is also not ticked off yet. Any reason? Above you stated it is fixed.

brummer-simon, 2017-04-26:
It is. I think it was just not crossed of the list.

brummer-simon, 2017-04-26:
At least i am not aware the any line is longer than 100 characters

PeterKietzmann, 2019-07-23:
@brummer-simon any news?

miri64, 2020-01-24:
Ping @brummer-simon how far along is this issue by now?

brummer-simon, 2020-01-24:
Since you are reviewing everything TCP related it is up to you to decide if this Issue is solved.

From my point of view the last remaining checkboxes are address by now.

miri64, 2020-07-01:
Agreed!

closed 2020-07-01 by miri64