RIOT-OS/RIOT #6484: boards/nucleo-l1: hard faults with shell?

PeterKietzmann, 2017-01-26:
Running e.g. tests/shell, examples/default or examples/gnrc_networking on a nucelo-l1 hard-faults. I didn't check other nucleo boards nor did I look into details

haukepetersen, 2017-01-26:
Other nucleo boards seem to be fine, but run into the same hard-fault on the l1. Seems to me, that something is broken with the UART interrupt, looking into it now.

haukepetersen, 2017-01-27:
hm, seems like we get hardfaults on any interrupt -> the `xtimer_msg` test reboots every second...

haukepetersen, 2017-01-27:
this is getting creepy: `tests/periph_timer` works as expected. `xtimer_msg` and `xtimer_usleep` work fine with `DEVELHELP` enabled, but hard fault once `DEVELHELP` is disabled...

haukepetersen, 2017-01-27:
one step closer: git bisect lead me to `807a190e2086793225e774a5d64983467cf221b7` (cpu: cortexm: adapt to periph/pm) as the bad commit. Strange.

haukepetersen, 2017-01-27:
hmm, setting, or better waking up the CPU from sleep, is the problem here. So what is the L1 doing (or not doing), so it crashes when sleeping?!

haukepetersen, 2017-01-27:
fixed by #6499

closed 2017-01-27 by PeterKietzmann

PeterKietzmann, 2017-01-27:
#6499 was merged

kYc0o, 2018-01-08:
I think a regression of this bug came back after #7385 was merged. Should we reopen this one or keep going on #8024 ?

PR #6499 (2017-01-27): cpu/cortexm: disable interrupts before sleeping
fixes #6484 (though temporarily for now)

A real fix will be provided with #6239