RIOT-OS/RIOT #6490: ATOMU

melshuber, 2017-01-26:
OFF Topic:

I think i just found an issue in ATOMIC_VALUE on avr8

```
#include <atomic.h>

atomic_int_t atomic = ATOMIC_INIT(0);

int atomic_val(void) {
    return ATOMIC_VALUE(atomic);
}

int main(void)
{
    while (1);
}
```

copiles to
```
$ BOARD=arduino-mega2560 make
$ avr-objdump -S bin/arduino-mega2560/atomic/main.o

Disassembly of section .text.atomic_val:

00000000 <atomic_val>:
#include <atomic.h>

atomic_int_t atomic = ATOMIC_INIT(0);

int atomic_val(void) {
    return ATOMIC_VALUE(atomic);
   0:	80 91 00 00 	lds	r24, 0x0000
   4:	90 91 00 00 	lds	r25, 0x0000
}
   8:	08 95       	ret

Disassembly of section .text.startup.main:

00000000 <main>:

int main(void)
{
   0:	00 c0       	rjmp	.+0      	; 0x2 <__zero_reg__+0x1>
```

you see two `lds` instructions in atomic_val.c

Problem case: Suppose a higher priority thread interrupts just after the first `lds` and changes the value itself to 0xFFFF. In this case an inconsistent value would be returned (endianness depended either 0xFF or 0xFF00)

melshuber, 2017-01-26:
sorry i did accidently hit enter

miri64, 2017-01-28:
Can you please give the issue a more informative title.

miri64, 2017-01-28:
Ah nvm. Just saw #6491.

closed 2017-01-28 by miri64