RIOT-OS/RIOT #6491: ATOMIC_VALUE might return inconsistent values on avr8 (or other 8bit mcu)

melshuber, 2017-01-26:
threre might an issue in ATOMIC_VALUE(...) on avr8

```
#include <atomic.h>

atomic_int_t atomic = ATOMIC_INIT(0);

int atomic_val(void) {
    return ATOMIC_VALUE(atomic);
}

int main(void)
{
    while (1);
}
```

copiles to
```
$ BOARD=arduino-mega2560 make
$ avr-objdump -S bin/arduino-mega2560/atomic/main.o

Disassembly of section .text.atomic_val:

00000000 <atomic_val>:
#include <atomic.h>

atomic_int_t atomic = ATOMIC_INIT(0);

int atomic_val(void) {
    return ATOMIC_VALUE(atomic);
   0:	80 91 00 00 	lds	r24, 0x0000
   4:	90 91 00 00 	lds	r25, 0x0000
}
   8:	08 95       	ret

Disassembly of section .text.startup.main:

00000000 <main>:

int main(void)
{
   0:	00 c0       	rjmp	.+0      	; 0x2 <__zero_reg__+0x1>
```

you see two `lds` instructions in atomic_val()

Problem case: Suppose a higher priority thread interrupts just after the first `lds` and changes the value itself to 0xFFFF. In this case an inconsistent value would be returned (endianness depended either 0xFF or 0xFF00)

The correct behavior should be return either the old or the new value

gebart, 2017-01-26:
I agree, it's not correct. The best solution is in my opinion already PRed in #5688

OlegHahm, 2017-01-26:
Does the AVR toolchain actually has C11 support?

gebart, 2017-01-26:
It should work on AVR since it has a more recent gcc than the msp430 does.

OlegHahm, 2017-01-27:
Which is not so difficult. ;-)

melshuber, 2017-01-27:
as far as i have seen since gcc 4.9 the header stdatomic.h exists, but i think the functions are not implemented

I just built a tool chain with ct-ng (gcc 6.3; all default)

It seems there are implementations missing (e.g. __atomic_load_2, probably others as well)
```
$ avr-gcc -v
Using built-in specs.
Reading specs from /home/melshuber/x-tools/avr/lib/gcc/avr/6.3.0/device-specs/specs-avr2
COLLECT_GCC=avr-gcc
COLLECT_LTO_WRAPPER=/home/melshuber/x-tools/avr/libexec/gcc/avr/6.3.0/lto-wrapper
Target: avr
Configured with: /home/melshuber/Work/avr/.build/src/gcc-6.3.0/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=avr --prefix=/home/melshuber/x-tools/avr --with-local-prefix=/home/melshuber/x-tools/avr/avr --without-headers --with-newlib --enable-threads=no --disable-shared --with-pkgversion='crosstool-NG crosstool-ng-1.22.0-484-g462c3c0' --disable-__cxa_atexit --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --with-gmp=/home/melshuber/Work/avr/.build/avr/buildtools --with-mpfr=/home/melshuber/Work/avr/.build/avr/buildtools --with-mpc=/home/melshuber/Work/avr/.build/avr/buildtools --with-isl=/home/melshuber/Work/avr/.build/avr/buildtools --enable-lto --with-host-libstdcxx='-static-libgcc -Wl,-Bstatic,-lstdc++,-Bdynamic -lm' --enable-target-optspace --disable-nls --enable-multiarch --enable-languages=c,c++
Thread model: single
gcc version 6.3.0 (crosstool-NG crosstool-ng-1.22.0-484-g462c3c0)
```

```
#include <stdatomic.h>

atomic_int atomic = ATOMIC_VAR_INIT(0);

int atomic_val(void) {
    return atomic_load(&atomic);
}

int main(void)
{
    volatile int a = atomic_val();
    (void)a;
    while (1);
}
```

```
$ BOARD=arduino-mega2560 QUIET=0 make
Building application "atomic" for "arduino-mega2560" with MCU "atmega2560".

'/home/melshuber/Work/riotfork/dist/tools/genconfigheader/genconfigheader.sh' '/home/melshuber/Work/riotfork/tests/atomic/bin/arduino-mega2560/riotbuild/riotbuild.h' -DDEVELHELP -Werror -mmcu=atmega2560   -ffunction-sections -fdata-sections -ggdb -g3 -Os -DCOREIF_NG=1 -ffunction-sections -fdata-sections -DBOARD_ARDUINO_MEGA2560=\"arduino-mega2560\" -DRIOT_BOARD=BOARD_ARDUINO_MEGA2560 -DCPU_ATMEGA2560=\"atmega2560\" -DRIOT_CPU=CPU_ATMEGA2560 -DMCU_ATMEGA2560=\"atmega2560\" -DRIOT_MCU=MCU_ATMEGA2560 -std=c99 -fno-delete-null-pointer-checks -Wstrict-prototypes -Werror=strict-prototypes -Wold-style-definition -Werror=old-style-definition -fno-common -Wall -DMODULE_ATMEGA_COMMON -DMODULE_AUTO_INIT -DMODULE_BOARD -DMODULE_CORE -DMODULE_CORE_MSG -DMODULE_CPU -DMODULE_ISRPIPE -DMODULE_PERIPH -DMODULE_PERIPH_COMMON -DMODULE_SYS -DMODULE_TSRB -DMODULE_UART_STDIO -DRIOT_VERSION=\"2017.01-devel-829-g8fcc5-melshuber-vm.lan-rmutex-rebased\"
Keeping old /home/melshuber/Work/riotfork/tests/atomic/bin/arduino-mega2560/riotbuild/riotbuild.h
DIRS=" " "make" -C /home/melshuber/Work/riotfork/tests/atomic -f /home/melshuber/Work/riotfork/Makefile.application
make[1]: Entering directory '/home/melshuber/Work/riotfork/tests/atomic'
avr-gcc \
	-DRIOT_FILE_RELATIVE=\"tests/atomic/main.c\" \
	-DRIOT_FILE_NOPATH=\"main.c\" \
	-Werror -mmcu=atmega2560 -ffunction-sections -fdata-sections -ggdb -g3 -Os -ffunction-sections -fdata-sections -std=c99 -fno-delete-null-pointer-checks -Wstrict-prototypes -Werror=strict-prototypes -Wold-style-definition -Werror=old-style-definition -fno-common -Wall -include '/home/melshuber/Work/riotfork/tests/atomic/bin/arduino-mega2560/riotbuild/riotbuild.h'  -I/home/melshuber/Work/riotfork/core/include -I/home/melshuber/Work/riotfork/drivers/include -I/home/melshuber/Work/riotfork/sys/include -I/home/melshuber/Work/riotfork/cpu/atmega2560/include -I/home/melshuber/Work/riotfork/boards/arduino-mega2560/include -I/home/melshuber/Work/riotfork/cpu/atmega_common/include -isystem/home/melshuber/Work/riotfork/cpu/atmega_common/avr-libc-extra -I/home/melshuber/Work/riotfork/sys/libc/include  -MD -MP -c -o /home/melshuber/Work/riotfork/tests/atomic/bin/arduino-mega2560/atomic/main.o /home/melshuber/Work/riotfork/tests/atomic/main.c
"make" -C /home/melshuber/Work/riotfork/boards/arduino-mega2560
"make" -C /home/melshuber/Work/riotfork/core
make[2]: Entering directory '/home/melshuber/Work/riotfork/core'
make[2]: Leaving directory '/home/melshuber/Work/riotfork/core'
"make" -C /home/melshuber/Work/riotfork/cpu/atmega2560
make[2]: Entering directory '/home/melshuber/Work/riotfork/cpu/atmega2560'
"make" -C /home/melshuber/Work/riotfork/cpu/atmega_common
make[3]: Entering directory '/home/melshuber/Work/riotfork/cpu/atmega_common'
"make" -C /home/melshuber/Work/riotfork/cpu/atmega_common/periph
make[3]: Leaving directory '/home/melshuber/Work/riotfork/cpu/atmega_common'
make[2]: Leaving directory '/home/melshuber/Work/riotfork/cpu/atmega2560'
"make" -C /home/melshuber/Work/riotfork/drivers
make[2]: Entering directory '/home/melshuber/Work/riotfork/drivers'
"make" -C /home/melshuber/Work/riotfork/drivers/periph_common
make[3]: Entering directory '/home/melshuber/Work/riotfork/drivers/periph_common'
make[3]: Leaving directory '/home/melshuber/Work/riotfork/drivers/periph_common'
make[2]: Leaving directory '/home/melshuber/Work/riotfork/drivers'
"make" -C /home/melshuber/Work/riotfork/sys
make[2]: Entering directory '/home/melshuber/Work/riotfork/sys'
"make" -C /home/melshuber/Work/riotfork/sys/auto_init
make[3]: Entering directory '/home/melshuber/Work/riotfork/sys/auto_init'
[... cut ...] /bin/arduino-mega2560/atomic.map -Wl,--cref -mmcu=atmega2560   -ggdb -g3 -Os -e reset_handler -Wl,--gc-sections -static -lgcc
/home/melshuber/Work/riotfork/tests/atomic/bin/arduino-mega2560/atomic.a(main.o): In function `atomic_val':
/home/melshuber/Work/riotfork/tests/atomic/main.c:6: undefined reference to `__atomic_load_2'
collect2: error: ld returned 1 exit status
/home/melshuber/Work/riotfork/tests/atomic/../../Makefile.include:275: recipe for target 'all' failed
make: *** [all] Error 1
```

gebart, 2017-01-27:
@melshuber could you try it with #5688 applied? It should provide everything you need.

melshuber, 2017-01-27:
on #5688 it seem reasonable on avr8 (main.c see above)

```
$ git log --oneline -n15
692a944 sys/posix: Add missing Doxygen comment on spinlock flag member
93be91d doc/doxygen: Exclude msp430-common imported newlib headers
26b41f3 doc/doxygen: Sort exclude patterns
95e89da dist/tools/licenses: Add BSD 1 clause style license from newlib sys/features.h
31b709e core: Remove atomic in favour of C11 stdatomic
ee19d1e x86: Remove atomic_arch
6e07833 cortexm_common: Remove atomic_arch
d99c6b2 tests/unittests/core-atomic: Use stdatomic.h instead of custom atomic.h
01f20ee tests/emb6: Use stdatomic.h instead of custom atomic.h
cf5535a sys/posix/pthread: Use C11 atomic instead of atomic.h
91a2948 msp430_common: Add supplemental __clzhi2, __ctzhi2, for old MSPGCC
501d5ba msp430-common: Add stdatomic.h, sys/cdefs.h, sys/features.h from newlib
a13462e core: rename c11_atomic -> atomic_c11
3c6e197 core: add __sync_x library helpers for <gcc-4.7
2fd55e1 core/c11_atomic.c: Add remaining C11 atomic library helpers for GCC
```
```
$ avr-gdb -batch -ex 'file bin/arduino-mega2560/atomic.elf' -ex 'disassemble atomic_val'
Dump of assembler code for function atomic_val:
   0x00000152 <+0>:	ldi	r22, 0x05	; 5
   0x00000154 <+2>:	ldi	r23, 0x00	; 0
   0x00000156 <+4>:	ldi	r24, 0xD6	; 214
   0x00000158 <+6>:	ldi	r25, 0x02	; 2
   0x0000015a <+8>:	jmp	0x1f2	;  0x1f2 <__atomic_load_2>
End of assembler dump.
```
```
$ avr-gdb -batch -ex 'file bin/arduino-mega2560/atomic.elf' -ex 'disassemble __atomic_load_2'
Dump of assembler code for function __atomic_load_2:
   0x000001f2 <+0>:	push	r28
   0x000001f4 <+2>:	push	r29
   0x000001f6 <+4>:	movw	r28, r24
   0x000001f8 <+6>:	call	0x73c	;  0x73c <irq_arch_disable>
   0x000001fc <+10>:	ld	r0, Y+
   0x000001fe <+12>:	ld	r29, Y
   0x00000200 <+14>:	mov	r28, r0
   0x00000202 <+16>:	call	0x754	;  0x754 <irq_arch_restore>
   0x00000206 <+20>:	movw	r24, r28
   0x00000208 <+22>:	pop	r29
   0x0000020a <+24>:	pop	r28
   0x0000020c <+26>:	ret
End of assembler dump.```

gebart, 2017-02-09:
hmm, GH didn't auto-close this when #5688 was merged.
Fixed via #5688.

closed 2017-02-09 by gebart