RIOT-OS/RIOT #6501: boards/nucleo: Pin usage collision (SPI clk vs. LED0)

haukepetersen, 2017-01-27:
On most (all) STM nucleo boards we have a pin usage collision on pin PA05. It is used for both `LED0` and `SPI_DEV(0) sclk`. Now after the SPI remodeling was merged, the SPI busses are for some boards now initialized before the LED pin, leading to an overridden pin config and a not working SPI clock line...

Now the question is how to proceed here. 

@kaspar030: would you mind to check the pin configuration (MODER register) for your F334/encx24 setup quickly? Thx

kaspar030, 2017-01-30:
> @kaspar030: would you mind to check the pin configuration (MODER register) for your F334/encx24 setup quickly? Thx

I can confirm that by commenting the led configuration in board.c, the encx24j600 works on nucleo-f334, on current master. With the led setup in place, it doesn't work.

bergzand, 2017-02-05:
Confirmed this for the nucleo-f401, the LED is now a really nice SPI activity LED

haukepetersen, 2017-02-07:
Awesome board layout...

But any idea how we could fix this? The simplest way would be not to expose the LED, but do we want this?

aabadie, 2017-03-28:
Any solution found regarding this problem ?

bergzand, 2017-03-28:
I would disable the exposure of the LED. Maybe, but not necessarily guarded with some kind of define. 

My main arguments are that this LED has caused enough hard to debug SPI problems and I don't consider it a critical feature of the board. It is easier for most people to strap another LED to some GPIO than to move the clock line of the SPI.

PeterKietzmann, 2017-03-28:
Disable LED for SPI

haukepetersen, 2017-03-28:
+1 for dropping the LED!

At some point (once the FEATURES thingies are working es expected) we could make this dependent on SPI being present, but for now I fear we need to disable the LED per default...

aabadie, 2017-03-28:
hopefully, this LED is defined in a single place for nucleo-64 boards ;)

aabadie, 2017-03-28:
I think I read too fast.

So in the end, to summarize, the idea is to remove the `gpio_init(LED0_PIN)` call in board.c for nucleo-64 boards (I'll also check if the problem is there with the 32 and 144 pins versions) ?

haukepetersen, 2017-03-28:
yes. Or more precise, ifdef it out per default and enable it only if stated explicitly by the make system for now.

aabadie, 2017-03-28:
ok, got it. Is it ok if I take it ?

PR #6813 (2017-03-30): boards/nucleos: deactivate LED0 auto init by default
Should fix #6501 

The problem doesn't exist with 32 and 144 pins versions of the nucleo boards.

I accept any better proposition for the macro name.
closed 2017-03-30 by aabadie