RIOT-OS/RIOT #6502: cpu/stm32l1: RTC broken

haukepetersen, 2017-01-27:
Seems like #6448 broke the RTC for the `stm32l1` (at least on the `nucleo-l1`): the CPU is stuck when calling `rtc_init()`. Investigating this now...

haukepetersen, 2017-01-27:
seems like they are all related...

haukepetersen, 2017-01-27:
the `l1` was somehow broken really really bad -> we need to work on our regression testing!

haukepetersen, 2017-01-27:
found the error: my `nucleo-l1` does not have an external low-speed oscillator equipped... Time for some more granular configuration I guess...

vincent-d, 2017-01-27:
@haukepetersen I was about to ask if there was a LSE. I remember when reading the nucleo user manual, that some parts number have small differencies regarding the LSE.

It's not easy to solve, but I can imagine different options:
- duplicate the boards for the different part numbers: definititely not optimal
- improve the periph_conf.h comments, it's then up to the developper to update the file: maybe the easiest
- improve the rtc driver to avoid looping infinitely waiting for the LSE to be ready, and return an error (but rtc_init is `void rtc_init(void)`) or fallback to the LSI, but the LSI might not be precise enough to run the rtc correctly: more complicated and we're not sure to cover every corner case

haukepetersen, 2017-01-27:
The solution is IMHO simple:  #6504

aabadie, 2017-11-04:
This one can be closed since #7504 was merged.

closed 2017-11-04 by aabadie