RIOT-OS/RIOT #6510: Rename methods uart_stdio_init/read/write

antmerlino, 2017-01-27:
I believe renaming uart_stdio_init to simply stdio_init allows for a simpler way of switching where the stdio gets routed.  For any module currently using stdio functionality, uart_stdio_init/read/write are called even if the functionality isn't actually a uart. 

For example, on cortexm processors stdio is configured through syscall implementations.  The init call gets called before main through the C runtime startup procedure.  _write and _read are by default implemented, using uart_stdio.  

Segger RTT was implemented and has to "act" like a uart driver.
There are discussions of implementing USB and using a CDC VCP implementation to service stdio.
ETHOS is also implemented inside of the uart_stdio driver using defines.

What I propose is to refactor the interface to simply be:
```
void stdio_init (void);
int stdio_read (char* buffer, int count);
int stdio_write (const char* buffer, int len);
```

Then, anytime the application wants to utilize stdio, it should include a single XXX_stdio module.

1. ethos_stdio would become it's own module, which would essentially just be a similar version of uart_stdio but without the defines and 'normal' UART code.
2. uart_stdio would stay its own module but without ethos code
3. rtt could remove the concept of "uart" by simply refactoring
4. When USB support is done, a usb_stdio module can be made that initializes USB and sets up a CDC VCP configuration.

Of course, it also allows easier expansion for other types of stdio modules as well.  

One concern of course is including more than one stdio module, but that seems to at least be partially an issue already since rtt_stdio and uart_stdio would be defining the same functions.

Let me know if there is anything I'm not considering.  Otherwise, I can take care of the refactoring (although I'm not sure I can be the one testing RTT or Ethos).

Best,
Anthony

antmerlino, 2017-01-28:
@kaspar030 I have started (and potentially finished this update).  See PR #6513

miri64, 2018-10-01:
I think this was implemented by @haukepetersen in https://github.com/RIOT-OS/RIOT/pull/9503. Reopen if you disagree.

closed 2018-10-01 by miri64