RIOT-OS/RIOT #6519: driver/mrf24j40: broken on stm32f4discovery

haukepetersen, 2017-01-30:
the newly merged driver for the `mrf24j40` is broken on the `stm32f4discovery`. From quick analyis, the driver hangs during initialization while setting the channel, explicitly it loops in the function `mrf24j40_reset_state_machine()`. IMHO, this is not an SPI issues (as we expected earlier), as all SPI transfers before that, and also this specific one look just fine on the logic analyzer. I expect rather something to be broken in the initialization code, but won't have time to look into this any deeper myself...

PeterKietzmann, 2017-01-30:
Probably this is timing related. Following "test" makes the driver work for me with stm32f4discovery. It is still strange that [this loop](https://github.com/RIOT-OS/RIOT/blob/master/drivers/mrf24j40/mrf24j40_getset.c#L476) needs several SPI transfers until the device returns `MRF24J40_RFSTATE_RX` whereas with an Atmel there is just one call.

@bergzand can you confirm and have a look at this?

```
From 3ef44199c9ba544875833c7e32000136126a08fb Mon Sep 17 00:00:00 2001
From: PeterKietzmann <peter.kietzmann@haw-hamburg.de>
Date: Mon, 30 Jan 2017 17:06:27 +0100
Subject: [PATCH] uncomment spi opt

---
 cpu/stm32_common/periph/spi.c | 44 +++++++++++++++++++++----------------------
 1 file changed, 22 insertions(+), 22 deletions(-)

diff --git a/cpu/stm32_common/periph/spi.c b/cpu/stm32_common/periph/spi.c
index 60f0a85..a8cbe7c 100644
--- a/cpu/stm32_common/periph/spi.c
+++ b/cpu/stm32_common/periph/spi.c
@@ -165,34 +165,34 @@ void spi_transfer_bytes(spi_t bus, spi_cs_t cs, bool cont,
         gpio_clear((gpio_t)cs);
     }
 
-    /* transfer data, use shortpath if only sending data */
-    if (!inbuf) {
-        for (size_t i = 0; i < len; i++) {
-            while (!(dev(bus)->SR & SPI_SR_TXE));
-            *DR = outbuf[i];
-        }
-        /* wait until everything is finished and empty the receive buffer */
-        while (dev(bus)->SR & SPI_SR_BSY) {}
-        while (dev(bus)->SR & SPI_SR_RXNE) {
-            dev(bus)->DR;       /* we might just read 2 bytes at once here */
-        }
-    }
-    else if (!outbuf) {
-        for (size_t i = 0; i < len; i++) {
-            while (!(dev(bus)->SR & SPI_SR_TXE));
-            *DR = 0;
-            while (!(dev(bus)->SR & SPI_SR_RXNE));
-            inbuf[i] = *DR;
-        }
-    }
-    else {
+    // /* transfer data, use shortpath if only sending data */
+    // if (!inbuf) {
+    //     for (size_t i = 0; i < len; i++) {
+    //         while (!(dev(bus)->SR & SPI_SR_TXE));
+    //         *DR = outbuf[i];
+    //     }
+    //     /* wait until everything is finished and empty the receive buffer */
+    //     while (dev(bus)->SR & SPI_SR_BSY) {}
+    //     while (dev(bus)->SR & SPI_SR_RXNE) {
+    //         dev(bus)->DR;       /* we might just read 2 bytes at once here */
+    //     }
+    // }
+    // else if (!outbuf) {
+    //     for (size_t i = 0; i < len; i++) {
+    //         while (!(dev(bus)->SR & SPI_SR_TXE));
+    //         *DR = 0;
+    //         while (!(dev(bus)->SR & SPI_SR_RXNE));
+    //         inbuf[i] = *DR;
+    //     }
+    // }
+    // else {
         for (size_t i = 0; i < len; i++) {
             while (!(dev(bus)->SR & SPI_SR_TXE));
             *DR = outbuf[i];
             while (!(dev(bus)->SR & SPI_SR_RXNE));
             inbuf[i] = *DR;
         }
-    }
+    // }
 
     /* make sure the transfer is completed before continuing, see reference
      * manual(s) -> section 'Disabling the SPI' */
-- 
1.9.1
```

PeterKietzmann, 2017-01-30:
@bergzand do you even connect the `WAKE` line of the radio? I never did but now I've seen that we actually set the "Immediate Wake-up mode" [here](https://github.com/RIOT-OS/RIOT/blob/master/drivers/mrf24j40/mrf24j40.c#L66) and according to the datasheet sec. 3.15.2 one has to apply a high level to wake up the device.

bergzand, 2017-01-30:
I've optimized the wake pin away since the same functionality can be achieved via SPI commands, but I forgot to remove the SPI commands that enable the pin.  I should have some time this week to test out your suggestion on a nucleo and discovery board. It also makes sense to see if the delay can be tuned such that the SPI loop here is not needed.

PeterKietzmann, 2017-01-31:
@bergzand that would be great, thanks! I will analyse the SPI driver and all configurations on the stm32f4discovery later (again). In case everything looks as expected, I will count it as a radio driver "bug" and mark the mrf24j40 support as "initial" or "experimental" in the 2017.01 release notes.

bergzand, 2017-02-07:
@PeterKietzmann I tested a bit today. I checked and apparently I did not forget to remove the WAKEPAD enable bit, thus the wake pin was always disabled and the level applied to it should not matter. The driver is currently using the second method to enable and disable sleep mode (register based sleep and wake mode). The immwake register means that the radio wake immediately when signalled and not periodically based on a timer.

I tried your patch, and somehow seem to fix the initialization sequence. Everything looks working then, including the radio with some basic ping tests

PeterKietzmann, 2017-02-08:
Thanks for testing @bergzand. Are you aware of any SPI timing requirements for the radio? Additionally we need to analyse the stm32_common SPI driver in more depth.

phectori, 2017-03-01:
The stm32 datasheet states:

> During discontinuous communications, there is a 2 APB clock period delay between the write operation to SPI_DR and the BSY bit setting. As a consequence, in transmit-only mode, it is mandatory to wait first until TXE is set and then until BSY is cleared after writing the last data.

I think adding `while (!(dev(bus)->SR & SPI_SR_TXE)) {}` before line [175](https://github.com/RIOT-OS/RIOT/blob/master/cpu/stm32_common/periph/spi.c#L175) solves the problem. (so waiting for TXE to be set before checking the BSY bit.)

It solves the problem for me on a nucleo-f401 board. I do not have a stm32f4discovery at my disposal. 

I hope this solves the problem on the stm32f4discovery too.

bergzand, 2017-03-01:
@phectori thanks for the research on this issue. I should have some time again this weekend for RIOT-os, so I will test your suggestion then.

bergzand, 2017-03-17:
@phectori I've tested your sugestion and it works. What I think happens is that after writing to the SPI_DR register, we check the BSY flag. Due to the mentioned delay, this is not yet set and thus no wait until the BSY flag is cleared here. The RXNE is still cleared at this point and the SPI_DR is never read, the buffer thus never cleared.

My hypothesis is that with the long register read at [127](https://github.com/RIOT-OS/RIOT/blob/master/drivers/mrf24j40/mrf24j40_internal.c#L127) being a 3 byte SPI transfer, the result returned is actually the result of the second spi_transfer_byte instead of the third transfer. I would have to test whether this is a correct assumption, since the structure with a short register read is not much different, but those seem functional.

PeterKietzmann, 2017-03-30:
I will give it another try and close the issue again then

PeterKietzmann, 2017-03-30:
Works great now :-)!

closed 2017-03-30 by PeterKietzmann

PeterKietzmann, 2017-06-08:
@phectori could you check if the mrf24j40 driver still works on nucleo-f401 with current master?!

PeterKietzmann, 2017-06-09:
@phectori no need for this any more. At least not for nucleo-f401.

PR #6764 (2017-03-30): stm32_common/spi: fix receive after transmit-only error
This line fixes SPI transmissions on at least the faster class of stm32 devices.

The main problem is that there is a slight delay between the SPI transfer register being written and the BUSY bit being set. From the [datasheet of the stm32f405..439](http://www.st.com/resource/en/reference_manual/dm00031020.pdf) (20MB warning)
```
During discontinuous communications, there is a 2 APB clock period delay between the
write operation to SPI_DR and the BSY bit setting. As a consequence, it is mandatory to
wait first until TXE=1 and then until B [... cut]