RIOT-OS/RIOT #6520: Issue with microcoap_server example

paterson, 2017-01-30:
Hi there!

I'm relatively new to Riot, so first off, many thanks for the extensive documentation and work that has gone into the project. I'm looking to create a simple BLE + IPV6 + CoAP stack, so the microcoap_server example is perfect. I have two issues that are probably quite straightforward but I can't quite deal with. For reference, I'm using an NRF52dk board and have a Mac running macOS Sierra.

1. My `main` function in `main.c` does not run at all if I include the `gnrc_netdev_default` module in my Makefile like this:

       USEMODULE += gnrc_netdev_default

      I have tested this using the microcoap_server and the gnrc_minimal examples, which both fail to ever invoke the main function as they have that module in their makefiles, and also by adding it to the makefile of the hello_world example, which breaks after adding it (even though I don't use it at all). Is there any reason my main.c wouldn't be invoked as a result of this? Seems odd that adding it to the Makefile but never using it would cause this.

2. Even when I know that `main.c` is invoked (by toggling LED macros), I can't seem to log anything to the pyterm window. Irrespective of whether I use `printf` or `puts`, no output appears after flashing and calling `make term`. Is there something I am missing to get logging configured on the device? I am getting a warning saying `'No handlers could be found for logger "root"`, is that the issue? If so, is there a way to resolve it? 

If anyone has thoughts on these issues, I would really appreciate it. I feel like they are probably simple fixes but a little lost on what to do to fix them.

Many thanks!

closed 2017-01-30 by paterson

OlegHahm, 2017-01-31:
First of all: welcome to RIOT and thanks for the nice words about our documentation. ;-)

Out of curiosity: Did you manage to solve your problem?

paterson, 2017-02-02:
@OlegHahm Yes thank you! It was just a faulty board, so moving to an older NRF51 did the trick! Thanks!