RIOT-OS/RIOT #6532: native broken on OS X

kYc0o, 2017-01-31:
#4058 broke the building on native for OS X:

```console
RIOT/core/sched.c:61:47: error: argument to 'section' attribute is not valid for this target: mach-o section
      specifier requires a segment and section separated by a comma
__attribute__((used)) __attribute__((section (".openocd")))
                                              ^
RIOT/core/sched.c:67:47: error: argument to 'section' attribute is not valid for this target: mach-o section
      specifier requires a segment and section separated by a comma
__attribute__((used)) __attribute__((section (".openocd")))
                                              ^
2 errors generated.
```

@cgundogan is Jenkins now also compiling for OS X?
@kaspar030 What's Murdock state on this?

zpymyyn, 2017-02-01:
Me also met with this, I think it might relate with the Tuntap not updated to EI captain?

smlng, 2017-02-01:
@zpymyyn this is a compiler issue on macOS related to some optimizations for openocd, I don't think its related to the tuntap issue on macOS. Please try the fix in  #6534 and report back if it works or not, thanks!

zpymyyn, 2017-02-01:
@smlng  Thanks a lot! It works!

PR #6534 (2017-02-01): core: fix osx compile issue
fixes #6532
closed 2017-02-01 by kYc0o