RIOT-OS/RIOT #6533: tests/lwip target board for python test is hardcoded to native

BytesGalore, 2017-02-01:
The python driven [test](https://github.com/RIOT-OS/RIOT/blob/master/tests/lwip/tests/01-run.py) does not run for boards other then `native`
But since the default `BOARD` for is set to `iotlab-m3` for this test, I guess they should be the primary target.

Setting the boards and ttys manually in the 01_test.py [1]
``` python
Board("iotlab-m3", None, "/dev/ttyUSBX")
```
 forces the script to start with these parameters but it fails when trying to call reset [2] before going on with testing:
```
RIOT/tests/lwip$ make BOARD=iotlab-m3 test
./tests/01-run.py
Testing for (<Board 'iotlab-m3',port=None,serial='/dev/ttyUSB0'>, <Board 'iotlab-m3',port=None,serial='/dev/ttyUSB2'>): 
Traceback (most recent call last):
  File "./tests/01-run.py", line 270, in <module>
    [test_ipv6_send, test_udpv6_send, test_dual_send])
  File "./tests/01-run.py", line 172, in execute
    board_group.reset()
  File "./tests/01-run.py", line 153, in reset
    board.reset(application, env)
  File "./tests/01-run.py", line 124, in reset
    self.reset_strategy.execute(application, env)
  File "./tests/01-run.py", line 71, in execute
    super(ResetStrategy, self).__run_make(application, ("reset",), env)
AttributeError: 'super' object has no attribute '_ResetStrategy__run_make'
```

[1] https://github.com/RIOT-OS/RIOT/blob/master/tests/lwip/tests/01-run.py#L264
[2] https://github.com/RIOT-OS/RIOT/blob/master/tests/lwip/tests/01-run.py#L168

kYc0o, 2018-01-16:
Actually I'm not sure we can use the test script for any board as is, since it configures two tap interfaces to execute the tests.

Therefore, another test for radio platforms should be created.

Can we close the issue @BytesGalore ?

miri64, 2020-07-01:
Hopefully fixable once #11406 and follow-ups are in.

maribu, 2022-09-16:
@miri64: ping?

miri64, 2022-09-19:
See https://github.com/RIOT-OS/RIOT/pull/14874