RIOT-OS/RIOT #6535: Problem flashing Riot to Arduino UNO/Mega2560 boards

yijiazi, 2017-02-01:
Hi, 

I'm trying to flash Riot to Arduino boards --

mali, 2017-02-01:
Hum .. don't seem related with arduino API ?
@yijiazi  Which avrdude version ?

yijiazi, 2017-02-01:
@mali : the avrdude is 

 avrdude -v

avrdude: Version 6.2
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

I think it's the default one provided by the vagrant VM

kYc0o, 2017-02-02:
I think the problem lies on the fact that you're using the vagrant VM. It comes with predefined default access to several boards which provide UART connectivity, and usually it works good so far. At the same time, if the resources of the host are not so high, of course it has a consequence on the communication with external hardware.

Which kind of native OS are you using? Be aware that RIOT can be built in any Linux or OS X based system.

cgundogan, 2017-02-02:
@yijiazi could you try `BOARD=arduino-uno make flash` instead of the direct call to `avrdude`? One thing I spotted is that you use a baud rate of `115200`, while RIOT is using `9600` (see `RIOT/board/arduino-common/Makefile.include`

OlegHahm, 2017-02-02:
> Hum .. don't seem related with arduino API ?

Totally true - my fault.

mali, 2017-02-02:
I use avrdude: Version 6.1, compiled on Sep 11 2014 at 20:00:34 and never had this kind of problem.
As suggested by @Kycoo, can you try flashing directly from your MacOS host ?

yijiazi, 2017-02-03:
@cgundogan : What I used was `BOARD=arduino-uno make flash `, I simply copied the part about the flashing. The default baud rate used by the `BOARD=arduino-uno make flash` was 11520. 
I also tried with other rates from 9600, it doesn't work either. 

But I'll try Riot without using the VM as @mali and @kYc0o suggested, and come back to you later. 

thanks a lot.

yijiazi, 2017-02-03:
OK, I tried it without using VM -- apparently, it works fine now. 

Now the problem is solved -- thanks very much!

OlegHahm, 2017-02-03:
I guess we can close then?

cgundogan, 2017-02-04:
Before we close this issue I would actually like to confine the problem. AFAIK, flashing arduino boards (at least used to) work with a linux host. @yijiazi is using a OSX machine and we had several problems before with OSX. Can anyone with an OSX host ( @kYc0o  ?) try to reproduce this error by flashing an `arduino-uno` from within vagrant?

shaolinkhoa, 2017-03-17:
hi,
I have the problem while flashing Arduino Mega2560 board too.

khoa@khoa: ~/RIOT/examples/arduino_hello-world$ BOARD=arduino-uno make flash

Building application "arduino_hello-world" for "arduino-uno" with MCU "atmega328p".

In file included from /home/khoa/RIOT/drivers/include/periph/uart.h:48:0,
                 from /home/khoa/RIOT/sys/arduino/include/serialport.hpp:24,
                 from /home/khoa/RIOT/sys/arduino/include/arduino.hpp:29,
                 from /home/khoa/RIOT/examples/arduino_hello-world/_sketches.cpp:1:
/usr/lib/gcc/avr/4.9.2/include/stdint.h:9:26: fatal error: stdint.h: No such file or directory
 # include_next <stdint.h>
                          ^
compilation terminated.
/home/khoa/RIOT/Makefile.base:70: recipe for target '/home/khoa/RIOT/examples/arduino_hello-world/bin/arduino-uno/arduino_hello-world/_sketches.o' failed
make[1]: *** [/home/khoa/RIOT/examples/arduino_hello-world/bin/arduino-uno/arduino_hello-world/_sketches.o] Error 1
/home/khoa/RIOT/examples/arduino_hello-world/../../Makefile.include:275: recipe for target 'all' failed
make: *** [all] Error 2

Would you mind helping me ?

aabadie, 2017-03-17:
Hi,

Did you try on a Debian-like system ? If yes, can you try this:
```
sudo apt-get install avr-libc
```

mali, 2017-03-17:
arduino Mega2560 or Uno ?
if it's a Mega2560 you should use `make BOARD=arduino-mega2560`  ?
Have you installed avr-libc ?

shaolinkhoa, 2017-03-17:
Sorry for my misstake, I copied when I tried with Uno.
After installed avr-libc, it works. Thanks all.

closed 2017-03-17 by aabadie