RIOT-OS/RIOT #6549: Riot OS + CoAP + OSX

paterson, 2017-02-02:
Hi Guys,

Thanks for all the documentation that comes with Riot; it's been great for getting started as a newbie!

Just a quick question. I have my NRF51 board running the `microcoap_server` example, and it works great on that side, but when I'm trying to request the payload from `/riot/board`, I can't seem to get it working. 

I run the following command:

      coap-client "coap://[fe80::ff:fe00:e71a]/riot/board"

Where `fe80::ff:fe00:e71a` is the inet6 addr for '../128  scope: local [multicast]'. This outputs:

      v:1 t:0 tkl:0 c:1 id:29332
      Feb 02 18:38:58 CRIT coap_send: sendto

And just hangs there. I suspect I have not set up my bridge/tap properly, or have not used the correct url IP Address.

Would anyone have ideas about this or know how to run this example on OSX? Many thanks!

smlng, 2017-02-02:
Whats your network/communication setup between your Mac and the board? How are they connected?

paterson, 2017-02-03:
Hey! Thanks for replying! Right now all I have is the USB connection, so I figure I need to setup the bridge/tap stuff right? That's where I'm a little lost so any info on that for OSX would be great!

smlng, 2017-02-09:
@paterson bridge/tap isn't fully functional on macOS compared to Linux, however, basic stuff is possible. You have to install [tuntap](http://tuntaposx.sourceforge.net/), e.g. via [Homebrew](http://brew.sh): `brew cask install tuntap`. You may have to adapt permission, please refer to tuntap help/FAQ. 

Afterwards, you should be able to use RIOTs `tapsetup` script in `dist/tools/tapsetup`. What works is RIOT native <--> macOS via `bridge0`, not working is the RIOT <--bridge--> RIOT setup with 2 RIOT native instances. Please have a look a this [README](https://github.com/smlng/riot-tools/tree/master/tcp_listen) might be helpful, too.

paterson, 2017-02-09:
Hi @smlng. Right I get you, thanks for the clarification! So because I'm looking to communicate between a RIOT board and my host operating system, I need to essentially find commands that do the equivalent of this, but on mac:

sudo ip tuntap add tap0 mode tap user ${USER}
sudo ip link set tap0 up

Is that possible? I.e RIOT board <--> macOS?

Thanks again!
Niall

smlng, 2017-02-09:
RIOT native <--> macOS is working (somewhat) as described above. RIOT board <--> macOS would require something like ethos or slip, IIRC both are not working on macOS right now. But I haven't tried lately ...

As an alternative, you could run Linux inside a VM and pass the USB where the board is connected to the VM. Use Vagrant for an easy setup, look up the [tutorials](https://github.com/RIOT-OS/Tutorials) for further infos.

smlng, 2017-03-06:
@paterson: are you getting along, and is this solved?

smlng, 2017-07-06:
@paterson what's your status on this, did you get it working, so we can close this issue?

PeterKietzmann, 2019-02-21:
No response for > 1,5 years. *examples/microcoap * doesn't exists anymore but this seemd more like a problem with `tap` on macOS anyway. @paterson if you ever face this issue again, please use the mailing list and give an updated report of that issue.

closed 2019-02-21 by PeterKietzmann