RIOT-OS/RIOT #6571: cpu/lpc11u34 : GPIO mode

ks156, 2017-02-07:
Is it safe to mask the mode that way : https://github.com/RIOT-OS/RIOT/blob/master/cpu/lpc11u34/periph/gpio.c#L77-L78 ?

Can we be sure that the gpio_mode_t members will have a correct binary value ?
I've connected a debugger, and here's the respective member values : 

* GPIO_IN -> 0
* GPIO_IN_PD -> 8
* GPIO_IN_PU -> 16
* GPIO_OUT-> 1
* GPIO_OD -> 1025
* GPIO_OD_PU -> 1041

These values looks counterintuitive, and I don't know how to interpret them. Anyway, if these values are really those used to set the GPIO mode, it will not works as expected.

Why not assigning values to this enum : 

```
typedef enum {
    GPIO_IN = 0,               /**< configure as input without pull resistor */
    GPIO_IN_PD = 1,             /**< configure as input with pull-down resistor */
    GPIO_IN_PU = 2,             /**< configure as input with pull-up resistor */
    GPIO_OUT = 3,               /**< configure as output in push-pull mode */
    GPIO_OD = 4,                /**< configure as output in open-drain mode without
                             *   pull resistor */
    GPIO_OD_PU = 5              /**< configure as output in open-drain mode with
                             *   pull resistor enabled */
} gpio_mode_t;
```

In any case, I think even if the new method is much more elegant than a switch statement, the switch is the only way to configure the mode correctly.

Maybe I'm missing something. All comments and informations are welcome

gebart, 2017-02-07:
See https://github.com/RIOT-OS/RIOT/blob/master/cpu/lpc11u34/include/periph_cpu.h#L70-L89 for a piece of the puzzle.
The lpc11u34 CPU periph driver has its own device specific gpio_mode_t where some hardware register bit masks are encoded as the values of the enum. The default gpio_mode_t definition that you quoted above is from the generic periph header under drivers/include, but it will not be used on lpc11u34 because of `HAVE_GPIO_MODE_T` in periph_cpu.h

ks156, 2017-02-07:
Wow, how have I missed this ? That's no **a** piece, that's **the** piece ! Thanks a lot @gebart 👍 

Now I understand the values I read when I debug the code.

(oh, and this issue can be closed, of course)

closed 2017-02-07 by miri64