RIOT-OS/RIOT #6578: RIOT+OpenThread

rmadhuraj, 2017-02-08:
Hi group,

I am working on RIOT with OpenThread which is cloned from 
https://github.com/jremmert-phytec-iot/RIOT/tree/kw2xrf
https://github.com/RIOT-OS/RIOT/pull/5552
Now, I want to test the functionality of openthread on SAMR21. Can anyone help me on how to test the openthread functionality. 
1. **How many minimum nodes required?**
2. How can I make sure that openthread application is working properly.

Thanks & Regards
Madhu

kYc0o, 2017-02-08:
See #5552, maybe you can synchronise with @jia200x .

kYc0o, 2017-02-08:
For extensive tests the IoT-Lab testbed was used.

jia200x, 2017-02-08:
Hi @rmadhuraj 

As @kYc0o pointed out, the #5552 is the current PR for OpenThread. There's an ongoing update (OT has changed a lot since July 2015). The new version is compiling from OT building, but stopped working properly (I'm on it ;) ).

The PR version is just for testing Thread capabilities, but still wouldn't use it for real cases because lacks commissioning and border router. Anyway if you want to try mesh network communication, UDP, etc., you can use it.

Concerning your questions:

1. There's no minimum (but of course at least you need 2 if you want to test communication). The thread network will automatically select a single leader, routers and children based in the number of nodes.
2. IoT-Lab (https://www.iot-lab.info/) is a good tool to test this kind of networks. We tested communication in rows of nodes (4 hops from end to end) and worked. When nodes density is high, sometimes a couple of nodes crashes (but this for sure will be fixed in the update). 

I suggest to try the test/openthread_test and examples/openthread_posix_example to test OT. It should work "as it" on SAMR21. If you have any questions don't hesitate to ask again.

Cheers.

rmadhuraj, 2017-02-09:
Hi jia200x,

                Thanks . I tried  compiling tests/othread ,, but its throwing me the error...

madhu.r@CPU-245U:~/Opentread+RIOT/RIOT-kw2xrf/tests/othread$ make
Building application "openthread_example" for "native" with MCU "native".

"make" -C /home/madhu.r/Opentread+RIOT/RIOT-kw2xrf/pkg/openthread
make -C /home/madhu.r/Opentread+RIOT/RIOT-kw2xrf/tests/othread/bin/pkg/native/openthread
/home/madhu.r/Opentread+RIOT/RIOT-kw2xrf/tests/othread/main.cpp:4:28: fatal error: cli/cli-serial.h: No such file or directory
 #include <cli/cli-serial.h>
                            ^
compilation terminated.
make[1]: *** [/home/madhu.r/Opentread+RIOT/RIOT-kw2xrf/tests/othread/bin/native/openthread_example/main.o] Error 1
make: *** [all] Error 2

so where can i get that missing library ??

jia200x, 2017-02-09:
Hi @rmadhuraj 

You are compiling it for "native". You have to compile it for SAMR21:

`make BOARD=samr21-xpro all flash`

Then:

`make BOARD=samr21-xpro term`

Cheers!

rmadhuraj, 2017-02-10:
Hi @jia200x 

                  Thanks for your time. I did the same for SAMR21-XPRO as well . But still i am facing the same Error...

 **madhu.r@CPU-245U:~/Opentread+RIOT/RIOT-kw2xrf/tests/othread$ make BOARD=samr21-xpro all flash**
Building application "openthread_example" for "samr21-xpro" with MCU "samd21".

"make" -C /home/madhu.r/Opentread+RIOT/RIOT-kw2xrf/pkg/openthread
make -C /home/madhu.r/Opentread+RIOT/RIOT-kw2xrf/tests/othread/bin/pkg/samr21-xpro/openthread
/home/madhu.r/Opentread+RIOT/RIOT-kw2xrf/tests/othread/main.cpp:4:28: fatal error: cli/cli-serial.h: No such file or directory
 #include <cli/cli-serial.h>
                            ^
compilation terminated.
make[1]: *** [/home/madhu.r/Opentread+RIOT/RIOT-kw2xrf/tests/othread/bin/samr21-xpro/openthread_example/main.o] Error 1
make: *** [all] Error 2
**madhu.r@CPU-245U:~/Opentread+RIOT/RIOT-kw2xrf/tests/othread$** 

Can you help me ?!

jia200x, 2017-02-10:
Hi @rmadhuraj 

Please try the code available at #5552. I'm not sure about the setup in the Phytec branch.
The code in #5552 was developed using a SAMR21-xpro, so it should work out of the box.

Cheers

jia200x, 2017-10-25:
@rmadhuraj is the initial issue solved?

OpenThread is now available in RIOT's master

rmadhuraj, 2017-10-26:
@jia200x yes, it is solved. Thank you. The reason is older gcc version.

closed 2017-10-26 by rmadhuraj