RIOT-OS/RIOT #6583: native broken on Mint17 (gcc 4.8.4)

haukepetersen, 2017-02-09:
#5688 breaks native for on my host:
```
/home/hauke/dev/riot/RIOT/core/atomic_c11.c:40:23: fatal error: stdatomic.h: No such file or directory
```
using `gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4` the default stock gcc on Mint 17

I know the gcc is old, but I guess I am not the only one using the default stock gcc on their linux machine (and not willing to upgrade once every half a year?!). So any idea to solve this?

haukepetersen, 2017-02-09:
Actually the same seems to be true for the `avr8` platforms -> `avr-gcc (GCC) 4.8.2`.

haukepetersen, 2017-02-09:
solved it by upgrading to some newer gcc version. Still I am wondering, what we can do to help people that run these old compiler versions, as it seems to me this might be something quite a few people run into?!

kaspar030, 2017-02-09:
> I know the gcc is old, but I guess I am not the only one using the default stock gcc on their linux machine (and not willing to upgrade once every half a year?!). 

Mint 17 is based on Ubuntu 14.04 LTS. We sometimes ago decided that we'd support only the newest LTS, with a grace period of maybe half a year...

> So any idea to solve this?

Upgrade GCC! Should be possible. Maybe just use the ubuntu 16.04 packages.

haukepetersen, 2017-02-09:
> solved it by upgrading to some newer gcc version

did. 

> We sometimes ago decided that we'd support only the newest LTS

good to know, seems like I missed that. So fine with me, I consider this issue as solved.

closed 2017-02-09 by haukepetersen