RIOT-OS/RIOT #6588: mips: Build system is not aligned with other platforms

gebart, 2017-02-10:
There are some issues with the current Makefile.include for the recently added mips CPU family.

 1. [ ] `USEMODULE += newlib` is missing/commented out
 2. [ ] Makefile requires `$(MIPS_ELF_ROOT)/share/mips/rules/mipshal.mk`
 3. [ ] Doesn't build with BUILD_IN_DOCKER=1 unless the host system also has the tool chain installed, effectively eliminating the benefit of using a Docker image with all tools.
 4. [ ] Does not use the shared TOOLCHAIN= and Makefile.gnu settings files.

1. There are parts of the code, especially in #5616, where we need to take a particular path if newlib is used as the C library (as opposed to msp-libc or avr-libc). It requires that we can test `#if MODULE_NEWLIB` within the C code.

2. I didn't get the mipshal.mk file when I installed a mips bare-metal toolchain from scratch on my system (GCC+Newlib). Looking at the file in the riot/riotbuild Docker image I can see that we could just copy parts of it into cpu/mips_common and use it from there.

3. The Docker issues would be resolved if the makefile did not have the ifeq/error clause at the top and did not depend on the mipshal.mk file being found on the system.

4. Using the shared toolchain Makefiles makes it easier to support other toolchains in the future, for example LLVM/Clang, as alternatives to the standard GCC toolchain. It also takes care of setting all relevant tool command names so that we don't have to repeat the commands for every CPU family which uses GCC.

neiljay, 2017-02-13:
I'll address these issues one at a time:

> USEMODULE += newlib is missing/commented out

This compiles in /sys/newlib/syscalls, which conflicts with the implementation in our toolchain crt:

`/user/neil.jones/src/RIOT/examples/default/bin/mips-malta/newlib_syscalls_default/syscalls.o: In function `_init':
/user/neil.jones/src/RIOT/sys/newlib/syscalls.c:63: multiple definition of `_init'
/opt/imgtec/Toolchains/mips-mti-elf/2016.05-03/bin/../lib/gcc/mips-mti-elf/4.9.2/mipsel-r2-hard-newlib/lib/crti.o:(.init+0x0): first defined here
/user/neil.jones/src/RIOT/examples/default/bin/mips-malta/newlib_syscalls_default/syscalls.o: In function `_init':
/user/neil.jones/src/RIOT/sys/newlib/syscalls.c:63: multiple definition of `_fini'
/opt/imgtec/Toolchains/mips-mti-elf/2016.05-03/bin/../lib/gcc/mips-mti-elf/4.9.2/mipsel-r2-hard-newlib/lib/crti.o:(.fini+0x0): first defined here
/user/neil.jones/src/RIOT/examples/default/bin/mips-malta/newlib_syscalls_default/syscalls.o: In function `_init':
/user/neil.jones/src/RIOT/sys/newlib/syscalls.c:63: multiple definition of `_exit'
`
Note we need to use our crt for multiple reasons.

This also relates to your point 1. I Think a better scheme to detect if newlib is in use is need other than including this module.

neiljay, 2017-02-13:
>  Makefile requires $(MIPS_ELF_ROOT)/share/mips/rules/mipshal.mk
> I didn't get the mipshal.mk file when I installed a mips bare-metal toolchain from scratch on my system (GCC+Newlib). 

Where did you get this toolchain from and what version ? We could potentially move this file into RIOT but I want to avoid potentially locking us into specific versions of the toolchain. (there are interdependencies between this file and the linker scripts which also reside in the toolchain).

neiljay, 2017-02-13:
> Doesn't build with BUILD_IN_DOCKER=1 unless the host system also has the tool chain installed, effectively eliminating the benefit of using a Docker image with all tools.

Really ?

`
$ cd /opt/imgtec/Toolchains/

$ sudo mv mips-mti-elf/ was_mips-mti-elf

$ cd ~/RIOT

$ docker run -i -t -u $UID -v ~/RIOT:/data/riotbuild --env BOARD=mips-malta riotbuild make -C examples/default clean all
make: Entering directory '/data/riotbuild/examples/default'
Building application "default" for "mips-malta" with MCU "mips32r2_common".

"make" -C /data/riotbuild/boards/mips-malta
"make" -C /data/riotbuild/core
"make" -C /data/riotbuild/cpu/mips32r2_common
"make" -C /data/riotbuild/cpu/mips32r2_common/periph
"make" -C /data/riotbuild/drivers
"make" -C /data/riotbuild/drivers/saul
"make" -C /data/riotbuild/sys
"make" -C /data/riotbuild/sys/auto_init
"make" -C /data/riotbuild/sys/auto_init/saul
"make" -C /data/riotbuild/sys/fmt
"make" -C /data/riotbuild/sys/phydat
"make" -C /data/riotbuild/sys/ps
"make" -C /data/riotbuild/sys/saul_reg
"make" -C /data/riotbuild/sys/shell
"make" -C /data/riotbuild/sys/shell/commands
   text    data     bss     dec     hex filename
  57652    2428    5980   66060   1020c /data/riotbuild/`

> The Docker issues would be resolved if the makefile did not have the ifeq/error clause at the top
Other  devs asked for this, I have no issues using docker ?

kaspar030, 2017-02-13:
>> Doesn't build with BUILD_IN_DOCKER=1 unless the host system also has the tool chain installed, effectively eliminating the benefit of using a Docker image with all tools.
> 
> $ sudo mv mips-mti-elf/ was_mips-mti-elf

Try unsetting the mips variables (which ought not to be set if the toolchain is not installed.

Also, please try ```BUILD_IN_DOCKER=1 BOARD=mips-malta make -C examples/hello-world```, which should currently be broken, unless the MIPS_* variables are st.

The docker image has the toolchain installed, so if the build is started from within, building works fine. That's why the CI doesn't show a problem.

neiljay, 2017-02-13:
> Does not use the shared TOOLCHAIN= and Makefile.gnu settings files

Makefile.GNU must be getting pulled in somehow see: 
https://github.com/RIOT-OS/RIOT/pull/6066/commits/d273ecd36ba5ed5e61bb4bcf4fa4c1e9aa17c8e6

We need to do this as objcopy is broken in our toolchain, so having some flexibility to override these setting is required, we should be able to move the MIPS port to use TOOLCHAIN and Makefile.gnu explicitly though fairly easily (as long as we can override the variables from board/cpu makefiles if/when needed).

neiljay, 2017-02-13:
> Try unsetting the mips variables (which ought not to be set if the toolchain is not installed.

That still works if you manually invoke docker:

$ export MIPS_ELF_ROOT=
16:17[njones@njones-linux(x86_64)~/RIOT]
$ docker run -i -t -u $UID -v ~/RIOT:/data/riotbuild --env BOARD=mips-malta riotbuild make -C examples/default clean all
make: Entering directory '/data/riotbuild/examples/default'
Building application "default" for "mips-malta" with MCU "mips32r2_common".

"make" -C /data/riotbuild/boards/mips-malta
"make" -C /data/riotbuild/core
"make" -C /data/riotbuild/cpu/mips32r2_common
"make" -C /data/riotbuild/cpu/mips32r2_common/periph
"make" -C /data/riotbuild/drivers
"make" -C /data/riotbuild/drivers/saul
"make" -C /data/riotbuild/sys
"make" -C /data/riotbuild/sys/auto_init
"make" -C /data/riotbuild/sys/auto_init/saul
"make" -C /data/riotbuild/sys/fmt
"make" -C /data/riotbuild/sys/phydat
"make" -C /data/riotbuild/sys/ps
"make" -C /data/riotbuild/sys/saul_reg
"make" -C /data/riotbuild/sys/shell
"make" -C /data/riotbuild/sys/shell/commands
   text    data     bss     dec     hex filename
  57652    2428    5980   66060   1020c /data/riotbuild/examples/default/bin/mips-malta/default.elf
make: Leaving directory '/data/riotbuild/examples/default'

It looks like the issue with BUILD_IN_DOCKER is it initially invokes make on the host system, how is that ever supposed to work ?

gebart, 2017-02-13:
@neiljay what does your bundled crti `_init` function actually do? Maybe it belongs in the cpu initialization in RIOT?

If we don't have the RIOT syscalls then nothing in RIOT will be able to interact with newlib. I guess you have implemented basic uart write for printf, but I don't see how any of the other calls will work without cooperating with the OS. This will in particular break the POSIX compatibility functions in #5616.

gebart, 2017-02-13:
@neiljay 
>It looks like the issue with BUILD_IN_DOCKER is it initially invokes make on the host system, how is that ever supposed to work ?

This is a convenience helper for letting new users build stuff for all platforms without having to manually install a toolchain for the target (you still need GNU make, obviously) and to avoid having to type a really long `docker run` command line.

If you try it for the other targets (e.g. `make BUILD_IN_DOCKER=1 BOARD=stm32f4discovery`) you'll see that the Makefile will set up the environment in Docker for you, including all possible extra environment variables that may affect the outcome, as well as mapping volumes for both the OS and the application project. The compiler existence checks and include path mappings are skipped on the host side and only executed by the containerized make inside Docker.

Regarding my toolchain, I just built it using Gentoo's crossdev, which uses the vanilla GCC sources, newlib and binutils to build the toolchain. I built mine for the target triple `mips-mti-elf`, simply to match what was suggested in the Makefile.include for mips-common. Are the headers and extra files (mipshal.mk etc.) available stand-alone or only as part of the pre-built toolchain package on imgtec.com?

neiljay, 2017-02-13:
@gebart 

>I guess you have implemented basic uart write for printf

"Use the source Luke" ;-)  
https://github.com/RIOT-OS/RIOT/blob/master/cpu/mips32r2_common/thread_arch.c

No the toolchain provides a full semi-hosting layer which can interact with a debugger or bootloader or the application can intercept the syscalls themselves and choose to handle them (we can optionally handle the write syscall to stdout or stderr in the riot/mips port forwarding to the uart).

The spec is here:

http://wiki.prplfoundation.org/w/images/4/42/UHI_Reference_Manual.pdf

We could still support a VFS we just need to plug in at a different point.

> what does your bundled crti _init function actually do?

Architecture specific initialisation, GP setup, Correct C++ initialisation is the big one, Why duplicate this code into RIOT, we would then have to maintain 2 copies, again source can be found in:

$MIPS_ELF_ROOT/share/mips/hal/

> Are the headers and extra files (mipshal.mk etc.) available stand-alone or only as part of the pre-built toolchain package on imgtec.com?

You need the pre-built toolchains from IMG, upstream GCC is lacking in latest architecture features, let alone all the extra stuff that come with libgloss. I did state this clearly on the original pull request and once the PIC32 board stuff gets merged I'll add lots of info to the Wiki for building on MIPS. (We are working on upstreaming everything but its a slow process).

neiljay, 2017-02-13:
@gebart 

wrt issue 3, how about:

`
diff --git a/cpu/mips32r2_common/Makefile.include b/cpu/mips32r2_common/Makefile.include
index d0c9b7a..a9e0dbd 100644
--- a/cpu/mips32r2_common/Makefile.include
+++ b/cpu/mips32r2_common/Makefile.include
@@ -1,6 +1,8 @@
 ifndef MIPS_ELF_ROOT
+ifneq ($(BUILD_IN_DOCKER),1) #Don't error when BUILD_IN_DOCKER=1 as it _is_ set in DOCKER
     $(error "Please set $$(MIPS_ELF_ROOT) and ensure $$(MIPS_ELF_ROOT)/bin is on your PATH")
 endif
+endif

 \# Target triple for the build.
 export TARGET_ARCH ?= mips-mti-elf
@@ -10,7 +12,9 @@ export MEMORY_BASE=0x80000000
 export MEMORY_SIZE=1M
 export APP_START=0x80000000

+ifneq ($(BUILD_IN_DOCKER),1) #Don't error when BUILD_IN_DOCKER=1 as MIPS_ELF_ROOT _is_ set in DOCKER
 include $(MIPS_ELF_ROOT)/share/mips/rules/mipshal.mk
+endif

 \# define build specific options
 export CFLAGS_CPU   = -EL -march=mips32r2 -std=gnu99
~
`
This works for me:

`
$ export MIPS_ELF_ROOT=
18:52[njones@njones-linux(x86_64)~/RIOT/examples/default]
$ make BUILD_IN_DOCKER=1 BOARD=mips-malta
Launching build container using image "riot/riotbuild:latest".
docker run --rm -i -t -u "$(id -u)" \
            -v '/home/ldap/njones/RIOT:/data/riotbuild/riotbase' \
            -v '/home/ldap/njones/RIOT/cpu:/data/riotbuild/riotcpu' \
            -v '/home/ldap/njones/RIOT/boards:/data/riotbuild/riotboard' \
            -v '/home/ldap/njones/RIOT:/data/riotbuild/riotproject' \
            -v /etc/localtime:/etc/localtime:ro \
            -e 'RIOTBASE=/data/riotbuild/riotbase' \
            -e 'CCACHE_BASEDIR=/data/riotbuild/riotbase' \
            -e 'RIOTCPU=/data/riotbuild/riotcpu' \
            -e 'RIOTBOARD=/data/riotbuild/riotboard' \
            -e 'RIOTPROJECT=/data/riotbuild/riotproject' \
            -e 'BOARD=mips-malta' \
            -w '/data/riotbuild/riotproject/examples/default/' \
            'riot/riotbuild:latest' make  'BOARD=mips-malta'
Building application "default" for "mips-malta" with MCU "mips32r2_common".

"make" -C /data/riotbuild/riotbase/core
"make" -C /data/riotbuild/riotbase/drivers
"make" -C /data/riotbuild/riotbase/drivers/saul
"make" -C /data/riotbuild/riotbase/sys
"make" -C /data/riotbuild/riotbase/sys/auto_init
"make" -C /data/riotbuild/riotbase/sys/auto_init/saul
"make" -C /data/riotbuild/riotbase/sys/fmt
"make" -C /data/riotbuild/riotbase/sys/phydat
"make" -C /data/riotbuild/riotbase/sys/ps
"make" -C /data/riotbuild/riotbase/sys/saul_reg
"make" -C /data/riotbuild/riotbase/sys/shell
"make" -C /data/riotbuild/riotbase/sys/shell/commands
"make" -C /data/riotbuild/riotboard/mips-malta
"make" -C /data/riotbuild/riotcpu/mips32r2_common
"make" -C /data/riotbuild/riotcpu/mips32r2_common/periph
   text    data     bss     dec     hex filename
  57644    2428    5980   66052   10204 /data/riotbuild/riotproject/examples/default/bin/mips-malta/default.elf
`

neiljay, 2017-02-15:
@gebart points 3 and 4 are now addressed in #6066 and #6092 see:
https://github.com/RIOT-OS/RIOT/pull/6092/commits/1c827f3c1786aabd93c3d4130085691b8e50ba31
and
https://github.com/RIOT-OS/RIOT/pull/6066/commits/4413a76aaa571fd8c1a4fa573ea943051c769968

gebart, 2017-02-17:
@neiljay I think we should try to find the compiler in `$PATH` before erroring out on a missing environment variable. The toolchain root can be found relative from the compiler executable. Saves on setup if you have the compiler already in your path.

Btw, I looked at the mipshal.mk file and I still think it should be merged into `mips-common/Makefile.include`. There's basically only one relevant code path in that file, the one that chooses the 32 bit linker script and sets the cflags and linker flags for it. The tool command names should already be handled by Makefile.gnu which chooses the correct gcc, objcopy, objdump etc. command names.

neiljay, 2017-11-16:
I believe all these issue are resolved by #6892.
There does appear to be further issues in the MIPS build setup as discussed in #8019 and #8032.
Can we close this and concentrate on fixing the new issues ?

gebart, 2017-11-17:
Yes, I agree

closed 2017-11-17 by gebart