RIOT-OS/RIOT #6595: Regression: Build with TOOLCHAIN=llvm fails on cortexm and native

pyropeter, 2017-02-13:
```
$ make -C examples/hello-world/ TOOLCHAIN=llvm -j1 BOARD=iotlab-m3
make: Entering directory '/$RIOTBASE/examples/hello-world'
Building application "hello-world" for "iotlab-m3" with MCU "stm32f1".

"make" -C /$RIOTBASE/boards/iotlab-m3
"make" -C /$RIOTBASE/boards/iotlab-common
"make" -C /$RIOTBASE/core
/$RIOTBASE/core/atomic_c11.c:354:6: error: definition of builtin function '__sync_synchronize'
void __sync_synchronize(void) {
     ^
1 error generated.
make[2]: *** [/$RIOTBASE/Makefile.base:64: /$RIOTBASE/examples/hello-world/bin/iotlab-m3/core/atomic_c11.o] Error 1
make[1]: *** [/$RIOTBASE/Makefile.base:20: ALL--/$RIOTBASE/core] Error 2
make: *** [/$RIOTBASE/examples/hello-world/../../Makefile.include:275: all] Error 2
make: Leaving directory '/$RIOTBASE/examples/hello-world'
```

The same error occurs on `native`, but there are further issues on that platform:

```
$ make -C examples/hello-world/ TOOLCHAIN=llvm BOARD=native
make: Entering directory '/$RIOTBASE/examples/hello-world'
Building application "hello-world" for "native" with MCU "native".

"make" -C /$RIOTBASE/boards/native
"make" -C /$RIOTBASE/core
"make" -C /$RIOTBASE/cpu/native
"make" -C /$RIOTBASE/drivers
"make" -C /$RIOTBASE/sys
"make" -C /$RIOTBASE/boards/native/drivers
"make" -C /$RIOTBASE/cpu/native/periph
"make" -C /$RIOTBASE/sys/auto_init
/$RIOTBASE/core/atomic_sync.c:146:1: error: conflicting types for '__sync_lock_test_and_set_1'
TEMPLATE_SYNC_LOCK_TEST_AND_SET_N(1)        /* __sync_lock_test_and_set_1 */
^
/$RIOTBASE/core/atomic_sync.c:47:10: note: expanded from macro 'TEMPLATE_SYNC_LOCK_TEST_AND_SET_N'
    I##n __sync_lock_test_and_set_##n (I##n *ptr, I##n desired) \
         ^
<scratch space>:22:1: note: expanded from here
__sync_lock_test_and_set_1
^
/$RIOTBASE/core/atomic_sync.c:146:1: note: '__sync_lock_test_and_set_1' is a builtin with type 'char (volatile char *, char, ...)'
/$RIOTBASE/core/atomic_sync.c:47:10: note: expanded from macro 'TEMPLATE_SYNC_LOCK_TEST_AND_SET_N'
    I##n __sync_lock_test_and_set_##n (I##n *ptr, I##n desired) \
         ^
<scratch space>:22:1: note: expanded from here
__sync_lock_test_and_set_1
^
/$RIOTBASE/core/atomic_sync.c:146:1: error: definition of builtin function '__sync_lock_test_and_set_1'
TEMPLATE_SYNC_LOCK_TEST_AND_SET_N(1)        /* __sync_lock_test_and_set_1 */
^
/$RIOTBASE/core/atomic_sync.c:47:10: note: expanded from macro 'TEMPLATE_SYNC_LOCK_TEST_AND_SET_N'
    I##n __sync_lock_test_and_set_##n (I##n *ptr, I##n desired) \
         ^
<scratch space>:22:1: note: expanded from here
__sync_lock_test_and_set_1
^
/$RIOTBASE/core/atomic_sync.c:147:1: error: conflicting types for '__sync_lock_test_and_set_2'
TEMPLATE_SYNC_LOCK_TEST_AND_SET_N(2)        /* __sync_lock_test_and_set_2 */
^
/$RIOTBASE/core/atomic_sync.c:47:10: note: expanded from macro 'TEMPLATE_SYNC_LOCK_TEST_AND_SET_N'
    I##n __sync_lock_test_and_set_##n (I##n *ptr, I##n desired) \
         ^
<scratch space>:24:1: note: expanded from here
__sync_lock_test_and_set_2
^
/$RIOTBASE/core/atomic_sync.c:147:1: note: '__sync_lock_test_and_set_2' is a builtin with type 'short (volatile short *, short, ...)'
/$RIOTBASE/core/atomic_sync.c:47:10: note: expanded from macro 'TEMPLATE_SYNC_LOCK_TEST_AND_SET_N'
    I##n __sync_lock_test_and_set_##n (I##n *ptr, I##n desired) \
[... 192 lines cut ...]
    bool __sync_bool_compare_and_swap_##n (I##n *ptr, I##n expected, I##n desired) \
         ^
<scratch space>:24:1: note: expanded from here
__sync_bool_compare_and_swap_2
^
/$RIOTBASE/core/atomic_sync.c:157:1: note: '__sync_bool_compare_and_swap_2' is a builtin with type 'bool (volatile short *, short, short, ...)'
/$RIOTBASE/core/atomic_sync.c:77:10: note: expanded from macro 'TEMPLATE_SYNC_BOOL_COMPARE_AND_SWAP_N'
    bool __sync_bool_compare_and_swap_##n (I##n *ptr, I##n expected, I##n desired) \
         ^
<scratch space>:24:1: note: expanded from here
__sync_bool_compare_and_swap_2
^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
/$RIOTBASE/core/atomic_c11.c:354:6: error: definition of builtin function '__sync_synchronize'
void __sync_synchronize(void) {
     ^
1 error generated.
make[2]: *** [/$RIOTBASE/Makefile.base:64: /$RIOTBASE/examples/hello-world/bin/native/core/atomic_c11.o] Error 1
make[2]: *** Waiting for unfinished jobs....
20 errors generated.
make[2]: *** [/$RIOTBASE/Makefile.base:64: /$RIOTBASE/examples/hello-world/bin/native/core/atomic_sync.o] Error 1
make[1]: *** [/$RIOTBASE/Makefile.base:20: ALL--/$RIOTBASE/core] Error 2
make: *** [/$RIOTBASE/examples/hello-world/../../Makefile.include:275: all] Error 2
make: Leaving directory '/$RIOTBASE/examples/hello-world'
```

smlng, 2017-02-13:
Just tried on macOS, can confirm this issue. Needs further investigation!

Would be good to have llvm tested for native by our CI.

gebart, 2017-02-13:
Easiest workaround is to just ifdef the whole sync file away when using
clang. The file is only needed when using a gcc from 4.6 or earlier, i.e.
the msp boards.

gebart, 2017-02-14:
btw, what version of Clang was it that caused this error?
I don't see this error on Linux with `clang version 3.9.1 (tags/RELEASE_391/final)`

closed 2017-02-14 by kaspar030

pyropeter, 2017-02-14:
I used `clang version 3.9.1 (tags/RELEASE_391/final)` on Linux, too. Why did you not get the error?

PR #6598 (2017-02-14): llvm: hide redefines of atomic sync operations
fixes #6595