RIOT-OS/RIOT #6603: native: 64 Bit support ?

brummer-simon, 2017-02-14:
Hi,

I have read today that Arch Linux seems to discontinue x86 32-bit 
platform support in summer 2017.

Other distributions might follow in the future. Are there any plans to 
get rid of natives dependencies to its manditory 32-bit libraries?

miri64, 2017-02-14:
This doesn't make much sense IMHO. We don't have support for any other 32-bit platform.

brummer-simon, 2017-02-15:
@miri64 - I dont get what you mean. I am just saying that Linuxes support for i386 starts to vanish sooner or later, therfore "native" should buildable under x86_64 without dependencies to i386 libraries.

kaspar030, 2017-02-15:
@brummer-simon Arch only discuntinues 32bit only installations. they will not drop support for 32bit userland binaries.

Having native compile for 32bit would be nice-to-have, but IMHO is not worth the effort of fixing the codebase (yet).

miri64, 2017-02-15:
@brummer-simon 
> @miri64 - I dont get what you mean. I am just saying that Linuxes support for i386 starts to vanish sooner or later, therfore "native" should buildable under Linux on x86_64 without dependencies to i386 libraries.

I mean, why making the native platform vastly different, only because one distro is dropping toolchain support for it (which it isn't, see @kaspar030's comment). There are always ways to install software beyond the official repositories of a Linux distribution (in Arch even easier than most other distros IMHO), so why enforce a very different architecture (64bit on native vs (32bit, 16bit or 8bit) on all other platforms) just because you can't `apt-get install gcc-multilib` anymore?

@kaspar030
> Having native compile for 32bit would be nice-to-have, but IMHO is not worth the effort of fixing the codebase (yet).

You mean 64bit?

miri64, 2017-02-15:
> […] just because you can't `apt-get install gcc-multilib` anymore?

I mean of course `pacman -S gcc-multilib` ^^"

LudwigKnuepfer, 2017-02-25:
| We don't have support for any other 32-bit platform.

You mean 64bit?

LudwigKnuepfer, 2017-02-25:
I highly doubt any major Linux distribution will drop 32-bit support (available via multilib for the x86_64 platform) in the foreseeable future.
That being said there are pros and cons.

Pro:
- improve RIOT's code base by fixing 64 related errors
- simplify toolchain setup for native

Con:
- effort / gain (as it's unlikely we will support any other 64 bit platform, fixing potential 64 bit errors in RIOT is less beneficial)
- decreased 32 bit testing (unless we make 32 the default in which case the toolchain setup argument is diminished)

From my perspective the pros don't justify pulling manpower from other tasks.

LudwigKnuepfer, 2017-02-25:
Closing as memo, reopen if you disagree.

LudwigKnuepfer, 2017-02-25:
PS:
I added my answer to the [FAQ](https://github.com/RIOT-OS/RIOT/wiki/FAQ#why-is-there-no-64-bit-support-at-least-in-native) - if I missed some important point please update there as well.

miri64, 2017-02-27:
>> We don't have support for any other 32-bit platform.
>
> You mean 64bit?

Yes ^^"

daid, 2018-01-03:
Just as an extra note on this issue: Installing "gcc-multilib" on debian jessie removes certain cross compilers. It's not uncommon to have cross compilers installed on embedded development environments.

> daid@monolith:~/RIOT/examples/hello-world$ sudo apt-get install gcc-multilib
> [sudo] password for daid:
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> The following packages will be REMOVED:
>  crossbuild-essential-armhf g++-4.9-arm-linux-gnueabihf g++-arm-linux-gnueabihf gcc-4.9-arm-linux-gnueabihf gcc-arm-linux-gnueabihf

If support for 16bit and 8bit architectures is already there, then it feels odd to say: But 64bit is too much effort without gain. Any 32->64bit bug you will have is a hidden bug for 32->16/8bit, effecting most likely performance/memory usage.

miri64, 2019-11-14:
From https://github.com/RIOT-OS/RIOT/issues/6603#issuecomment-282468710

> I highly doubt any major Linux distribution will drop 32-bit support (available via multilib for the x86_64 platform) in the foreseeable future.

Apparently this is now happening (spearheaded by Ubuntu):

- https://ubuntu.com/blog/statement-on-32-bit-i386-packages-for-ubuntu-19-10-and-20-04-lts
- https://steamcommunity.com/app/221410/discussions/0/1638675549018366706/ (Steam was one of the major opponents of 32-bit removal in Ubuntu as many games are still written for 32-bit architectures and the container approach seems to be a compromise solution to make 32-bit applications still be able to run on Linux without 32-bit support, so Ubuntu might go ahead and remove 32-bit support after all)

So I think we need to re-open this discussion. 

>     * decreased 32 bit testing (unless we make 32 the default in which case the toolchain setup argument is diminished)

We wouldn't have to. We could have basically three "boards": `native32` (equivalent to the current `native` board), `native64` (`native` with 64-bit support) and `native` (which is an alias for the version most suitable for the platform it is running on). For testing we could still run `native32` (if we find a container it can run in).

miri64, 2019-12-03:
I've [heard rumors](https://github.com/RIOT-OS/RIOT/pull/12517#issuecomment-561088672) that @x3ro is working on `native64`, so I'm assigning him to this issue.

miri64, 2019-12-03:
@x3ro did you have a look at https://github.com/RIOT-OS/RIOT/issues/6603#issuecomment-554131105?

x3ro, 2019-12-05:
I have now 😅 So yeah, I started working on this, I think it was beginning of this year, and did get it to compile on Linux 64-bit, but was then scared off by debugger segfaults that I didn't care to debug 🤷‍♂ I might have some time to look into this later this month, potentially also at 36C3, but at this point I can't promise much. I'll try to attend the January Hack'n'Ack though.

x3ro, 2019-12-16:
I've started a branch over at https://github.com/x3ro/RIOT/tree/x64 – I've been focussing only on Intel/Linux x64 for now. It currently segfaults because the context switching assembly is still broken, but I have a rough idea what needs to be done. But if you ignore this, it does compile and run the hello world example on 64-bit native :D

InspireSemi, 2022-05-31:
Any update on adding support for 64 bit.. I really think RIOT will quickly be left behind if this is not done.

chrysn, 2024-03-16:
We now do have a native64 board; was anything from this issue left unaddressed?

maribu, 2024-05-02:
Fixed by https://github.com/RIOT-OS/RIOT/pull/20315

closed 2024-05-02 by maribu