RIOT-OS/RIOT #6611: SPI for samr21 board in  RIOT-2017.01

shreeharsha22, 2017-02-15:
May I know if SPI driver is tested and stable for samr21 board in latest RIOT-2017.01. If any one has already worked on it.  Facing some clock related issues.

Thanks in advance.

Regards,
shreeharsha I

PeterKietzmann, 2017-02-15:
Yes it was tested. You can have a look at [this table](https://docs.google.com/spreadsheets/d/1_IrBtKyhkOyKK8NxQIHOJ2fdyExqNqMucxDWPLdBNQI/edit?pli=1#gid=0) which contains information about what was tested. However, we introduced a complete SPI rework with this release and we collect leftover topics in #6437. Can you elaborate on your clock related issues?

shreeharsha22, 2017-02-15:
hai am doing the below steps for spi communication
where I have selected 2nd spi device in i,e configuring for sercom3
static const spi_conf_t spi_config[] = {
        {:
        },
        {
        },
         }
        .dev      = &SERCOM3->SPI,
        .miso_pin = GPIO_PIN(PA, 16),
        .mosi_pin = GPIO_PIN(PA, 18),
        .clk_pin  = GPIO_PIN(PA, 19),
        .miso_mux = GPIO_MUX_D,
        .mosi_mux = GPIO_MUX_D,
        .clk_mux  = GPIO_MUX_D,
        .miso_pad = SPI_PAD_MISO_0,
        .mosi_pad = SPI_PAD_MOSI_2_SCK_3
    }
};

while(1){
spi_acquire(2,SPI2_CS,SPI2_MODE,SPI_CLK_400KHZ);
spi_transfer_bytes(2,SPI2_CS, false, headerBuffer, dummyread, headerLength);  /*performing write dummmy read*/
 spi_release(2);

spi_acquire(2,SPI2_CS,SPI2_MODE,SPI_CLK_400KHZ);
spi_transfer_bytes(2,SPI2_CS,false,dummywrite,readBuffer,readlength);  /*performing read*/
spi_release(2);
}

Thanks in advance.
Regards,
Shreeharsha I

PeterKietzmann, 2017-02-16:
- please check your braces
- can you try your code with a given [default configuration](https://github.com/RIOT-OS/RIOT/blob/master/boards/samr21-xpro/include/periph_conf.h#L183), e.g. the second defined SPI port has pinouts
- what exactly is the problem? Please state what you see and what you've expected.

immesys, 2017-02-24:
I will also confirm that something is very wrong with SPI in SAMR21 in master now. Deep sleep mode now consumes 1.5mA instead of 4uA. I have bisected it to between d587cf38df5a630804212864c5644fe9e9981e53 and 236f1eddddb2755c9fefbdf118e3e98f97075b4b which isn't super helpful because more than 60k lines changed there, most related to SPI rework. I'm not finished searching yet but wanted to confirm that something is up with the new spi code.

PeterKietzmann, 2017-11-27:
On that device we provide a configuration for the second SPI bus which is 
```
    {
        .dev      = &SERCOM5->SPI,
        .miso_pin = GPIO_PIN(PB, 2),
        .mosi_pin = GPIO_PIN(PB, 22),
        .clk_pin  = GPIO_PIN(PB, 23),
        .miso_mux = GPIO_MUX_D,
        .mosi_mux = GPIO_MUX_D,
        .clk_mux  = GPIO_MUX_D,
        .miso_pad = SPI_PAD_MISO_0,
        .mosi_pad = SPI_PAD_MOSI_2_SCK_3
    }
```
I've tested this configuration and it works. Power consumption has been discussed/addressed elsewhere. @shreeharsha22 I consider this issue as solved and close it. Please give me a sign if you need further help

closed 2017-11-27 by PeterKietzmann