RIOT-OS/RIOT #6620: gcoap: broken handling of ACK messages?

haukepetersen, 2017-02-16:
I am not quite sure, but is the behavior as shown below as intended? It shows sending a confirmable GET request from  a coap client (libcoap on linux) to a `gcoap` server (`native` RIOT on the same host). From my understanding, the second ACK should not be send, right? Or couldn't `gcoap` pickyback the content in an ACK message right away?

[image]

kb2ma, 2017-02-16:
Apologies for the confusion, @haukepetersen. gcoap does not support confirmable messaging at present, only non-confirmable. You are seeing gcoap copy the message type from the request into the response, which is a poor way to deal with the limitation.

The gcoap example [README](https://github.com/RIOT-OS/RIOT/tree/master/examples/gcoap) describes the limitations. The source documentation does not call it out though. I would be happy to add a quick PR to state this limitation clearly at the top of the documentation. Let me know what you think.

Presently I am working on implementation of the Observe extension (PR #6469). As I mentioned in my [last comment there](https://github.com/RIOT-OS/RIOT/pull/6469#issuecomment-278930486), I plan to add confirmable messaging next. I plan to remove the WIP label on the Observe PR this weekend. I would like to get confirmable messaging into 2017.04. The infrastructure for timing out the wait for a response already is there.

haukepetersen, 2017-02-17:
All good, I might as well read the doc... :-)

Maybe I can squeeze in some time next week to look into implementing support for confirmable message beginning of next week if that is alright with you.

Regarding this issue: I consider it solved and close it again.

closed 2017-02-17 by haukepetersen

kb2ma, 2017-02-17:
Of course, I'd be happy for your contributions! The client side is the challenging part, but gcoap already times out requests. See the use of gcoap_request_memo_t and the GCOAP_MEMO* macros. Also, @kaspar030 has implemented confirmable messaging for nanocoap in [nanocoap_sock.c](https://github.com/kaspar030/sock/blob/master/nanocoap/nanocoap_sock.c). Notice the constants already defined there that should be reused.

kb2ma, 2017-02-17:
One more thing -- why I implemented the timeout and callback on failure for NON. IMO, whether it's a NON request or a CON request, the application still may need to deal with the situation where it does not get a response to the request.

kb2ma, 2017-02-17:
I also maintain some wiki pages on gcoap. The [status page](https://github.com/kb2ma/RIOT/wiki/gcoap-Status) is a good place to start, and the features page will give you more background on the approach.