RIOT-OS/RIOT #6624: core: thread_flags seem to be broken on native

haukepetersen, 2017-02-17:
Thread flags seem to be broken on native. I was trying to run this code
```c
    thread_flags_clear(TFLAGS_ANY);

    while (1) {
        DEBUG("setting timer\n");
        xtimer_set(&timer, timeout);
        thread_flags_wait_any(TFLAGS_TIMEOUT);
        // xtimer_remove(&timer);
        DEBUG("timeout\n");
        DEBUG("flags are 0x%04x\n", (int)sched_active_thread->flags);
        DEBUG("cb and arg are %i and %p\n", (int)timer.callback, timer.arg);
    }
```
on `native`. But after setting the timer a second time, it gets stuck:
```
setting timer
timeout
flags are 0x0000
cb and arg are 134531956 and 0x807118c
setting timer
 -> stuck here
```

Next I tried to run `test/thread_flags`, but the test also failed:
```
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2017.04-devel-138-g8037a-schnubbi-add_mqttsn)
main starting
thread(): waiting for 0x1...
main(): setting flag 0x0001
thread(): received flags: 0x0001
thread(): waiting for 0x1 || 0x64...
main(): setting flag 0x0064
thread(): received flags: 0x0064
thread(): waiting for 0x2 && 0x4...
main(): setting flag 0x0001
main(): setting flag 0x0008
main(): setting flag 0x0002
main(): setting flag 0x0004
thread(): received flags: 0x0006
thread(): waiting for any flag, one by one
thread(): received flags: 0x0001
thread(): waiting for any flag, one by one
thread(): received flags: 0x0008
  --> stuck here
```

using `gcc (Ubuntu 6.2.0-3ubuntu11~14.04) 6.2.0 20160901` on Mint 17.3

kaspar030, 2017-02-21:
> Next I tried to run test/thread_flags, but the test also failed:

Actually, it doesn't fail. It just doesn't end. The thread waiting for flags exits, end main ends up in ```while(1) {}```. Will PR a fix.

kaspar030, 2017-02-21:
See #6636.

kaspar030, 2017-02-21:
I run your code. Seems like the second timer interrupt is not triggered.

kaspar030, 2017-02-21:
Problem is native specific, on samr21 your code runs fine.

kaspar030, 2017-02-21:
Somehow, the second "thread_yield_higher()" tries to switch to native's ISR context, but never arrives. Don't have time to dig deeper.

haukepetersen, 2017-02-22:
Would you mind opening a more specific issue and closing this one then? Thx

haukepetersen, 2017-02-23:
closed in favor of #6642

closed 2017-02-23 by haukepetersen