RIOT-OS/RIOT #6629: Add support to (all) ST Nucleo boards

aabadie, 2017-02-20:
There's an ongoing work which goal is to add support for (all) [ST Nucleo boards](http://www.st.com/en/evaluation-tools/stm32-mcu-nucleo.html) to RIOT.

The goal if this issue is to keep track of this work. For all boards, it lists some information, the actual support status (available upstream, available PR, not supported) and give the link to the documentation in the Wiki (if any).

| # | Board | Board Family | MCU Family | Status | Documentation | 
| -- | -------- | ------------------ | ------------------ | -------- | --------------------- |
| 1 | [Nucleo-L011K4](http://www.st.com/en/evaluation-tools/nucleo-l011k4.html) |  Nucleo 32 |   M0+ |  Not supported | No doc | 
| 2 | [Nucleo-F031K6](http://www.st.com/en/evaluation-tools/nucleo-f031k6.html)  |  Nucleo 32 |  M0 |  PR #6345 | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo32-F031) | 
| 3 | [Nucleo-F042K6](http://www.st.com/en/evaluation-tools/nucleo-f042k6.html)  |  Nucleo 32 |  M0 |  Supported | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo32-F042) | 
| 4 | [Nucleo-L031K6](http://www.st.com/en/evaluation-tools/nucleo-l031k6.html) |  Nucleo 32 |  M0+ |  Not supported | No doc | 
| 5 | [Nucleo-F303K8](http://www.st.com/en/evaluation-tools/nucleo-f303k8.html) |  Nucleo 32 |  M4 |  Supported | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo32-F303) | 
| 6 | [Nucleo-L432KC](http://www.st.com/en/evaluation-tools/nucleo-l432kc.html) |  Nucleo 32 |  M4 |  Not supported | No doc | 
| 7 | [Nucleo-F030R8](http://www.st.com/en/evaluation-tools/nucleo-f030r8.html) |  Nucleo 64  |  M0 |  Supported | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo-F030) | 
| 8 | [Nucleo-F302R8](http://www.st.com/en/evaluation-tools/nucleo-f302r8.html) |  Nucleo 64 |  M4 | PR #6615 | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo-F302) | 
| 9 | [Nucleo-F334R8](http://www.st.com/en/evaluation-tools/nucleo-f334r8.html)  |  Nucleo 64 |  M4 |  Supported | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo-F334)  | 
| 10 | [Nucleo-L053R8](http://www.st.com/en/evaluation-tools/nucleo-l053r8.html)  |  Nucleo 64 |  M0+ |  PR #6628  |  No Doc | 
| 11 | [Nucleo-F070RB](http://www.st.com/en/evaluation-tools/nucleo-f070rb.html)   |  Nucleo 64 |  M0 |  Supported | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo-F070)  | 
| 12 | [Nucleo-F072RB](http://www.st.com/en/evaluation-tools/nucleo-f072rb.html)  |  Nucleo 64 |  M0 |  Supported | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo-F072)  | 
| 13 | [Nucleo-F103RB](http://www.st.com/en/evaluation-tools/nucleo-f103rb.html)  |  Nucleo 64 |  M0 |  Supported | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo-F103)  | 
| 14 | [Nucleo-F410RB](http://www.st.com/en/evaluation-tools/nucleo-f410rb.html)  |  Nucleo 64 |  M4 |  PR #6025  | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo-F410)  | 
| 15 | [Nucleo-L073RZ](http://www.st.com/en/evaluation-tools/nucleo-l073rz.html)  |  Nucleo 64 |  M0+ |  PR #6433  | No doc  | 
| 16 | [Nucleo-F091RC](http://www.st.com/en/evaluation-tools/nucleo-f091rc.html)  |  Nucleo 64 |  M0 |  Supported | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo-F091)  | 
| 17 | [Nucleo-F303RE](http://www.st.com/en/evaluation-tools/nucleo-f303re.html)  |  Nucleo 64 |  M4 |  Supported | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo-F303)  | 
| 18 | [Nucleo-L152RE](http://www.st.com/en/evaluation-tools/nucleo-l152re.html)  |  Nucleo 64 |  M3 |  Supported | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo-L1)  | 
| 19 | [Nucleo-F401RE](http://www.st.com/en/evaluation-tools/nucleo-f401re.html)  |  Nucleo 64 |  M4 |  Supported | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo-F401)  | 
| 20 | [Nucleo-F446RE](http://www.st.com/en/evaluation-tools/nucleo-f446re.html)  |  Nucleo 64 |  M4 |  Supported | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo-F446)  | 
| 21 | [Nucleo-F411RE](http://www.st.com/en/evaluation-tools/nucleo-f411re.html)  |  Nucleo 64 |  M4 |  PR #5709  | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo-F411)  | 
| 22 | [Nucleo-L476RG](http://www.st.com/en/evaluation-tools/nucleo-l476rg.html)  |  Nucleo 64 |  M4 |  Not supported | No doc  | 
| 23 | [Nucleo-F303ZE](http://www.st.com/en/evaluation-tools/nucleo-f303ze.html)  |  Nucleo 144 |  M4 |  PR #6625  | No doc  | 
| 24 | [Nucleo-F446ZE](http://www.st.com/en/evaluation-tools/nucleo-f446ze.html)  |  Nucleo 144 |  M4 |  PR #6553  | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo144-F446)  | 
| 25 | [Nucleo-F722ZE](http://www.st.com/en/evaluation-tools/nucleo-f722ze.html)  |  Nucleo 144 |  M7 |  Not supported | No doc  | 
| 26 | [Nucleo-F207ZG](http://www.st.com/en/evaluation-tools/nucleo-f207zg.html)  |  Nucleo 144 |  M3 |  Supported | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo144-F207)  | 
| 27 | [Nucleo-F412ZG](http://www.st.com/en/evaluation-tools/nucleo-f412zg.html)  |  Nucleo 144 |  M4 |  Not supported | No doc  | 
| 28 | [Nucleo-F746ZG](http://www.st.com/en/evaluation-tools/nucleo-f746zg.html)  |  Nucleo 144 |  M7 |  Not supported | No doc  | 
| 29 | [Nucleo-F413ZH](http://www.st.com/en/evaluation-tools/nucleo-f413zh.html)  |  Nucleo 144 |  M4 |  PR #6565  | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo144-F413)  | 
| 30 | [Nucleo-F429ZI](http://www.st.com/en/evaluation-tools/nucleo-f429zi.html)  |  Nucleo 144 |  M4 |  Not supported | [Doc available](https://github.com/RIOT-OS/RIOT/wiki/Board%3A-Nucleo144-F429)  | 
| 31 | [Nucleo-F767ZI](http://www.st.com/en/evaluation-tools/nucleo-f767zi.html)  |  Nucleo 144 |  M7 |  Not supported | No Doc  |

aabadie, 2017-02-22:
Updated Nucleo-F411.

aabadie, 2017-02-23:
Updated Nucleo32-F031

aabadie, 2017-02-24:
Updated Nucleo144-F413

LudwigKnuepfer, 2017-02-25:
Just out of curiosity: why is this a goal?

aabadie, 2017-02-27:
>  why is this a goal?

It's a goal because it's difficult to follow the status of all ports for this family of boards : which one is in the code base, which one has a PR opened, which one is documented in the wiki. I think it also helps others following this work.
Not sure if that answers your question.

LudwigKnuepfer, 2017-02-28:
@aabadie I was interested in why it is a goal to support ALL Nucleo boards.

aabadie, 2017-02-28:
@LudwigKnuepfer, last year at the Paris Open Source Summit, I remember there was one guy talking about IoT who was comparing different OSes. He gave his opinion about RIOT : promising but with limited hardware support.
I think that supporting all Nucleo boards could give a good starting counter argument of what he said (even if he was more thinking of NXP MCUs).

Those boards covers a wide set of ST microcontrollers (from 2KB ram to more than 256KB, L0 to F7). I know some of them has limited interest for IoT (Nucleo-L011, the smallest for example) but it would be a good advertisement for RIOT to say : RIOT is supporting all Nucleo boards (like mbed) or RIOT is supporting all Arduino boards, etc
Let's target exhaustive hardware support ! (pipe dream inside)

LudwigKnuepfer, 2017-03-13:
I see, thank you for the explanation!
Also: thank you for the effort, sounds legit ;)

soleilos, 2017-08-11:
When I flash the Hello-world example on  NUCLEO-F767ZI, I receive the following flashing error:

/home/vagrant/RIOT/boards/nucleo144-f767/dist/openocd.cfg:5: Error: Can't find target/stm32f7x.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 60
at file "/home/vagrant/RIOT/boards/nucleo144-f767/dist/openocd.cfg", line 5
/home/vagrant/RIOT/examples/hello-world/../../Makefile.include:368: recipe for target 'flash' failed

Any thoughts is appreciated

gebart, 2017-08-11:
@soleilos most likely you need to update your OpenOCD version. It looks like your version is too old to support the stm32f7 CPUs.

soleilos, 2017-08-11:
@gebart Thank you, I updated it to openocd 0.10.0 and could flash it as well.

The new issue is that nothing is displayed on the terminal for a simple hello-world or riot_and_cpp examples, PyTerm:

Done flashing
/home/vagrant/RIOT/dist/tools/pyterm/pyterm -p "/dev/ttyACM0" -b "115200"
No handlers could be found for logger "root"
2017-08-11 12:38:38,573 - INFO # Connect to serial port /dev/ttyACM0
Welcome to pyterm!
Type '/exit' to exit.
ps
help
/exit

Tenderson, 2017-10-11:
Hi @gebart , 

I updated the OpenOCD and it flashed successfully. However, when I try ps and help, nothing is shown on the terminal, although these modules are added to the program.

I have the following settings:

$ gcc --version
gcc (Ubuntu/Linaro 6.3.0-18ubuntu2~16.04) 6.3.0 20170519
$ openocd --version
Open On-Chip Debugger 0.10.0
$ pip --version
pip 9.0.1 from /home/vagrant/.local/lib/python2.7/site-packages (python 2.7)

Here is the result.

2017-10-11 12:58:49,095 - INFO # Connect to serial port /dev/ttyACM0
Welcome to pyterm!
Type '/exit' to exit.
ps
help
ps

Any thoughts is appreciated.

aabadie, 2017-10-13:
@Tenderson, try to avoid cross posting for the same issue and create a dedicated one.

Maybe you need to update the ST Link of your board. See [this page](http://www.st.com/en/development-tools/stsw-link007.html) for more information.

kYc0o, 2017-10-13:
>Maybe you need to update the ST Link of your board. See this page for more information.

@aabadie I have already given that advice, but apparently it didn't work

PeterKietzmann, 2017-11-27:
@aabadie sorry if I don't get it from the above list, but arent't all boards ported now? If so, why don't we close this issue?

aabadie, 2017-11-27:
> If so, why don't we close this issue?

There are some remaining and the documentation is missing for some of the listed boards. I'd like to keep this one open for tracking the porting status.

PeterKietzmann, 2017-11-27:
Oh completely overseen them two open PRs. Will 'ask around' if someone has the boards around.

PeterKietzmann, 2019-07-17:
I updated the above list now that #8939 got merged. The only remaining board is Nucleo-H743ZI. Is it planned to support it? Otherwise I don't know why we should keep this issue open.

miri64, 2020-07-01:
@aabadie ping? Is there support planned for the Nucleo-H743ZI?

aabadie, 2020-07-01:
> Is there support planned for the Nucleo-H743ZI?

Yep, I got one in the office. I think @fjmolinas is interested by it hahaha ;)

maribu, 2022-09-16:
Still a goal?

I do have an STM32-H7 nucleo lying around that I would still like to see support for :)

aabadie, 2022-09-16:
> Still a goal?

I think so, but I doubt this issue will be resolved one day with ST releasing new boards all the time :)

Enoch247, 2024-03-15:
FYI: I started work on the STM32H7 family. I am using a Nucleo-STM32H723ZG (not listed in the table above). Made pretty good progress on the clock configuration today, which I will try to push before the weekend.

https://github.com/Enoch247/RIOT/tree/add-stm32h7

Enoch247, 2024-03-15:
> FYI: I started work on the STM32H7 family. I am using a Nucleo-STM32H723ZG (not listed in the table above). Made pretty good progress on the clock configuration today, which I will try to push before the weekend.
> 
> https://github.com/Enoch247/RIOT/tree/add-stm32h7

I won't be able to post today's code yet, but I've got it running on the HSI and can blink the board's user LEDs. Seem like things should get easier from here on out now.