RIOT-OS/RIOT #6632: PAN Coordinator Support

antmerlino, 2017-02-20:
I've seen this mentioned in a few different issues, but it seems there is no stand-alone issue regarding support for PAN configuration via a coordinator. 

I am interested in working on this feature but am unsure if I'd be able to do it alone without at minimum guidance, or more likely, actual help writing it.  Are there reasons that this hasn't been completed yet that I may not be considering?  Is there a general interest for this?

smlng, 2017-02-21:
Good point bringing this up (again), IIRC this is also an ongoing discussion for linux-wpan, maybe you chat with @alexaring on that topic, too.

PeterKietzmann, 2017-02-22:
@antmerlino I would say the general interest exists and there is no explicit reason (besides of time resources) why it hasn't been implemented yet. Concerning guidance I would say just start and come around with concrete questions in your PR, here or on the devel mailing list.

kaspar030, 2017-02-22:
@antmerlino I'd be really happy to see PAN coordinator support!

miri64, 2020-07-01:
Assigning @jia200x for now as he does a lot with PHY and MAC stuff lately.

jia200x, 2020-07-01:
In order to support this we actually need to implement a IEEE802.15.4 MAC. It will come right after #13943 :)

maribu, 2023-05-17:
@jia200x: Ping?